espressif/esp-idf #11758: ESP32: A fatal error occurred: Failed to connect to ESP32: No serial data received. after flash bootloader

FirzzyFunTWO, 2023-06-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

ESP32: A fatal error occurred: Failed to connect to ESP32: No serial data received. After I flash it

Hello, I am using ESP32DEVKITV1. I Flashed the ESP Cause no serial data to receive using https://espressif.github.io/esptool-js/ after all the flashing work but when I open the Arduino IDE. Serial communication show: 
```
rst:0x10 
 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT) invalid header: 0xffffffff invalid header: 0xffffffff invalid header: 0xffffffff invalid header: 0xffffffff invalid header: 0xffffffff invalid header: 0xffffffff invalid header: 0xffffffff
```

. so i reset it again using button in esp32 it works tho, but:
 

> Sketch uses 1064969 bytes (81%) of program storage space. Maximum is 1310720 bytes. Global variables use 38556 bytes (11%) of dynamic memory, leaving 289124 bytes for local variables. Maximum is 327680 bytes. esptool.py v4.5.1 Serial port COM6 Connecting...................................... A fatal error occurred: Failed to connect to ESP32: No serial data received. For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html Failed uploading: uploading error: exit status 2

conclusion esp32 serial data not receive not hardware problem. changing usb cable didn't fix

FirzzyFunTWO, 2023-06-28:
couldn't found esp32devkitv1 in technical documents

dobairoland, 2023-06-28:
`invalid header: 0xffffffff`

This indicates that flashing has not been done. Perhaps you have flashed the application but forgot to flash the bootloader and the partition table?

radimkarnis, 2023-06-28:
Hello @FirzzyFunTWO,
the serial communication works (you've received the `rst:0x10 
 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT) invalid header: 0xffffffff` message). Therefore the `No serial data received` error message seems weird.

some things to check:
- Have you selected the correct COM port when flashing?
- Does it work if you manually switch to download mode (press the RST button while holding the BOOT button)?
- Can you try downloading [esptool](https://github.com/espressif/esptool) and using it standalone? You can follow the [quick start guide](https://docs.espressif.com/projects/esptool/en/latest/esp32/#quick-start).

Thank you.

FirzzyFunTWO, 2023-06-28:
I selected right COM when flashing. pressing en while holding the boot button doens't solve it.
yes i can download the esptool and get in on enviroment

FirzzyFunTWO, 2023-06-28:
> i don't know i am following some tutorials in youtube

FirzzyFunTWO, 2023-06-28:
> Hello @FirzzyFunTWO, the serial communication works (you've received the `rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT) invalid header: 0xffffffff` message). Therefore the `No serial [...]

tried esptool. when connection its show no serial data receive tried to update com driver won't work?

radimkarnis, 2023-07-24:
Hi @FirzzyFunTWO, 
sorry for the delay, have you managed to solve your issue?

If you are able to see the boot log (the `rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 ...` part), that's a confirmation that the serial connection works. 

The `No serial data received` error can then only be explained by a non-functional automatic bootloader reset. You need to perform a manual reset and try flashing any binary with esptool.py. You can read about the manual reset [here](https://docs.espressif.com/projects/esptool/en/latest/esp32/advanced-topics/boot-mode-selection.html#manual-bootloader) or see [this video](https://www.youtube.com/live/zh-Y_s4X6zs?feature=share&t=1279) explaining everything.

If that still doesn't help, I am afraid there might be something wrong with your devkit.

radimkarnis, 2023-07-31:
Closing for inactivity. Feel free to reopen if needed.

closed 2023-07-31 by radimkarnis