espressif/esp-idf #11760: if IO can connect to the multi pad in touch slider?

swortain, 2023-06-28:
### Is your feature request related to a problem?

Hi, I got a touch slider on ESP32-S3(ESP-IDF 5.0), which have 6 pad, but I want to set the channel like this, the TOUCH_PAD_NUM11 and TOUCH_PAD_NUM13 used more than once. but it is possible to calculate the unique position of slider.

```
static const touch_pad_t channel_array[TOUCH_SLIDER_CHANNEL_NUM] = {
    // Touch slider channels array
    TOUCH_PAD_NUM11,
    TOUCH_PAD_NUM12,
    TOUCH_PAD_NUM13,
    TOUCH_PAD_NUM14,
    TOUCH_PAD_NUM11,
    TOUCH_PAD_NUM13,
};

```
if that is valid? I do remember it is valid when I read the docs somewhere, but I cann't find it out.
What's more I test it on my pcb, but it doesn't work.....

### Describe the solution you'd like.

It is theoretically possible to calculate the unique position, but it seemed that the SDK don't support it.

L-KAYA, 2023-06-29:
Unfortunately, we can't bind multiple touch channel to a same GPIO. Typically, the analog signals are fixed to a particular GPIO, only digital signals can be connected to different GPIOs via IO_MUX.

You can refer to the [S3 Technical Reference Manual](https://www.espressif.com.cn/sites/default/files/documentation/esp32-s3_technical_reference_manual_en.pdf) at chapter `39.2.4 Capacitive Touch Pins`

L-KAYA, 2023-06-29:
BTW, could you describe more  explicitly about how to `calculate the unique position of slider`?

swortain, 2023-06-29:
It's not multiple touch channel to a same GPIO, but multiple touchpad(the pad of touch slider) to a same touch channel.
For example we got 6 touchpad connect to GPIO 1,2,3,4,1,3, when finger touched touchpad 1, the value of IO1 it highest, IO2 it higher than IO 3,4, when finger touched  touchpad 5, the value of GPIO1 is highest as same, but what's different IO3,4 is higher than IO2. So we can recognize the difference between the two situation.
I do remember read the way somewhere but maybe not in espressif's document...

L-KAYA, 2023-06-29:
Ah, you're right! I got what you mean. Please allow me to take some time on evaluating this feature, as there should be some strict limitations to use a same pad multiple times. For example, the number of pad between two same pads, the adjacent pad should be different, etc.

swortain, 2023-06-29:
ok, it will be greet if we can make it by software because my PCB is produced, hahhaha.

L-KAYA, 2023-06-29:
It seems more like a mathematic problem to design a generic algorithm that covering all such cases. But it might be easier and more flexible if we provide an interface for users to implement their own algorithm that specified only for a particular pad order.

swortain, 2023-06-29:
OK, that's would be great offering an API to implement this feature.

closed 2023-06-29 by swortain