espressif/esp-idf #11761: #include "mbedtls/x509_crt.h" missing from esp_tls.h

Hairy-Lee, 2023-06-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Framework Version: esp-idf-v4.4.5

#include "mbedtls/x509_crt.h" is missing from esp_tls.h. 

mbedtls_x509_crt is used within esp_tls_t and without x509_crt.h mbedtls_x509_crt is an unknown type.

`/**
 * @brief      ESP-TLS Connection Handle
 */
typedef struct esp_tls {
#ifdef CONFIG_ESP_TLS_USING_MBEDTLS
    mbedtls_ssl_context ssl;                                                    /*!< TLS/SSL context */

    mbedtls_entropy_context entropy;                                            /*!< mbedTLS entropy context structure */

    mbedtls_ctr_drbg_context ctr_drbg;                                          /*!< mbedTLS ctr drbg context structure.
                                                                                     CTR_DRBG is deterministic random
                                                                                     bit generation based on AES-256 */

    mbedtls_ssl_config conf;                                                    /*!< TLS/SSL configuration to be shared
                                                                                     between mbedtls_ssl_context
                                                                                     structures */

    mbedtls_net_context server_fd;                                              /*!< mbedTLS wrapper type for sockets */

    **mbedtls_x509_crt** cacert;                                                    /*!< Container for the X.509 CA certificate */

    **mbedtls_x509_crt** *cacert_ptr;                                               /*!< Pointer to the cacert being used. */

    **mbedtls_x509_crt** clientcert;                                                /*!< Container for the X.509 client certificate */

    mbedtls_pk_context clientkey;                                               /*!< Container for the private key of the client
                                                                                     certificate */
#ifdef CONFIG_ESP_TLS_SERVER
    **mbedtls_x509_crt** servercert;                                                /*!< Container for the X.509 server certificate */

    mbedtls_pk_context serverkey;                                               /*!< Container for the private key of the server
                                                                                   certificate */
#endif
#elif CONFIG_ESP_TLS_USING_WOLFSSL
    void *priv_ctx;
    void *priv_ssl;
#endif
    int sockfd;                                                                 /*!< Underlying socket file descriptor. */

    ssize_t (*read)(struct esp_tls  *tls, char *data, size_t datalen);          /*!< Callback function for reading data from TLS/SSL
                                                                                     connection. */

    ssize_t (*write)(struct esp_tls *tls, const char *data, size_t datalen);    /*!< Callback function for writing data to TLS/SSL
                                                                                     connection. */

    esp_tls_conn_state_t  conn_state;                                           /*!< ESP-TLS Connection state */

    fd_set rset;                                                                /*!< read file descriptors */

    fd_set wset;                                                                /*!< write file descriptors */

    bool is_tls;                                                                /*!< indicates connection type (TLS or NON-TLS) */

    esp_tls_role_t role;                                                        /*!< esp-tls role
                                                                                     - ESP_TLS_CLIENT
                                                                                     - ESP_TLS_SERVER */

    esp_tls_error_handle_t error_handle;                                        /*!< handle to error descriptor */

} esp_tls_t;`

Alvin1Zhang, 2023-06-28:
Thanks for reporting.

mahavirj, 2023-06-29:
@Hairy-Lee `mbedtls/ssl.h` includes the `x509_crt.h` header file. Any specific build issue you are facing? If yes, could you please share detailed instructions to recreate it?

Hairy-Lee, 2023-06-29:
It wasnt detected during a build but when I linted my project (PC lint Plus 2.0):

esp_tls.h %userprofile%\.espressif\frameworks\esp-idf-v4.4.5\components\esp-tls
error 5915: unknown type name 'mbedtls_x509_crt' [Ln 309, Col5]
error 5915: unknown type name 'mbedtls_x509_crt' [Ln 311, Col5]
error 5915: unknown type name 'mbedtls_x509_crt' [Ln 313, Col5]
error 5915: unknown type name 'mbedtls_x509_crt' [Ln 632, Col1]

I agree that `mbedtls/ssl.h` includes the `x509_crt.h` header file, but only if `MBEDTLS_X509_CRT_PARSE_C` is defined. CRL and CRS are defined within sdkconfig but not CRT

CONFIG_MBEDTLS_X509_CRL_PARSE_C=y
CONFIG_MBEDTLS_X509_CSR_PARSE_C=y

closed 2023-07-06 by espressif-bot