espressif/esp-idf #11763: Unable to Enable PSRAM Clock Speed to 120MHz in ESP32-S3 N8R8

nikhil-robinson, 2023-06-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am currently working with an ESP32-S3 N8R8 module and trying to enable the PSRAM clock speed to 120MHz. However, I am facing difficulties in finding the option for setting this clock speed in the menuconfig tool.

Steps to Reproduce:

Use the ESP32-S3 N8R8 module.
Attempt to enable the PSRAM clock speed to 120MHz.
Open the menuconfig tool.
Search for the option to configure the PSRAM clock speed.
Unable to locate the option for setting the clock speed to 120MHz.
Expected Behavior:
I expected to find an option in the menuconfig tool to set the PSRAM clock speed to 120MHz for the ESP32-S3 N8R8 module.

Actual Behavior:
The menuconfig tool does not provide an option for configuring the PSRAM clock speed to 120MHz. The available options only allow for lower clock speeds.

I have checked the documentation and datasheet for the ESP32-S3 N8R8 module but couldn't find any specific information regarding enabling a clock speed of 120MHz for PSRAM.
Please let me know if there are any additional steps or information required to resolve this issue.

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/11893537/sdkconfig.txt)

Icarus113, 2023-06-28:
Which idf version are you using?

Since v5.1, we introduce the 120MHz DDR Octal PSRAM (ESP32-S3 N8R8 is using an Octal PSRAM) as an experimental feature, you can refer to this [doc](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/flash_psram_config.html#all-supported-modes-and-speeds) to know the risks and how to enable it.

Prior to v5.1, we don't support 120MHz DDR Octal PSRAM.

@nikhil-robinson

Icarus113, 2023-07-04:
@nikhil-robinson  Hi, may I know if there's any updates?

Icarus113, 2023-07-07:
This issue will be closed.

Feel free to reopen this, if there's any further questions. @nikhil-robinson

closed 2023-07-07 by Icarus113