espressif/esp-idf #11764: readdir returning "?" as file name

ysba, 2023-06-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello,

I need some help to debug an issue that I'm facing. I'm not quite sure if this is whether a bug or not.

### Application details

I have a hardware application that uses an ESP32-C3-MINI-1 module. This application has partition in the internal flash memory used to store a about dozen of files (2 configuration files, 4 log files, and up to 10 certificate files) - surely not more than that.

I have created a task which has the purpose of getting the total amount of bytes used by the log files, so I can estimate how much space is still available. The following snippet shows the main part of this task:

``` C
static void storage_update_free_space() {
    char path[64];
    FILE *fp = NULL;
    DIR *d;
    struct dirent *dir; 
    uint32_t total_bytes = 0;
    uint32_t file_len = 0;
    d = opendir(STORAGE_BASE_PATH);
    if(d) {
        ESP_LOGD(LOG_TAG_STORAGE, "%s, loop init", __func__);
        while((dir = readdir(d)) != NULL) {
            sprintf(path, "%s/%s", STORAGE_BASE_PATH, dir->d_name);
            fp = fopen(path, "rb");
            ESP_LOGD(LOG_TAG_STORAGE, "%s, d_name: %s / d_ino: %d", __func__, dir->d_name, dir->d_ino);
            if(fp == NULL) {
                ESP_LOGW(LOG_TAG_STORAGE, "%s, fopen error: %s", __func__, path);
            }
            else if(dir->d_name[0] == 'l' && dir->d_name[1] == 'o' && dir->d_name[2] == 'g') {  
                fseek(fp, 0, SEEK_END);
                file_len = ftell(fp);
                fclose(fp);
                ESP_LOGW(LOG_TAG_STORAGE, "%s, file: %s, %d bytes", __func__, dir->d_name, file_len);
                total_bytes += file_len;
            }
            else {
                ESP_LOGD(LOG_TAG_STORAGE, "%s, skipping, not a log file: %s", __func__, path);
                fclose(fp);
            }
        }

        ESP_LOGD(LOG_TAG_STORAGE, "%s, loop end", __func__);

        closedir(d); 

    /* calculate and print available space based on "total_bytes" */
}
```

### The problem
This application is in production already. The symptom here described happens only in some units. All of them run the very same firmware version of my application. The bunch of files are created during run time, so at this point only the one or two files may bre preset.

Here follows a section of the output log, begining in the "loop init" line:

```
D (578) storage: storage_update_free_space, loop init
D (579) wl_flash: read - src_addr= 0x00002000, size= 0x00001000
V (579) wl_flash: calcAddr - addr= 0x00002000 -> result= 0x00002000, dummy_addr= 0x00039000
V (581) vfs_fat: vfs_fat_open: path="/?", flags=0, mode=1b6
D (581) vfs_fat: vfs_fat_open: fresult=6
D (582) storage: storage_update_free_space, d_name: ? / d_ino: 0
W (582) storage: storage_update_free_space, fopen error: /spiflash/?
V (583) vfs_fat: vfs_fat_open: path="/?", flags=0, mode=1b6
D (584) vfs_fat: vfs_fat_open: fresult=6
D (584) storage: storage_update_free_space, d_name: ? / d_ino: 0
W (585) storage: storage_update_free_space, fopen error: /spiflash/?
V (586) vfs_fat: vfs_fat_open: path="/?", flags=0, mode=1b6
D (586) vfs_fat: vfs_fat_open: fresult=6
D (587) storage: storage_update_free_space, d_name: ? / d_ino: 0
W (587) storage: storage_update_free_space, fopen error: /spiflash/?
```

The log keeps on this sequence about 50 times, then one of my files is found (``config.jsn``):
```
V (768) vfs_fat: vfs_fat_open: path="/?", flags=0, mode=1b6
D (768) vfs_fat: vfs_fat_open: fresult=6
D (769) storage: storage_update_free_space, d_name: ? / d_ino: 0
W (769) storage: storage_update_free_space, fopen error: /spiflash/?
V (770) vfs_fat: vfs_fat_open: path="/config.jsn", flags=0, mode=1b6
D (772) storage: storage_update_free_space, d_name: config.jsn / d_ino: 0
D (772) storage: storage_update_free_space, skipping, not a log file: /spiflash/config.jsn
V (773) vfs_fat: vfs_fat_open: path="/?", flags=0, mode=1b6
D (774) vfs_fat: vfs_fat_open: fresult=6
D (774) storage: storage_update_free_space, d_name: ? / d_ino: 0
W (775) storage: storage_update_free_space, fopen error: /spiflash/?
```

The sequence is kept for more 50 or 60 times, another file is found (``log_type1.bin``), and then the loop ends:
```
V (921) vfs_fat: vfs_fat_open: path="/?", flags=0, mode=1b6
D (921) vfs_fat: vfs_fat_open: fresult=6
D (922) storage: storage_update_free_space, d_name: ? / d_ino: 0
W (923) storage: storage_update_free_space, fopen error: /spiflash/?
V (924) vfs_fat: vfs_fat_open: path="/log_type1.bin", flags=0, mode=1b6
D (925) storage: storage_update_free_space, d_name: log_type1.bin / d_ino: 0
D (926) vfs_fat: vfs_fat_lseek: offset=64, filesize:=64
D (926) wl_flash: read - src_addr= 0x0000e000, size= 0x00001000
V (927) wl_flash: calcAddr - addr= 0x0000e000 -> result= 0x0000e000, dummy_addr= 0x00039000
W (928) storage: storage_update_free_space, file: log_type1.bin, 64 bytes
D (929) wl_flash: read - src_addr= 0x00003000, size= 0x00001000
V (929) wl_flash: calcAddr - addr= 0x00003000 -> result= 0x00003000, dummy_addr= 0x00039000
D (931) storage: storage_update_free_space, loop end
```

Ok, so the thing is: I have only two valid files on my data partition (``config.jsn`` and ``log_type1.bin``), but the iteration using ``readdir`` gives about 120 entries with ``?`` as file name, which of course gives error in ``fopen``.

The verbose log line ``V (921) vfs_fat: vfs_fat_open: path="/?", flags=0, mode=1b6`` is the first log entry indicating ``?`` as filename, and the next line ``D (921) vfs_fat: vfs_fat_open: fresult=6`` indicates value 6 as return code, which means ``FR_INVALID_NAME`` (``FRESULT`` enum at ``components/fatfs/src/ff.h``).

### Questions
* How can I check what is going on in my data storage partition and figure out why ``readdir`` is giving this type of error?
* Why this problem would happen only in some units, and not every one?
* Is there some way to get more debug info beyod the verbose level? I have figured out that the ``vfs_fat`` and ``wl_flash`` tags print information when calling ``readdir``. Is there any other tag that could help in this situation?

### System information
* ESP32-C3-MINI-1 module
* Building with ESP-IDF ``v4.4.5`` for production, but the problem persists when updating the module with firmware built on ``v5.0.2`` (last stable at this time, as ``v5.1-rc2`` would not compile directly)
* Flash encryption is used
* Firmware is flashed in production using ``esptool``
  * First, the encryption key is flashed via ``espefuse``
  * Encrypted firmware files are flashed via ``esptool``
  * Fuses are burnt via ``espefuse`` (``SPI_BOOT_CRYPT_CNT``, ``DIS_DOWNLOAD_ICACHE``, ``DIS_USB_JTAG``, ``DIS_PAD_JTAG``, ``DIS_DIRECT_BOOT``, ``DIS_DOWNLOAD_MANUAL_ENCRYPT``, ``DIS_DOWNLOAD_MODE``)
* As download mode is disabled in production, I'm only able to update firmware using custom OTA bootloader via UART interface
* Partition table:
  ```
  # ESP-IDF Partition Table
  # Name,     Type,   SubType,    Offset,     Size,       Flags
  nvs,        data,   nvs,        0xA000,     24K,
  phy_init,   data,   phy,        ,           4K,
  otadata,    data,   ota,        ,           0x2000,
  ota_0,      app,    ota_0,      ,           0x180000,
  ota_1,      app,    ota_1,      ,           0x180000,
  storage,    data,   fat,        ,           0xC0000,    encrypted
  nvs_keys,   data,   nvs_keys,   ,           0x1000,     encrypted
  ```

Any help in this matter will be appreciated!
Cheers,
Yuri

adokitkat, 2023-07-02:
Hello, thank you for your report. If you are trying to just get an info about your FAT partition like total and free bytes, have you consider using [`esp_vfs_fat_info` function](https://github.com/espressif/esp-idf/blob/cf7e743a9b2e5fd2520be4ad047c8584188d54da/components/fatfs/vfs/esp_vfs_fat.h#L352) ([implementation](https://github.com/espressif/esp-idf/blob/cf7e743a9b2e5fd2520be4ad047c8584188d54da/components/fatfs/vfs/vfs_fat.c#L207))?

ysba, 2023-07-03:
Hi! I was not aware of the ``esp_vfs_fat_info`` function, maybe because I'm using v4.4 in production. I'll take a look, but as I could see it only returns the amount of bytes. Anyway, I'd like to stick with the ``readdir`` method because eventually I may need to debug which files are present in the fat as their amount and their names are created on the run.

adokitkat, 2023-07-13:
I would still encourage you to use `esp_vfs_fat_info`. As I said, it returns a number of total bytes in FATFS partition and a number of free bytes in the partition, so you can just subtract those to get a number of used bytes, which are values you care about, if I understand you correctly. Traversing FAT structures is not trivial and POSIX `readdir` is not meant to do so in my opinion. If your allocation unit (cluster) size is set to 16 kiB (on SD card) and your file is 100 B, it will still occupy 16 kiB in FATFS partition (this would be 4kiB for SPI flash).

Also if you keep your files open, i.e. don't call `fclose()` on them often, then you also should call `fsync(fd)` on them after a write operation and before any function which reports any information such as total and free bytes (e.g.  `esp_vfs_fat_info`) and keep `FF_FS_NOFSINFO` = 0 (default value) to get accurate information (no more orphan clusters).

If the content of your partition is not corrupted I think you can still iterate through filesystem using `readdir` to find files and just ignore `/?` entries. However I am not sure if your algorithm is completely correct.

Verbose level is the highest log level, however some components have separate log settings such as e.g. `VFS_SUPPRESS_SELECT_DEBUG_OUTPUT` used in `vfs.c` file => `Select() related functions might produce an unconveniently lot of debug outputs when one sets the default log level to DEBUG or higher. It is possible to suppress these debug outputs by enabling this option.` which is enabled by default in IDF and so even if your log level is verbose, ESP_LOG fucntions in `vfs.c` print nothing, so please be ware of this. You may try to disable this setting in menuconfig (`(Top) → Component config → Virtual file system`). I am not aware of something like this being in another files/components though.

adokitkat, 2023-07-27:
Hello. Do you have any update on this please? Or should this issue be closed?

ysba, 2023-07-31:
Hi! I'll follow your suggestion and use the ``esp_vfs_fat_info`` function. The reason for this issue is just to report the ``?`` as filename using ``readdir`` method. As far as I know, ``readdir`` uses ``.a`` libraries in the ``esp-idf`` framework, so I am not able to track down the source code to help and find the problem.

I just wrote the code in my application to use the ``esp_vfs_fat_info`` function. Next step is to see how it behaves on the specific units that have shown the ``?``.

adokitkat, 2023-08-07:
This is the source code for C stdlib `newlib` we are using: https://github.com/espressif/newlib-esp32

I will close this issue for now and if you have any problem or question regarding this issue, feel free to reopen it. If you were to have another problem or question, please open a new issue. Thank you.

closed 2023-08-07 by adokitkat