espressif/esp-idf #11765: 修改蓝牙名称后, ios搜索还是旧名称

Leadrive, 2023-06-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

4.4

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3

### Power Supply used.

USB

### What is the expected behavior?

ble name修改后，希望ios能搜索出新名称

### What is the actual behavior?

Android上正常，ios总是搜索出旧名称

### Steps to reproduce.

    esp_ble_gap_set_device_name(name);// BLUFI_DEVICE_NAME
    esp_ble_gap_config_adv_data(&blufi_adv_data);
搜索了几个bug，都没找到解决的办法。
都说是ios缓存的问题，请问如何解决？

zhp0406, 2023-06-29:
你可以更加具体的描述一下吗?
我使用 ios 下 EspBluFi 测试，下滑屏幕刷新后显示出来了更改后的新名字。

zhp0406, 2023-09-13:
@Leadrive 
请问你的问题是否已经解决?

Leadrive, 2023-09-23:
@zhp0406 没解决。

zhp0406, 2023-09-23:
@Leadrive 
你有没有下划屏幕刷新一下试试

closed 2023-12-07 by Indastri