espressif/esp-idf #11766: 修改蓝牙名称后, ios搜索还是旧名称

Leadrive, 2023-06-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

4.4

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3

### Power Supply used.

USB

### What is the expected behavior?

ble name修改后，希望ios能搜索出新名称

### What is the actual behavior?

Android上正常，ios总是搜索出旧名称

### Steps to reproduce.

    esp_ble_gap_set_device_name(name);// BLUFI_DEVICE_NAME
    esp_ble_gap_config_adv_data(&blufi_adv_data);
搜索了几个bug，都没找到解决的办法。
都说是ios缓存的问题，请问如何解决？

zhp0406, 2023-08-15:
@Leadrive 
你是使用的下面这个 example 吗？
https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/blufi

如果是这个 example，需要你下拉刷新才能显示出来新的名字。

Indastri, 2024-01-04:
Duplicated issue #11765. Closing.

closed 2024-01-04 by Indastri