espressif/esp-idf #11767: Flashing error

uzunab07, 2023-06-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

 1.79.2 (Universal)

### Operating System used.

macOS

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32 WROOM 32D

### Power Supply used.

USB

### What is the expected behavior?

It should be able to pass the flashing part 

### What is the actual behavior?

I tired flashing the esp but i am getting this error

### Steps to reproduce.

1.Build 
2.Flash

### Debug Logs.

```plain
A fatal error occurred: Unable to verify flash chip connection (Serial data stream stopped: Possible serial noise or corruption.).
```

radimkarnis, 2023-06-29:
Hello @uzunab07,
there seems to be an issue with the flash chip on your devkit. Esptool fails when it interrogates it for the first time.

- Is there anything connected to your devkit? If there is, please disconnect everything.
- Do you have other hardware to test with? Does the issue persist with other devkits?
- Can you try flashing without the stub flasher? (Menuconfig -> Serial flasher config -> Disable download stub)

radimkarnis, 2023-07-17:
Closing this for inactivity. Feel free to reopen if needed.

closed 2023-07-17 by radimkarnis