espressif/esp-idf #11768: RMT: Cannot transmit and receive on same GPIO pin when `io_loop_back=true`

kevinhikaruevans, 2023-06-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.0.2

### Operating System used.

Linux

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32-s3-devkitc-1

### Power Supply used.

USB

### What is the expected behavior?

When using RMT with RX/TX channels using the same GPIO with `.flags.io_loop_back=true`, I expect to be able to transmit and receive on both channels.

### What is the actual behavior?

When using an rx channel, the GPIO pin is pulled always low. Specifically, calling `rmt_new_rx_channel(...)` pulls the GPIO pin low and I cannot receive or transmit on that pin.

When I disable the rx channel, I can transmit just fine.

### Steps to reproduce.

1. Setup RMT with tx and rx channels
2. Observe the line is pulled low

I basically followed the NEC RMT example, but tweaked a few things, so I can transmit 10-bit serial. Feel free to comment if I'm doing something wrong or missing it.

```c
#include "freertos/FreeRTOS.h"
#include "freertos/task.h" 
#include "freertos/queue.h"
#include "driver/rmt_tx.h"
#include "driver/rmt_rx.h"
#include "driver/rmt_encoder.h"

#include "esp_check.h"
#include "esp_log.h"
#include <stdint.h>

#define SERIAL_FREQ_HZ 1000000
#define SERIAL_GPIO_NUM 6

static const char TAG[] = "main";

const static rmt_symbol_word_t bits[2] = {
  // inverted!

  /* 0 */ {
    .level0 = 1,
    .duration0 = 5,
[... 253 lines cut ...]
␛[0;32mI (200) cpu_start: ELF file SHA256:  f9aa97c14fbafdc2...␛[0m
␛[0;32mI (206) cpu_start: ESP-IDF:          5.0.1␛[0m
␛[0;32mI (211) cpu_start: Min chip rev:     v0.0␛[0m
␛[0;32mI (216) cpu_start: Max chip rev:     v0.99 ␛[0m
␛[0;32mI (220) cpu_start: Chip rev:         v0.1␛[0m
␛[0;32mI (225) heap_init: Initializing. RAM available for dynamic allocation:␛[0m
␛[0;32mI (232) heap_init: At 3FC94E40 len 000548D0 (338 KiB): D/IRAM␛[0m
␛[0;32mI (239) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM␛[0m
␛[0;32mI (245) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM␛[0m
␛[0;32mI (252) heap_init: At 600FE010 len 00001FF0 (7 KiB): RTCRAM␛[0m
␛[0;32mI (259) spi_flash: detected chip: generic␛[0m
␛[0;32mI (263) spi_flash: flash io: dio␛[0m
␛[0;32mI (267) cpu_start: Starting scheduler on PRO CPU.␛[0m
␛[0;32mI (0) cpu_start: Starting scheduler on APP CPU.␛[0m
␛[0;32mI (287) main: create rmt tx channel␛[0m
␛[0;32mI (287) gpio: GPIO[6]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 ␛[0m
␛[0;32mI (297) gpio: GPIO[6]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 ␛[0m
␛[0;32mI (1307) main: transmitting␛[0m
␛[0;32mI (3307) main: transmitting␛[0m
␛[0;32mI (5307) main: transmitting␛[0m
␛[0;32mI (7307) main: transmitting␛[0m
␛[0;32mI (9307) main: transmitting␛[0m
␛[0;32mI (11307) main: transmitting␛[0m
```
```

kevinhikaruevans, 2023-06-28:
Interestingly, I can set the rx pin to a new pin, then jump the rx/tx together and it works as-expected.

suda-morris, 2023-06-29:
TLDR, you may need to install the RX channel first, then the TX channel. Please check the test code we used: https://github.com/espressif/esp-idf/blob/master/components/driver/test_apps/rmt/main/test_rmt_rx.c#L43

kevinhikaruevans, 2023-06-29:
> TLDR, you may need to install the RX channel first, then the TX channel. Please check the test code we used: https://github.com/espressif/esp-idf/blob/master/components/driver/test_apps/rmt/main/test_rmt_rx.c#L43

That worked. I called `rmt_new_rx_channel` before `rmt_new_tx_channel` to get it working correctly. 

It would be nice to have a [note on the RMT page](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/rmt.html#resource-allocation) under [`rmt_tx_channel_config_t::io_loop_back`](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/rmt.html#_CPPv4N23rmt_tx_channel_config_t12io_loop_backE) stating the rx channel must be initialized before tx.

suda-morris, 2023-06-30:
Because [this code](https://github.com/espressif/esp-idf/blob/master/components/driver/gpio/gpio.c#L216) will change the gpio matrix signal connection. If the TX was set first, then during the RX set up, the previous RMT channel signal will be overridden by the GPIO control signal.

closed 2023-07-06 by espressif-bot