espressif/esp-idf #11771: system assert when I stop ble task

luacerbb, 2023-06-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.4

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-DevKitC-V4

### Power Supply used.

USB

### What is the expected behavior?

stop ble task success

### What is the actual behavior?

sometimes assert when stop ble task.

### Steps to reproduce.

stop ble function:
static void _bd_ble_stop(void)
{
	log_debug(TAG,"ble stop");
	int32_t  ret = ble_gap_adv_stop();
	if (ret) 
	{
		log_debug(TAG, "Error in stopping advertisement with err code = %d",ret);
	}
	
//	if((ret = ble_gap_terminate(g_conn_handle, BLE_ERR_REM_USER_CONN_TERM)) != 0)
//	{
//		log_err(TAG,"ble terminate fail:%d",ret);
//	}

	//关闭nimble任务运行
	ret = nimble_port_stop();
	if (ret == 0) 
	{
		HAL_SleepMs(50);
		nimble_port_deinit();
		ret = esp_nimble_hci_and_controller_deinit();
		if (ret != ESP_OK) 
		{
			log_err(TAG, "esp_nimble_hci_and_controller_deinit() failed with error: %d", ret);
		}
		//释放RAM空间，减少很多	
		esp_bt_mem_release(ESP_BT_MODE_BTDM);
	}
	else
	{
		log_err(TAG,"stop ble err: %d",ret);
	}
}

### Debug Logs.

```plain
[dbg] 29:09:15:26 _bd_ble_stop(1331): ble stop
[dbg] 29:09:15:26 _bd_ble_stop(1335): Error in stopping advertisement with err code = 2
E (24673) NimBLE: ble_hs_stop_terminate_timeout_cb,1 connection(s) still up 

[inf] 29:09:15:26 bleprph_host_task(1011): BLE Host Task Stoped

assert failed: host_rcv_pkt esp_nimble_hci.c:378 (rc == 0)

Backtrace: 0x40081d7e:0x3ffe5290 0x4009202d:0x3ffe52b0 0x4009859d:0x3ffe52d0 0x40115e62:0x3ffe53f0 0x4012a836:0x3ffe5420 0x40119d36:0x3ffe5440 0x4001791d:0x3ffe5460 0x4011a5b5:0x3ffe5480 0x4011a692:0x3ffe54c0 0x400188f5:0x3ffe54f0 0x40119f77:0x3ffe5530 0x40044a63:0x3ffe5550 0x40045983:0x3ffe5570 0x40084db0:0x3ffe5590 0x40019d11:0x3ffe55c0 0x40055b4d:0x3ffe55e0 0x401184b7:0x3ffe5600 0x40118b15:0x3ffe5620
0x40081d7e: panic_abort at /mnt/hgfs/SUB_PROJECT/esp-idf-v4.4.4/components/esp_system/panic.c:408

0x4009202d: esp_system_abort at /mnt/hgfs/SUB_PROJECT/esp-idf-v4.4.4/components/esp_system/esp_system.c:137

0x4009859d: __assert_func at /mnt/hgfs/SUB_PROJECT/esp-idf-v4.4.4/components/newlib/assert.c:85

0x40115e62: host_rcv_pkt at /mnt/hgfs/SUB_PROJECT/esp-idf-v4.4.4/components/bt/host/nimble/esp-hci/src/esp_nimble_hci.c:372
 (inlined by) host_rcv_pkt at /mnt/hgfs/SUB_PROJECT/esp-idf-v4.4.4/components/bt/host/nimble/esp-hci/src/esp_nimble_hci.c:341

0x4012a836: vhci_send at ??:?

0x40119d36: r_eif_send at ??:?

0x4011a5b5: hci_tx_start at hci_tl.c:?

0x4011a692: r_hci_tl_send at ??:?

0x40119f77: r_hci_send_2_host_hack at ??:?

0x40084db0: ke_task_schedule at ke_task.c:?

0x401184b7: r_rw_schedule at ??:?

0x40118b15: btdm_controller_task at ??:?
```

RoshanESP, 2023-06-30:
Hi @luacerbb , 
Can you try the attached patch and check if this fixes the issue? 
[assert_fix.txt](https://github.com/espressif/esp-idf/files/11913398/assert_fix.txt)

luacerbb, 2023-07-05:
> Hi @luacerbb , Can you try the attached patch and check if this fixes the issue? [assert_fix.txt](https://github.com/espressif/esp-idf/files/11913398/assert_fix.txt)

thanks,I have try this patch,it don't occur assert again

closed 2023-07-05 by luacerbb