espressif/esp-idf #11774: ble multi advertising transmits on all channels.

jonathanMerlingPal, 2023-06-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

4.4.5

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

esp32-s3 custom board

### Power Supply used.

External 3.3V

### What is the expected behavior?

Ble advertising is supposed to transmit only on the three channels that are used for advertising.

### What is the actual behavior?

I ran bluetooth/bluedroid/ble_50/multi-adv example as is and connected it to Signal Hound BB60C Real-Time Spectrum Analyzer . with no one attempting to connect i got:
[image]
which is not up to Bluetooth standard.

for comparison I ran bluetooth/bluedroid/ble/gatt_server example as is on the same hardware and got:
[image]
which is up to standard.

### Steps to reproduce.

1-load example bluetooth/bluedroid/ble_50/multi-adv.
2-flash to chip.
3-connect spectrum analyzer.
4-run example and record Bluetooth signal on the spectrum analyzer.

### More Information.

i tried the same with esp-idf 5.0.2 and got the same result

zhp0406, 2023-07-04:
[image]

In BLE 5.0, the ADV_EXT_IND and AUX_ADV_IND are utilized. The ADV_EXT_IND is restricted to channels 37, 38, and 39 only, while the AUX_ADV_IND can be employed across all channels.

For BLE 4.2, it employs ADV_IND, which is limited to channel 37, 38, and 39 exclusively.

jonathanMerlingPal, 2023-07-05:
thanks for the info.

closed 2023-07-05 by jonathanMerlingPal