espressif/esp-idf #11776: I2C master changes clock after slave "NACK"

screwdriver48, 2023-06-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.0.2

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32-C3

### Power Supply used.

External 3.3V

### What is the expected behavior?

I setup an I2C master which communicates with two slaves. One of the two slaves is sometimes disconnected hence the response is a "NACK".
The clock speed of the I2C peripheral should never change without reconfiguration.
I2C_CLOCK: 20000Hz
SCL_GPIO: GPIO_NUM_18
SDA_GPIO: GPIO_NUM_19

### What is the actual behavior?

The I2C clock changes to 40000Hz on itself after 8 nacks. The I2C peripheral is initialized once and regular write/read operations are executed. 
The data received is always correct and I2C is fully functional, but the clock change is inexplicable.

The image shows the clock changing during runtime captured with a logic analyzer.
[image]

### Steps to reproduce.

1. Step: Setup an I2C master addressing two slaves. Slave 1 has to response, slave 2 has to nack the requests.
2. Step: Monitor the clock speed.
3. Step: After around 8 nacks from slave 2, the clock speed is changed and never recovers.

screwdriver48, 2023-07-03:
Update:
The issue appears after 10 "nack" responses from slave. This triggers the function "i2c_hw_fsm_reset".
Apperantly, the clock is not set properly in this function. Each time the function i2c_hw_fsm_reset is called, the i2c clock is changed to a value higher then initially set.

Workaround:
Save the initial i2c config locally in "i2c.h".
`esp_err_t i2c_param_config(i2c_port_t i2c_num, const i2c_config_t *i2c_conf)
{
    i2c_clock_source_t src_clk = I2C_CLK_SRC_DEFAULT;
    esp_err_t ret = ESP_OK;
    i2c_masterConfig = *i2c_conf;
`
In i2c_hw_fsm_reset, add following lines:
` i2c_clock_source_t src_clk = s_get_clk_src(i2c_masterConfig.clk_flags, i2c_masterConfig.master.clk_speed);
  i2c_hal_set_bus_timing(&(i2c_context[i2c_num].hal), i2c_masterConfig.master.clk_speed, src_clk, s_get_src_clk_freq(src_clk));`

Like this, the clock is not changed when i2c_hw_fsm_reset is called, but it is not a proper fix.

AxelLin, 2023-08-20:
@o-marshmallow 
Any comment about https://github.com/espressif/esp-idf/issues/11776#issuecomment-1617620249 ?

txinto, 2023-08-29:
Hi, I have a very similar issue.  After several times trying to access to a non-existent peripheral, one of my other (existing) devices begins issuing ACK errors.  I can recover the communication by executing this function:

```
void i2c_recover(void)
{
    i2c_param_config(i2c_master_port, &i2c_conf);
}
```

I've been doing a more specific recovery by reducing the contents of the i2c_param_config() to just re-config the clock source, and it worked.

This is the function added to ESP-IDF's i2c.c:

```
esp_err_t i2c_clk_source_config(i2c_port_t i2c_num, const i2c_config_t *i2c_conf)
{
    i2c_clock_source_t src_clk = I2C_CLK_SRC_DEFAULT;
    esp_err_t ret = ESP_OK;

    ESP_RETURN_ON_FALSE(i2c_conf != NULL, ESP_ERR_INVALID_ARG, I2C_TAG, I2C_ADDR_ERROR_STR);
    ESP_RETURN_ON_FALSE(i2c_conf->mode < I2C_MODE_MAX, ESP_ERR_INVALID_ARG, I2C_TAG, I2C_MODE_ERR_STR);

    if (i2c_conf->mode == I2C_MODE_MASTER) {
        src_clk = s_get_clk_src(i2c_conf->clk_flags, i2c_conf->master.clk_speed);
    }

    ESP_RETURN_ON_FALSE(src_clk != I2C_CLOCK_INVALID, ESP_ERR_INVALID_ARG, I2C_TAG, I2C_CLK_FLAG_ERR_STR);
    i2c_hal_set_bus_timing(&(i2c_context[i2c_num].hal), i2c_conf->master.clk_speed, src_clk, s_get_src_clk_freq(src_clk));

    return ESP_OK;
}

```

And this is the new recovery function:

```
void i2c_recover(void)
{
    i2c_clk_source_config(i2c_master_port, &i2c_conf);
}

```

I am using the ESP32C3, and the relevant peripherals are:

- The absent peripheral is the LM73.
- The present peripheral whose communication is lost, is the PCF8576.

If I don't try to access the LM73, the communication with the PCF8576 is not lost.
If I try to access the LM73 several times, then the comm with the PCF8576 is lost, error is ACK is not received.
If then I execute i2c_recover(), the communication with the PCF8576 is recovered.

This situation is repeating every 15 seconds approx, I am asking the LM73 approx 1 time every 1.2 seconds.

I've tried several ESP-IDF versions:

- With 4.4.1 I did not see the issue.
- With 4.4.5 the issue appeared.
- With 5.1.1 the issue is still present.

I've tested two chips:

- With ESP32C3, the issue exists (within the ESP-IDF versions shown above).
- With ESP32C6 I have not seen the issue.

It looks like some change done in the HAL of the ESP32C3 in 4.4.2 ... 4.4.5 created the issue, or activated it.  Maybe changes done in the I2C common part could be activated it too.

Hope this information helps.

AxelLin, 2023-10-20:
> Update: The issue appears after 10 "nack" responses from slave. This triggers the function "i2c_hw_fsm_reset". 

Now I observed this symptom on esp32c6 with v5.1.1-588-gc1c843f5e2.

AxelLin, 2024-03-17:
@o-marshmallow Any update regarding this issue?

mythbuster5, 2025-01-13:
The issue can be solved by this commit c0e5f2b73a65acc87d24d5023f968f6b24617d61.

closed 2025-01-13 by mythbuster5

AxelLin, 2025-01-13:
> The issue can be solved by this commit [c0e5f2b](https://github.com/espressif/esp-idf/commit/c0e5f2b73a65acc87d24d5023f968f6b24617d61).

I don't get it, above commit is only available on IDF-5.3+.
But this issue was reported on v5.0.2.
So how is the status of the fix for 5.1, 5.2 branches?