espressif/esp-idf #11778: Support Chunked Encoding in httpd_req_recv

xjzi, 2023-06-29:
### Is your feature request related to a problem?

I need to have a full duplex HTTP communication channel between the ESP httpd server and javascript on a website. One solution is to have both the transfer and receive use the Chunked Encoding, but the ESP HTTP server does not support receiving in chunks.

### Describe the solution you'd like.

I would like httpd_req_recv to implement Chunked Encoding.

### Describe alternatives you've considered.

- I tried having the website send its payload normally and receive concurrent messages with chunked encoding, but the fetch API blocks until the whole payload has been transferred. 
- I tried having the website send its payload with one HTTP request and receive concurrent messages on another one, but the ESP HTTP server doesn't support concurrent http requests.
- I know that websockets would work but plain HTTP would be more convenient.

ESP-YJM, 2023-10-18:
@xjzi Yes, we not support receive chunk data in esp_http_server component now. We will evaluate this function request internal first.

AxelLin, 2024-08-06:
> We will evaluate this function request internal first.

@ESP-YJM 
The status switched back to Opened after in "Selected for Development" for 10 Months.
How is the progress now?

ESP-YJM, 2024-08-06:
@mahavirj  Could you please help update the status?

Vanmaele, 2025-06-13:
I was wondering if there are any updates on this feature or plans to support Transfer-Encoding: chunked for incoming HTTP request bodies in esp_http_server? I'd like to express my interest in this feature as it would improve compatibility with modern web APIs and tooling.