espressif/esp-idf #11779: static const in a c++ Class goes to .bss instead of flash

diplfranzhoepfinger, 2023-06-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-rc2

### Operating System used.

Linux

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

i will provide a minimal code Snippet for it. 

static const should go to flash not .bss

### What is the actual behavior?

it goes to .bss

### Steps to reproduce.

1. Step
2. Step
3. Step
...

diplfranzhoepfinger, 2023-06-30:
i have a Idea: 

it might come from the Build-Script: 

[https://gitlab.com/meisterschulen-am-ostbahnhof-munchen/org.eclipse.4diac.forte/-/blob/develop_me3/setup_esp32_isobus_DualOut.sh](https://gitlab.com/meisterschulen-am-ostbahnhof-munchen/org.eclipse.4diac.forte/-/blob/develop_me3/setup_esp32_isobus_DualOut.sh)

Line 82..91

diplfranzhoepfinger, 2023-06-30:
@igrr 
@azoitl
i am Pretty sure the way we build it is the Reason. 

so let me explain you: 

1. Step we call "setup_esp32_isobus_DualOut.sh" which produces a libforte-static.a
2. Step this libforte-static.a is linked to a normal esp-idf project: https://gitlab.com/meisterschulen-am-ostbahnhof-munchen/4diac-forte-esp32-component/-/blob/feature/no_lua/CMakeLists.txt

diplfranzhoepfinger, 2023-06-30:
@igrr
@azoitl

i have made same as a Lib and App Architecture: 

it still works right. 
https://github.com/diplfranzhoepfinger/testSegments/tree/as_lib

so it must be some specifics of not using idf.py for the lib build, some Command-line Params etc.. ?

diplfranzhoepfinger, 2023-06-30:
one more Information: 

i have a "older" Project with the same Build Schema, but not using esp-idf v5.1-rc2 but 5.0.2, 
and the issue is the same. 

as well i have a "fully open Source" Project, 
here: https://gitlab.com/meisterschulen-am-ostbahnhof-munchen/esp32-4diac-example-application

and here: 
https://gitlab.com/meisterschulen-am-ostbahnhof-munchen/esp32-4diac-example-application-no-lua

where the same occurs.

igrr, 2023-06-30:
Hi @diplfranzhoepfinger, we can accept this as a bug report if you provide a simple example (similar to your testSegments project) which demonstrates the issue. Troubleshooting a complex third party project is not the type of support we can offer in this issue tracker.

franz-hoepfinger-at-hm-edu, 2023-07-01:
Hi @igrr this is very clear. I was more expecting a Simple hint from you what I could try to solve this issue. 
I have 2 big Projects now. esp-idf and eclipse 4diac... Both say check your settings...

For example: what is the Compiler Command Line esp-idf is using ?
Then I could test this.

franz-hoepfinger-at-hm-edu, 2023-07-01:
Other way round: is there a paid support who could debug the complex Project? Like buy a Ticket (CoDeSys offers such things)

igrr, 2023-07-01:
> what is the Compiler Command Line esp-idf is using ?

You can find the commands esp-idf is using in build/compile_commands.json.

As for the troubleshooting suggestion, here's one I can think of:

Try to find the object file corresponding to the .cpp file where this variable is defined. The object files are in the build directory. Run objdump -t on this file and look for the section into which the variable is placed. Is it in rodata?

diplfranzhoepfinger, 2023-07-01:
> Try to find the object file corresponding to the .cpp file where this variable is defined. The object files are in the build directory. Run objdump -t on this file and look for the section into which the variable is placed. Is it in rodata?

very cool hint thanks. 

i learn more and more about Compiler building. I had "Compilerbau" in School, but this is far away from practice.

diplfranzhoepfinger, 2023-07-01:
[https://gist.github.com/diplfranzhoepfinger/3ee2a01620de1febbb1d2f2648030d60](https://gist.github.com/diplfranzhoepfinger/3ee2a01620de1febbb1d2f2648030d60)

DualHysteresys.cpp --> source Code

DualHysteresys.cpp.obj.txt --> output of objdump -t DualHysteresys.cpp.obj

diplfranzhoepfinger, 2023-07-01:
so, @igrr as you expected: the Error is already in the very first stage of Compilation. not later.

diplfranzhoepfinger, 2023-07-01:
so, found it: 

the Constants in my Sample are in the cpp File. 

so the Compiler can see them. 

in the 4diac Project they are in the extra cpp File, not in a header. 

so the Compiler can only reserve space and not replace the Constants with the Values.

diplfranzhoepfinger, 2023-07-01:
i have reported this to eclipse 4diac.

if i experimentally copy in the Values, 

it moves the Data to the rodata segment instead of bss

[image]

[image]

diplfranzhoepfinger, 2023-07-01:
solved: 

there is a Option: 

    -DFORTE\_LINKED\_STRINGDICT=OFF \\

if you set this, the Segment goes to rodata instead of bss. 

@igrr thanks for your help. 

as always Espressif Support outperforms even big players!

closed 2023-07-01 by diplfranzhoepfinger