espressif/esp-idf #11780: How to compile idf.py using a single thread?

lbuque, 2023-06-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.4

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

When I compile micropython, I often encounter compilation errors; when using esp-idf v4.0.2, I can make the following modifications, and then no compilation errors will appear.

```shell
$ git diff
diff --git a/tools/idf.py b/tools/idf.py
index 430f1ce0ba..31e9d111d0 100755
--- a/tools/idf.py
+++ b/tools/idf.py
@@ -74,7 +74,7 @@ GENERATORS = [
     ("Ninja", ["ninja"], ["ninja", "--version"], "-v"),
     (
         MAKE_GENERATOR,
-        [MAKE_CMD, "-j", str(multiprocessing.cpu_count() + 2)],
+        [MAKE_CMD, "-j", "1"],
         [MAKE_CMD, "--version"],
         "VERBOSE=1",
     ),
```

**Now I can't find where to modify it.**

### What is the actual behavior?

None

### Steps to reproduce.

1. Step
2. Step
3. Step
...

dobairoland, 2023-06-30:
https://github.com/espressif/esp-idf/blob/release/v4.4/tools/idf_py_actions/constants.py#L32

lbuque, 2023-06-30:
@dobairoland thanks

closed 2023-06-30 by lbuque