espressif/esp-idf #11782: How can I print a taskHandle_t 's current stack

0xFEEDC0DE64, 2023-06-30:
I am trying to make debugging a watchdog timeout easier by printing the failed task's current stack.

A typical watchdog timeout log output looks like this:

```
E (116178) task_wdt task_wdt.c:519 task_wdt_isr(): Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (116178) task_wdt task_wdt.c:553 task_wdt_isr():  - main (CPU 0)                                       
E (116178) task_wdt task_wdt.c:567 task_wdt_isr(): Tasks currently running:                              
E (116178) task_wdt task_wdt.c:569 task_wdt_isr(): CPU 0: IDLE                                           
E (116178) task_wdt task_wdt.c:569 task_wdt_isr(): CPU 1: IDLE                                           
E (116178) task_wdt task_wdt.c:406 task_wdt_timeout_abort_xtensa(): Aborting.                            
E (116178) task_wdt task_wdt.c:424 task_wdt_timeout_abort_xtensa(): Print CPU 0 (current core) backtrace 
                                                                                                         
                                                                                                         
                                                                                                         
                                                                                                         
Backtrace: 0x400882bf:0x3ffc6900 0x400d4bf2:0x3ffc6920 0x4008aa73:0x3ffc6940 0x4008c62d:0x3ffc6960       
0x400882bf: xt_utils_wait_for_intr at /home/feedc0de/goecharger-esp32-firmware/esp-idf/components/xtensa/include/xt_utils.h:81
 (inlined by) esp_cpu_wait_for_intr at /home/feedc0de/goecharger-esp32-firmware/esp-idf/components/esp_hw_support/cpu.c:111
                                                                                                         
0x400d4bf2: esp_vApplicationIdleHook at /home/feedc0de/goecharger-esp32-firmware/esp-idf/components/esp_system/freertos_hooks.c:59
                                                                                                         
0x4008aa73: prvIdleTask at /home/feedc0de/goecharger-esp32-firmware/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:4327
                                                                                                         
0x4008c62d: vPortTaskWrapper at /home/feedc0de/goecharger-esp32-firmware/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162
```

Now I want to know what task "main" was doing at that moment.

I found the corresponding esp-idf code here: https://github.com/espressif/esp-idf/blob/master/components/esp_system/task_wdt/task_wdt.c#L553

In that code we have a handle to the failing task. How can I print the stack of that task handle?

I would like to have a similar output like `esp_backtrace_print()` already provides, just for that failing task.

So idf.py is able to parse the hex adresses and prints the method and linenumbers.

chipweinberger, 2023-10-03:
@igrr appreciate you linking that! Would be useful as a first class feature in ESP-IDF.

ESP-Marius, 2024-02-06:
I think we can consider this one closed by 3686689a2a85da74. If you think the use-case you need it for is still not supported then feel free to reopen the issue.

closed 2024-02-06 by ESP-Marius