espressif/esp-idf #11786: Issue with running idf.py flash and idf.py monitor on ESP32-S2FN4R2

loco-choco, 2023-07-01:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.2

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

For flash to finish the flashing process and for monitor to be able to monitor the esp output.

### What is the actual behavior?

Both fail with the error OSError: [Errno 71] Protocol error, `idf.py monitor` in
```shell
Exception in thread Thread-2 (_run_outer):
Traceback (most recent call last):
  File "/nix/store/1r6n7v2wam7gkr18gxccpg7p5ywgw551-python3-3.10.12/lib/python3.10/threading.py", line 1016, in _bootstrap_inner
    self.run()
  File "/nix/store/1r6n7v2wam7gkr18gxccpg7p5ywgw551-python3-3.10.12/lib/python3.10/threading.py", line 953, in run
    self._target(*self._args, **self._kwargs)
  File "/nix/store/10am01kfgmsidrw1jfnwbpqlvwhmsqn2-esp-idf-v5.0.2/tools/idf_monitor_base/stoppable_thread.py", line 47, in _run_outer
    self.run()
  File "/nix/store/10am01kfgmsidrw1jfnwbpqlvwhmsqn2-esp-idf-v5.0.2/tools/idf_monitor_base/serial_reader.py", line 55, in run
    self.serial.rts = low      # Set rts/dtr to the reset state
  File "/nix/store/7zgswij5795q5k9wkl33fk638b98pzbl-python3-3.10.12-env/lib/python3.10/site-packages/serial/serialutil.py", line 463, in rts
    self._update_rts_state()
  File "/nix/store/7zgswij5795q5k9wkl33fk638b98pzbl-python3-3.10.12-env/lib/python3.10/site-packages/serial/serialposix.py", line 706, in _update_rts_state
    fcntl.ioctl(self.fd, TIOCMBIS, TIOCM_RTS_str)
OSError: [Errno 71] Protocol error
```

and for `idf.py flash` in
```shell
Traceback (most recent call last):
  File "/nix/store/1r6n7v2wam7gkr18gxccpg7p5ywgw551-python3-3.10.12/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/nix/store/1r6n7v2wam7gkr18gxccpg7p5ywgw551-python3-3.10.12/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/nix/store/7zgswij5795q5k9wkl33fk638b98pzbl-python3-3.10.12-env/lib/python3.10/site-packages/esptool/__main__.py", line 9, in <module>
    esptool._main()
  File "/nix/store/7zgswij5795q5k9wkl33fk638b98pzbl-python3-3.10.12-env/lib/python3.10/site-packages/esptool/__init__.py", line 1032, in _main
    main()
  File "/nix/store/7zgswij5795q5k9wkl33fk638b98pzbl-python3-3.10.12-env/lib/python3.10/site-packages/esptool/__init__.py", line 674, in main
    esp = esp or get_default_connected_device(
  File "/nix/store/7zgswij5795q5k9wkl33fk638b98pzbl-python3-3.10.12-env/lib/python3.10/site-packages/esptool/__init__.py", line 921, in get_default_connected_device
    _esp.connect(before, connect_attempts)
  File "/nix/store/7zgswij5795q5k9wkl33fk638b98pzbl-python3-3.10.12-env/lib/python3.10/site-packages/esptool/loader.py", line 640, in connect
    last_error = self._connect_attempt(reset_strategy, mode)
  File "/nix/store/7zgswij5795q5k9wkl33fk638b98pzbl-python3-3.10.12-env/lib/python3.10/site-packages/esptool/loader.py", line 525, in _connect_attempt
    reset_strategy()  # Reset the chip to bootloader (download mode)
  File "/nix/store/7zgswij5795q5k9wkl33fk638b98pzbl-python3-3.10.12-env/lib/python3.10/site-packages/esptool/reset.py", line 85, in __call__
    self._setDTRandRTS(False, True)  # IO0=HIGH & EN=LOW, chip in reset
  File "/nix/store/7zgswij5795q5k9wkl33fk638b98pzbl-python3-3.10.12-env/lib/python3.10/site-packages/esptool/reset.py", line 58, in _setDTRandRTS
    fcntl.ioctl(self.port.fileno(), TIOCMSET, struct.pack("I", status))
OSError: [Errno 71] Protocol error
Connecting...
CMake Error at run_serial_tool.cmake:55 (message):
  
  /nix/store/7zgswij5795q5k9wkl33fk638b98pzbl-python3-3.10.12-env/bin/python3.10;;/nix/store/10am01kfgmsidrw1jfnwbpqlvwhmsqn2-esp-idf-v5.0.2/components/esptool_py/esptool/esptool.py;--chip;esp32s2
  failed

```

### Steps to reproduce.

1. Have nix with flakes enabled
2. Have the template project at https://github.com/espressif/esp-idf-template/tree/master
3. Run the dev shell `nix develop github:mirrexagon/nixpkgs-esp-dev#esp32s2-idf`
4. With the ESP32-S2FN4R2 connected, run either `idf.py monitor` or `idf.py flash`

### More Information.

Every time I run either command, the esp disconnects and reconnects with this pattern as shown in this dmesg log:
```shell
dmesg --follow-new                                                                                                              [23:17:10]
[10112.710585] usb 1-14: USB disconnect, device number 35
[10112.962802] usb 1-14: new full-speed USB device number 36 using xhci_hcd
[10113.090875] usb 1-14: New USB device found, idVendor=303a, idProduct=8141, bcdDevice= 1.00
[10113.090888] usb 1-14: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[10113.090894] usb 1-14: Product: DENEYAP MINI
[10113.090899] usb 1-14: Manufacturer: Turkish Technnology Team Foundation (T3)
[10113.090903] usb 1-14: SerialNumber: 48:27:e2:5d:f3:46
[10113.093930] cdc_acm 1-14:1.0: ttyACM0: USB ACM device
[10119.531869] usb 1-14: USB disconnect, device number 36
[10119.532010] cdc_acm 1-14:1.1: acm_start_wb - usb_submit_urb(write bulk) failed: -19
[10119.785742] usb 1-14: new full-speed USB device number 37 using xhci_hcd
[10119.913483] usb 1-14: New USB device found, idVendor=303a, idProduct=0002, bcdDevice= 7.23
[10119.913496] usb 1-14: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[10119.913502] usb 1-14: Product: ESP32-S2
[10119.913507] usb 1-14: Manufacturer: Espressif
[10119.913511] usb 1-14: SerialNumber: 0
[10119.916150] cdc_acm 1-14:1.0: ttyACM0: USB ACM device
[10122.106590] usb 1-14: USB disconnect, device number 37
[10122.359712] usb 1-14: new full-speed USB device number 38 using xhci_hcd
[10122.487845] usb 1-14: New USB device found, idVendor=303a, idProduct=8141, bcdDevice= 1.00
[10122.487858] usb 1-14: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[10122.487865] usb 1-14: Product: DENEYAP MINI
[10122.487869] usb 1-14: Manufacturer: Turkish Technnology Team Foundation (T3)
[10122.487874] usb 1-14: SerialNumber: 48:27:e2:5d:f3:46
[10122.490347] cdc_acm 1-14:1.0: ttyACM0: USB ACM device
[10126.421336] usb 1-14: USB disconnect, device number 38
[10126.421449] cdc_acm 1-14:1.1: acm_start_wb - usb_submit_urb(write bulk) failed: -19
[10126.672674] usb 1-14: new full-speed USB device number 39 using xhci_hcd
[10126.800554] usb 1-14: New USB device found, idVendor=303a, idProduct=0002, bcdDevice= 7.23
[10126.800569] usb 1-14: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[10126.800575] usb 1-14: Product: ESP32-S2
[10126.800580] usb 1-14: Manufacturer: Espressif
[10126.800584] usb 1-14: SerialNumber: 0
[10126.803040] cdc_acm 1-14:1.0: ttyACM0: USB ACM device
```
And I also have loaded a hello-world program from the arduino ide in it.

I originally had made this issue https://github.com/mirrexagon/nixpkgs-esp-dev/issues/22 on the repository of the nix flake I am using, but I am not sure if the problem is related to the way python is being packaged or something else.

peterdragun, 2023-07-04:
Hi, you are having issues with the driver not being able to toggle DTR and RTS to reset the chip to download mode. This could be caused by several reasons. Based on your comments, our chip is using USB-OTG peripheral for communication. Have you enabled [USB OTG console](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-guides/usb-otg-console.html#usb-otg-console)? Especially have you followed the part about [initial upload](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-guides/usb-otg-console.html#initial-upload)?

loco-choco, 2023-07-04:
That was indeed the issue, had to run with dfu-util and enable the USB console port on the project. Only after flashing with the tool that the board allowed me to flash with `idf.py flash` and use `idf.py monitor`. Thanks for the help!

closed 2023-07-04 by loco-choco