espressif/esp-idf #11787: Can you provide an example of using QSPI interface to drive LCD?

DoctorTag, 2023-07-01:
### Is your feature request related to a problem?

Can you provide an example of using QSPI interface to drive LCD?

andylinpersonal, 2023-07-02:
Uh, which LCD panel has a QSPI as control interface? 

Most LCD panels should have parallel, i2c, (1 or 2-line) SPI, MIPI DSI (out of scope) and / or eDP (out of scope).

DoctorTag, 2023-07-02:
For example: the LCD driver IC--GC9B71,please refer to the following，thanks.
[image]

modi12jin, 2023-07-02:
> 呃，哪种液晶面板有QSPI控制接口？
> 
> 大多数 LCD 面板应具有并行、i2c、（1 或 2 线）SPI、MIPI DSI（超出范围）和/或 eDP（超出范围）接口。

https://github.com/Lzw655/esp32-s3-lcd-ev-baord-docs/issues/3

Lzw655, 2023-07-06:
@DoctorTag @modi12jin Here is an example for ST77903 QSPI LCD([ST77903_QSPI_1_6_400_400.zip](https://github.com/espressif/esp-idf/files/11963401/ST77903_QSPI_1_6_400_400.zip)). For more details, please see its README. 

The hardware of the LCD Module:
[image]

modi12jin, 2023-07-18:
@Lzw655 Thank you for your reply. I have successfully driven the SPD2010 screen on the ESP32S3, but the touch screen has not been driven up.
[SPD2010_HelloWorld.zip](https://github.com/espressif/esp-idf/files/12086134/SPD2010_HelloWorld.zip)

[image]

https://github.com/espressif/esp-idf/assets/40233017/cdab8ba4-4154-4039-993d-09edf7eb3e99

Lzw655, 2023-07-19:
@modi12jin  For SPD2010 Touch, you can refer to the example: [SPD2010_SPI_2_2_180_454.zip](https://github.com/espressif/esp-idf/files/12090445/SPD2010_SPI_2_2_180_454.zip)

modi12jin, 2023-07-21:
@Lzw655 I'm having some issues, but I can read touch signals

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x2a (SPI_FAST_FLASH_BOOT)
Saved PC:0x42008255
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x1870
load:0x403c9700,len:0x4
load:0x403c9704,len:0xda4
load:0x403cc700,len:0x3130
entry 0x403c9934
[0;32mI (27) boot: ESP-IDF v5.1-dirty 2nd stage bootloader[0m
[0;32mI (27) boot: compile time Jul 20 2023 02:47:50[0m
[0;32mI (27) boot: Multicore bootloader[0m
[0;32mI (30) boot: chip revision: v0.1[0m
[0;32mI (34) qio_mode: Enabling default flash chip QIO[0m
[0;32mI (39) boot.esp32s3: Boot SPI Speed : 80MHz[0m
[0;32mI (44) boot.esp32s3: SPI Mode       : QIO[0m
[0;32mI (49) boot.esp32s3: SPI Flash Size : 8MB[0m
[0;32mI (54) boot: Enabling RNG early entropy source...[0m
[0;32mI (59) boot: Partition Table:[0m
[0;32mI (63) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (85) boot:  2 factory          factory app      00 00 00010000 00100000[0m
[0;32mI (92) boot: End of partition table[0m
[0;32mI (96) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=0d808h ( 55304) map[0m
[0;32mI (113) esp_image: segment 1: paddr=0001d830 vaddr=3fc94a00 size=027e8h ( 10216) load[0m
[0;32mI (116) esp_image: segment 2: paddr=00020020 vaddr=42000020 size=1f134h (127284) map[0m
[0;32mI (141) esp_image: segment 3: paddr=0003f15c vaddr=3fc971e8 size=00b4ch (  2892) load[0m
[0;32mI (142) esp_image: segment 4: paddr=0003fcb0 vaddr=40374000 size=109d4h ( 68052) load[0m
[0;32mI (167) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (167) boot: Disabling RNG early entropy source...[0m
[0;32mI (178) cpu_start: Multicore app[0m
[0;32mI (179) esp_psram: Found 8MB PSRAM device[0m
[0;32mI (179) esp_psram: Speed: 80MHz[0m
[0;32mI (180) cpu_start: Pro cpu up.[0m
[0;32mI (183) cpu_start: Starting app cpu, entry point is 0x403754ac[0m
[0;32mI (0) cpu_start: App cpu up.[0m
[0;32mI (1108) esp_psram: SPI SRAM memory test OK[0m
[0;32mI (1117) cpu_start: Pro cpu start user code[0m
[0;32mI (1117) cpu_start: cpu freq: 240000000 Hz[0m
[0;32mI (1117) cpu_start: Application information:[0m
[0;32mI (1120) cpu_start: Project name:     ESP32S3-IDF_SPD2010-QSPI-I2C[0m
[0;32mI (1127) cpu_start: App version:      1[0m
[0;32mI (1132) cpu_start: Compile time:     Jul 20 2023 02:47:21[0m
[0;32mI (1138) cpu_start: ELF file SHA256:  3f3893897a81ff8e...[0m
[0;32mI (1144) cpu_start: ESP-IDF:          v5.1-dirty[0m
[0;32mI (1149) cpu_start: Min chip rev:     v0.0[0m
[0;32mI (1154) cpu_start: Max chip rev:     v0.99 [0m
[0;32mI (1159) cpu_start: Chip rev:         v0.1[0m
[0;32mI (1164) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (1171) heap_init: At 3FC985F0 len 00051120 (324 KiB): DRAM[0m
[0;32mI (1177) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM[0m
[0;32mI (1184) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM[0m
[0;32mI (1190) heap_init: At 600FE010 len 00001FF0 (7 KiB): RTCRAM[0m
[0;32mI (1197) esp_psram: Adding pool of 8192K of PSRAM memory to heap allocator[0m
[0;32mI (1205) spi_flash: detected chip: gd[0m
[0;32mI (1209) spi_flash: flash io: qio[0m
[0;32mI (1213) sleep: Configure to isolate all GPIO pins in sleep state[0m
[0;32mI (1220) sleep: Enable automatic switching of GPIO sleep configuration[0m
[0;32mI (1227) app_start: Starting scheduler on CPU0[0m
[0;32mI (1232) app_start: Starting scheduler on CPU1[0m
[0;32mI (1232) main_task: Started on CPU0[0m
[0;32mI (1242) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations[0m
[0;32mI (1252) main_task: Calling app_main()[0m
[0;32mI (5252) example: Initialize touch IO (I2C)[0m
[... 10 lines cut ...]
[0;31mE (5322) lcd_panel_touch_spd2010: read_data(157): read data failed[0m
[0;31mE (5332) lcd_panel.io.i2c: panel_io_i2c_tx_buffer(177): i2c transaction failed[0m
[0;31mE (5332) lcd_panel_touch_spd2010: Read_TP_Status_Length(303): Tx failed[0m
[0;31mE (5342) lcd_panel_touch_spd2010: TP_Read_Data(458): Read status length failed[0m
[0;31mE (5352) lcd_panel_touch_spd2010: read_data(157): read data failed[0m
[0;31mE (5352) lcd_panel.io.i2c: panel_io_i2c_tx_buffer(177): i2c transaction failed[0m
[0;31mE (5362) lcd_panel_touch_spd2010: Read_TP_Status_Length(303): Tx failed[0m
[0;31mE (5372) lcd_panel_touch_spd2010: TP_Read_Data(458): Read status length failed[0m
[0;31mE (5382) lcd_panel_touch_spd2010: read_data(157): read data failed[0m
[0;31mE (5382) lcd_panel.io.i2c: panel_io_i2c_tx_buffer(177): i2c transaction failed[0m
[0;31mE (5392) lcd_panel_touch_spd2010: Read_TP_Status_Length(303): Tx failed[0m
[0;31mE (5402) lcd_panel_touch_spd2010: TP_Read_Data(458): Read status length failed[0m
[0;31mE (5412) lcd_panel_touch_spd2010: read_data(157): read data failed[0m
x:187,y:60
x:189,y:78
x:189,y:81
x:189,y:84
x:189,y:86
x:189,y:90
x:189,y:91
x:189,y:92
x:189,y:93
```

https://github.com/espressif/esp-idf/assets/40233017/fd8aadd3-f80f-4f93-8c0e-139d9f36d167

Lzw655, 2023-07-21:
I also don't know the reason. But are the coordinates right? If so, you can ignore these few errors.

modi12jin, 2023-07-21:
@Lzw655 Sorry to bother, I found the problem.
The chip reset and touch reset were connected together before. Then I divided them and reset them separately, and the error disappeared.

modi12jin, 2023-08-01:
@Lzw655 I recorded the SPD2010 demo video, when will the esp_lcd component support the QSPI interface?

> bilibili
[ESP32S3初步驱动QSPI屏幕_SPD2010]( https://www.bilibili.com/video/BV1Lj41197Je/?share_source=copy_web&vd_source=22bec63878894fb747aa6bda6a91f2ad)

> youtube
[ESP32S3 initially drives QSPI screen_SPD2010](https://www.youtube.com/watch?v=FQuWy-_OBsg)

> code
https://github.com/modi12jin/ESP32S3-IDF_SPD2010-QSPI-I2C
https://github.com/modi12jin/Arduino-ESP32-SPD2010

>Adapter board
https://oshwhub.com/modi_555/1-46-cun-qspi-yuan-ping-zhuan-jie-ban

Lzw655, 2023-08-01:
@modi12jin Good job! I just bought two QSPI screens (SPD2010, GC9B71), and I will try to drive them recently.

HongLiangGong1990, 2023-08-06:
> @modi12jin Good job! I just bought two QSPI screens (SPD2010, GC9B71), and I will try to drive them recently.

i have test qspi screens(mentioned above) on GC9B71,  use this code in https://github.com/DoctorTag/esp32S3_watch  .Only when refreshing the arc shape，error pixels will appear，but other shapes are normal.
i have test lvgl version between 8.0~8.3, all are the same.
in this video, SPD2010 has same error(https://www.bilibili.com/video/BV1Lj41197Je/), then I suspect this is a bug in lvgl，but I don't have any ideas
[image]

Lzw655, 2023-08-17:
@DoctorTag Here is a demo project for GC9B61. Please carefully read the "README.md" and configure ESP-IDF as required.

[GC9B71_QSPI_1_89_320_386.zip](https://github.com/espressif/esp-idf/files/12365856/GC9B71_QSPI_1_89_320_386.zip)

Lzw655, 2023-08-17:
> i have test qspi screens(mentioned above) on GC9B71, use this code in https://github.com/DoctorTag/esp32S3_watch .Only when refreshing the arc shape，error pixels will appear，but other shapes are [...]

Please refer to [here](https://esp32.com/viewtopic.php?p=118738#p118738).

HongLiangGong1990, 2023-08-17:
Amazing！Code runs perfectly！

HongLiangGong1990, 2023-08-17:
when i test example: lv_demo_widgets, i found a bug.
[image]
i chose red theme, then flashing stripes appear on the screen，other color theme is ok.
[image]

HongLiangGong1990, 2023-08-17:
By the way, forgot to add one point. Only  when ESP_LCD_IO_QSPI_GC9B71_CONFIG pclk_hz increase to 24M， this bug will  appear

Lzw655, 2023-08-18:
@HongLiangGong1990 Me too. I think it's the limits of GC9B71 itself. But I can't find the timing parameters of the QSPI interface in the datasheet. And I tested another QSPI LCD (SH8601). It worked well under 40MHz PCLK.

modi12jin, 2023-08-18:
@Lzw655 Thanks for your work! My SPD2010 circular screen is damaged and cannot be tested

modi12jin, 2023-08-20:
@Lzw655 Thanks for your code, I tested it perfectly in WEA2012

[image]

```
void lvgl_port_rounder_callback(struct _lv_disp_drv_t *disp_drv, lv_area_t *area) {
  uint16_t x1 = area->x1;
  uint16_t x2 = area->x2;

  // 将区域的起点向下舍入到最接近的偶数
  area->x1 = x1 > 0 ? (x1 - (x1 % 4)) : 0;

  // 将区域末尾四舍五入到最接近的奇数
  area->x2 = (x2 - (x2 % 4) - 1);
}

...

  disp_drv.flush_cb = my_disp_flush;
  disp_drv.rounder_cb = lvgl_port_rounder_callback;
```

> or write like this

```
void lvgl_port_rounder_callback(struct _lv_disp_drv_t *disp_drv, lv_area_t *area) {
  area->x1 &= ~0x03;
  uint32_t tmp = area->x2 % 4;
  if (tmp == 3)
    return;
  area->x2 += -tmp + 3;
}

...

  disp_drv.flush_cb = my_disp_flush;
  disp_drv.rounder_cb = lvgl_port_rounder_callback;
```

Lzw655, 2023-08-21:
@modi12jin Good! For `lvgl_port_rounder_callback()`, I prefer the following code:

```c
void lvgl_port_rounder_callback(struct _lv_disp_drv_t * disp_drv, lv_area_t * area)
{
    uint16_t x1 = area->x1;
    uint16_t x2 = area->x2;

    // round the start of area down to the nearest 4N number
    area->x1 = (x1 >> 2) << 2;

    // round the end of area up to the nearest 4M+3 number
    area->x2 = ((x2 >> 2) << 2) + 3;
}
```

modi12jin, 2023-08-21:
@Lzw655 I don’t know if it’s right to write it like this, it’s changed from 40M to 80M

```
.clock_speed_hz = SPI_MASTER_FREQ_80M,
```

[image]

Lzw655, 2023-08-21:
@modi12jin The highest frequency of ESP SPI (or QSPI) is 80Mhz. It's ok to set it from 40M to 80M if the LCD IC allows it (according to its datasheet).

modi12jin, 2023-08-21:
> @modi12jin The highest frequency of ESP SPI (or QSPI) is 80Mhz. It's ok to set it from 40M to 80M if the LCD IC allows it (according to its datasheet).

[image]

The manual says 40M, but I set it to 80M and it’s fine

Lzw655, 2023-08-21:
@modi12jin Yes, but during mass production, there might be a few products that experience issues.

modi12jin, 2023-08-21:
[ESP32S3-IDF_WEA2012-QSPI-I2C.zip](https://github.com/espressif/esp-idf/files/12396707/ESP32S3-IDF_WEA2012-QSPI-I2C.zip)

@Lzw655 This color problem is suspected to be similar to this problem https://github.com/espressif/esp-idf/issues/10459

[image]

Lzw655, 2023-09-12:
@modi12jin @HongLiangGong1990 The QSPI LCD drivers including [SPD2010](https://components.espressif.com/components/espressif/esp_lcd_spd2010), [SH8601](https://components.espressif.com/components/espressif/esp_lcd_sh8601), and [GC9B71](https://components.espressif.com/components/espressif/esp_lcd_gc9b71) have been uploaded to the Espressif Registry. There is a common example named qspi_with_ram for them and it can also be downloaded from Espressif Registry.

closed 2023-11-02 by wujiangang

Windsssss, 2023-11-16:
> @modi12jin @HongLiangGong1990 The QSPI LCD drivers including [SPD2010](https://components.espressif.com/components/espressif/esp_lcd_spd2010), [...]

@Lzw655 hello，i just bought a spd2010 lcd，and i test the example you upload to Espressif Registry，i found a problem below, the screen lit up for a few seconds and then went out.

https://github.com/espressif/esp-idf/assets/36975516/0e8ece32-9c4d-4320-8da0-1759cce07fb1

Lzw655, 2023-11-16:
Hi @Windsssss, please open a new issue and choose "Runtime bug report" in [esp-iot-solution](https://github.com/espressif/esp-iot-solution/issues). And it will be very helpful if you attach the detailed logs.

Windsssss, 2023-11-17:
> Hi @Windsssss, please open a new issue and choose "Runtime bug report" in [esp-iot-solution](https://github.com/espressif/esp-iot-solution/issues). And it will be very helpful if you attach the detailed logs.

@Lzw655 , https://github.com/espressif/esp-iot-solution/issues/312

Lzw655, 2023-11-17:
> > Hi @Windsssss, please open a new issue and choose "Runtime bug report" in [esp-iot-solution](https://github.com/espressif/esp-iot-solution/issues). And it will be very helpful if you attach the detailed logs.
> 
> @Lzw655 , [espressif/esp-iot-solution#312](https://github.com/espressif/esp-iot-solution/issues/312)

Ok, thank you!

mainy4210, 2024-04-24:
Are there any new routines for ST77903?