espressif/esp-idf #11788: adc_io_to_channel non-const pointer parameter

tpmlinz, 2023-07-01:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

In both cases below the `adc_uint_t*` parameter should almost certainly be a const pointer.

adc_common
`esp_err_t adc_io_to_channel(int, adc_unit_t*, adc_channel_t*)`

adc_oneshot
`esp_err_t adc_oneshot_io_to_channel(int, adc_unit_t*, adc_channel_t*)`

p.s. adc_continuous probably has the same issue.

p.p.s.
The first parameter `int io` should probably be of type `uint8_t` to be consistent with other parts of the API but regardless of size the sign seems superfluous.

Alvin1Zhang, 2023-07-03:
Thanks for reporting.

Icarus113, 2023-07-03:
@tpmlinz Thanks for the feedback~ However we cannot change the API signature at this moment. We'll consider this when introducing a next major release.

Icarus113, 2023-07-07:
I'll close this issue in next few days as we can't introduce breaking changes at this moment.
I've recorded this feedback and will re-consider this before next major release.

Will keep this issue open for few days. Let me know if there's any comment~ @tpmlinz

closed 2023-07-07 by Icarus113

Icarus113, 2023-08-02:
Hi @tpmlinz  We adopted the const pointer suggestion, about the `int io`, as it's a breaking change we'll still delay it to next major release and re-consider it.

The commit will be synced to GH master soon.