espressif/esp-idf #11789: bt_spp_acceptor example drops connection immediately with rsn:0x13

cwahn, 2023-07-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.2

### Operating System used.

macOS

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-WOORM

### Power Supply used.

USB

### What is the expected behavior?

The device should be connected with PC and discovered as serial device.

### What is the actual behavior?

The connection get dropped immediately after successful connection, with rsn:0x13. 

### Steps to reproduce.

1. Copy, build, and flash bt_spp_acceptor example to ESP32-WROOM
2. Connect to the device from PC(MacBook Pro, M1 Pro, macos Monterey ) 
3. Once it get connected successfully but the connection gets dropped immediately after connection. with following log message.

### Debug Logs.

```plain
boot: chip revision: v1.0
I (16) boot.esp32: SPI SLJ��ѹesp32: SPI Mode       : DIO
I (25) boot.esp32: SPI Flash Size : 2MB
I (29) boot: En�I (13) boot: ESP-IDF v5.0.2-dirty 2nd stage bootloader
I (13) boot: compile time 01:42:09
I (13) boot: chip revision: v1.0
I (16) boot.esp32: SPI Speed      : 40MHz
I (20) boot.esp32: SPI Mode       : DIO
I (25) boot.esp32: SPI Flash Size : 2MB
I (29) boot: Enabling RNG early entropy source...
I (35) boot: Partition Table:
I (38) boot: ## Label            Usage          Type ST Offset   Length
I (46) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (53) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (61) boot:  2 factory          factory app      00 00 00010000 00100000
I (68) boot: End of partition table
I (72) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1e78ch (124812) map
I (126) esp_image: segment 1: paddr=0002e7b4 vaddr=3ffbdb60 size=01864h (  6244) load
I (129) esp_image: segment 2: paddr=00030020 vaddr=400d0020 size=6f5a8h (456104) map
I (296) esp_image: segment 3: paddr=0009f5d0 vaddr=3ffbf3c4 size=02948h ( 10568) load
I (301) esp_image: segment 4: paddr=000a1f20 vaddr=40080000 size=18a38h (100920) load
I (356) boot: Loaded app from partition at offset 0x10000
I (356) boot: Disabling RNG early entropy source...
I (368) cpu_start: Pro cpu up.
I (368) cpu_start: Starting app cpu, entry point is 0x400813e8
0x400813e8: call_start_cpu1 at /Users/myname/esp/esp-idf/components/esp_system/port/cpu_start.c:141

I (0) cpu_start: App cpu up.
I (384) cpu_start: Pro cpu start user code
I (384) cpu_start: cpu freq: 160000000 Hz
I (384) cpu_start: Application information:
I (389) cpu_start: Project name:     bt_spp_acceptor_demo
I (395) cpu_start: App version:      1
I (400) cpu_start: Compile time:     Jul  3 2023 10:24:45
I (406) cpu_start: ELF file SHA256:  e00638ef06e4fb84...
I (412) cpu_start: ESP-IDF:          v5.0.2-dirty
I (417) cpu_start: Min chip rev:     v0.0
I (422) cpu_start: Max chip rev:     v3.99 
I (427) cpu_start: Chip rev:         v1.0
I (431) heap_init: Initializing. RAM available for dynamic allocation:
I (439) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (444) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (451) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (457) heap_init: At 3FFC7898 len 00018768 (97 KiB): DRAM
I (463) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (469) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (476) heap_init: At 40098A38 len 000075C8 (29 KiB): IRAM
I (483) spi_flash: detected chip: generic
I (487) spi_flash: flash io: dio
W (491) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (505) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (536) BTDM_INIT: BT controller compile version [2c56073]
I (536) system_api: Base MAC address is not set
I (536) system_api: read default base MAC address from EFUSE
I (546) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (1106) BT_BTM: btm_sec_set_security_level : sec: 0x0

I (1106) BT_BTM: BTM_SEC_REG[0]: id 0, is_orig 0, psm 0x0001, proto_id 0, chan_id 0

I (1106) BT_BTM:                : sec: 0x80, service name [Service Discovery] (up to 21 chars saved)

I (1116) BT_BTM: btm_sec_set_security_level : sec: 0x0

I (1126) BT_BTM: BTM_SEC_REG[0]: id 0, is_orig 1, psm 0x0001, proto_id 0, chan_id 0

I (1126) BT_BTM:                : sec: 0x80, service name [Service Discovery] (up to 21 chars saved)

I (1196) BT_BTM: BTM_SetPinType: pin type 0 [variable-0, fixed-1], code , length 0

I (1196) BT_BTM: btm_sec_set_security_level : sec: 0x0

I (1196) BT_BTM: BTM_SEC_REG[1]: id 42, is_orig 0, psm 0x0003, proto_id 3, chan_id 0

I (1206) BT_BTM:                : sec: 0x80, service name [RFC_MUX
] (up to 21 chars saved)

I (1216) BT_BTM: BTM_SetInquiryMode

I (1226) BT_BTM: BTM_SetPageScanType

I (1226) BT_BTM: BTM_SetInquiryScanType

I (1246) BT_BTM: BTM_SecAddDevice, link key type:4

I (1246) BT_BTM: BTM_SecAddDevice, link key type:4

I (1246) BT_BTM: Write Extended Inquiry Response to controller

I (1256) SPP_ACCEPTOR_DEMO: ESP_SPP_INIT_EVT
I (1266) BT_BTM: btm_sec_set_security_level : sec: 0x12

I (1266) BT_BTM: BTM_SEC_REG[2]: id 55, is_orig 0, psm 0x0003, proto_id 3, chan_id 1

I (1276) BT_BTM:                : sec: 0x1082, service name [JV PORT] (up to 21 chars saved)

I (1286) SPP_ACCEPTOR_DEMO: ESP_SPP_START_EVT handle:129 sec_id:55 scn:1
I (1296) BT_BTM: Write Extended Inquiry Response to controller

I (1296) BT_BTM: BTM_ReadDiscoverability

I (1306) BT_BTM: BTM_ReadConnectability

I (1306) BT_BTM: BTM_SetDiscoverability

I (1316) BT_BTM: BTM_SetDiscoverability: mode 2 [NonDisc-0, Lim-1, Gen-2], window 0x0012, interval 0x0800

I (1326) BT_BTM: BTM_SetConnectability

I (1326) BT_BTM: BTM_SetConnectability: mode 1 [NonConn-0, Conn-1], window 0x0012, interval 0x0800

I (1336) SPP_ACCEPTOR_DEMO: event: 10
I (1346) BT_BTM: BTM_SetPinType: pin type 0 [variable-0, fixed-1], code �?#, length 0

I (1346) SPP_ACCEPTOR_DEMO: Own address:[98:cd:ac:ab:c2:9e]
W (7316) BT_HCI: hcif conn complete: hdl 0x80, st 0x0
I (7316) BT_BTM: BTM_InqDbRead: bd addr [f4d48877c323]

I (7326) SPP_ACCEPTOR_DEMO: event: 16
I (7386) BT_BTM: BTM_SetPowerMode: pm_id 128 BDA: 8877c323 mode:0x0
I (7386) BT_BTM: btm_sec_set_peer_sec_caps: sm4: 0x11, rmt_support_for_secure_connections 1

I (7386) BT_BTM: btm_process_remote_ext_features: pend:0

I (7406) BT_BTM: btm_sec_set_peer_sec_caps: sm4: 0x11, rmt_support_for_secure_connections 1

I (7406) BT_BTM: btm_process_remote_ext_features: pend:0

W (7416) BT_HCI: hcif link supv_to changed: hdl 0x80, supv_to 8000
I (7546) BT_BTM: BTM_SetPowerMode: pm_id 128 BDA: 8877c323 mode:0x0
W (7626) BT_HCI: hcif link supv_to changed: hdl 0x80, supv_to 32000
W (11556) BT_HCI: hci cmd send: disconnect: hdl 0x80, rsn:0x13
W (11696) BT_HCI: hcif disc complete: hdl 0x80, rsn 0x16
```

### More Information.

It might be related with the open issue; https://github.com/espressif/esp-idf/issues/11681#issue-1759089560 .

xiongweichao, 2023-07-03:
Hi @cwahn ,

PC does not support SPP by default. You can refer to https://developer.apple.com/accessories/Accessory-Design-Guidelines.pdf

Thanks

cwahn, 2023-07-03:
> Hi @cwahn ,
> 
> PC does not support SPP by default. You can refer to https://developer.apple.com/accessories/Accessory-Design-Guidelines.pdf
> 
> Thanks

Hi @xiongweichao 

Thank you for the reply.
Then the issue should be closed.

By the way, is there any other way ESP32 could do serial communication with PC over bluetooth?

xiongweichao, 2023-07-03:
@cwahn  Sorry, I don't know about MacBook. But I think you can try this [way](https://unix.stackexchange.com/questions/92255/how-do-i-connect-and-send-data-to-a-bluetooth-serial-port-on-linux).

xiongweichao, 2023-09-07:
Hi @Talha-Babar ,

Since the mobile phone does not support SPP, the connection will be disconnected after the pairing is completed. If you want to establish a connection with the mobile phone but not send any data, it is recommended that you use other profiles([A2DP](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/classic_bt/a2dp_sink), [HFP](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/classic_bt/hfp_hf)).

Thanks

closed 2023-12-14 by Alvin1Zhang