espressif/esp-idf #11790: esp32s3 CVBS UVC - Negotiating streaming format failed

berkutta, 2023-07-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello

I'm trying to get the [UVC Host Example](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/host/uvc) working with a USB CVBS UVC converter. Unfortunately, it fails with the message `Negotiating streaming format failed, trying again...`

Is there something I can try to get past the negotiating? Or maybe a way to force it into a specific format?

Full Log:
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3820,len:0x1718
load:0x403c9700,len:0x4
load:0x403c9704,len:0xc08
load:0x403cc700,len:0x2f08
entry 0x403c990c
I (27) boot: ESP-IDF v5.2-dev-1349-gcf7e743a9b 2nd stage bootloader
I (27) boot: compile time Jul  3 2023 10:59:23
I (28) boot: Multicore bootloader
I (32) boot: chip revision: v0.1
I (35) boot.esp32s3: Boot SPI Speed : 80MHz
I (40) boot.esp32s3: SPI Mode       : DIO
I (45) boot.esp32s3: SPI Flash Size : 2MB
I (50) boot: Enabling RNG early entropy source...
I (55) boot: Partition Table:
I (59) boot: ## Label            Usage          Type ST Offset   Length
I (66) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (73) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (81) boot:  2 factory          factory app      00 00 00010000 00100000
I (88) boot: End of partition table
I (93) esp_image: segment 0: paddr=00010020 vaddr=3c030020 size=0e320h ( 58144) map
I (111) esp_image: segment 1: paddr=0001e348 vaddr=3fc92d00 size=01cd0h (  7376) load
I (113) esp_image: segment 2: paddr=00020020 vaddr=42000020 size=259f4h (154100) map
I (146) esp_image: segment 3: paddr=00045a1c vaddr=3fc949d0 size=00f5ch (  3932) load
I (147) esp_image: segment 4: paddr=00046980 vaddr=40374000 size=0ec6ch ( 60524) load
I (171) boot: Loaded app from partition at offset 0x10000
I (171) boot: Disabling RNG early entropy source...
I (182) cpu_start: Multicore app
I (183) esp_psram: Found 2MB PSRAM device
I (183) esp_psram: Speed: 40MHz
I (184) cpu_start: Pro cpu up.
I (187) cpu_start: Starting app cpu, entry point is 0x403752d8
0x403752d8: call_start_cpu1 at /home/user/esp/esp-idf/components/esp_system/port/cpu_start.c:154

I (0) cpu_start: App cpu up.
I (613) esp_psram: SPI SRAM memory test OK
I (622) cpu_start: Pro cpu start user code
I (622) cpu_start: cpu freq: 160000000 Hz
I (622) cpu_start: Application information:
I (625) cpu_start: Project name:     usb-uvc
I (630) cpu_start: App version:      v5.2-dev-1349-gcf7e743a9b
I (637) cpu_start: Compile time:     Jul  3 2023 10:59:16
I (643) cpu_start: ELF file SHA256:  cec1f690b...
I (648) cpu_start: ESP-IDF:          v5.2-dev-1349-gcf7e743a9b
I (655) cpu_start: Min chip rev:     v0.0
I (659) cpu_start: Max chip rev:     v0.99 
I (664) cpu_start: Chip rev:         v0.1
I (669) heap_init: Initializing. RAM available for dynamic allocation:
I (676) heap_init: At 3FC96208 len 00053508 (333 KiB): DRAM
I (682) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
[... 137 lines cut ...]
Selection Video Capture: crop_bounds, Left 0, Top 0, Width 640, Height 480, Flags: 
Streaming Parameters Video Capture:
        Capabilities     : timeperframe
        Frames per second: 30.000 (30/1)
        Read buffers     : 0

User Controls

                     brightness 0x00980900 (int)    : min=0 max=255 step=1 default=32 value=32
                       contrast 0x00980901 (int)    : min=0 max=255 step=1 default=132 value=148
                     saturation 0x00980902 (int)    : min=0 max=255 step=1 default=128 value=160
                            hue 0x00980903 (int)    : min=0 max=127 step=1 default=0 value=0
```

Linux dmesg:
```
[ 8740.247257] usb 1-4.1.4: new high-speed USB device number 37 using xhci_hcd
[ 8740.338926] usb 1-4.1.4: New USB device found, idVendor=534d, idProduct=0021, bcdDevice= 1.21
[ 8740.338930] usb 1-4.1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 8740.338931] usb 1-4.1.4: Product: AV TO USB2.0
[ 8740.338933] usb 1-4.1.4: Manufacturer: MACROSILICON
[ 8740.338933] usb 1-4.1.4: SerialNumber: 20150130
[ 8740.340373] usb 1-4.1.4: Found UVC 1.00 device AV TO USB2.0 (534d:0021)
[ 8740.340690] usb 1-4.1.4: UVC non compliance - GET_DEF(PROBE) not supported. Enabling workaround.
```

nopnop2002, 2023-07-19:
https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/host/uvc

The USB cameras that can be used with this sample are very limited.

This might help.

https://github.com/nopnop2002/esp-idf-video-streaming

roma-jam, 2023-07-31:
Hi @berkutta, 
Sorry for such a long response with this issue also. 

This is the same thing, as with the issue #11791.
The example tries to negotiate protocol with following parameters: 320x240, interval 1/30. 

To be able to work with other cameras, choose the interval that your camera has. 
You can find this intervals in the list during debug output of the example: 

                        FrameDescriptor(1)
                          capabilities: 00
                          size: 480x320                           <--- size here
                          bit rate: 768000-196608000
                          max frame size: 307200
                          default interval: 1/25                  <--- default interval here
                          interval[0]: 1/25
                        FrameDescriptor(2)
                          capabilities: 00
                          **size: 640x480**
                          bit rate: 768000-196608000
                          max frame size: 614400
                          **default interval: 1/25**
                          interval[0]: 1/25
                          interval[1]: 1/20
                          interval[2]: 1/15
                          interval[3]: 1/10
                        FrameDescriptor(3)
                          capabilities: 00
                          **size: 720x480**
                          bit rate: 768000-196608000
                          max frame size: 691200
                          **default interval: 1/25**
                          interval[0]: 1/25
                          interval[1]: 1/20
                          interval[2]: 1/15
                          interval[3]: 1/10

Feel free to choose any of available frame parameters (640x480 with the interval 1/5 should work well).

closed 2023-08-04 by Alvin1Zhang