espressif/esp-idf #11791: esp32s3 UVC Camera - Negotiating streaming format failed

berkutta, 2023-07-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello

I'm trying to get the [UVC Host Example](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/host/uvc) working with a USB UVC camera. Unfortunately, it fails with the message `Negotiating streaming format failed, trying again...`

Is there something I can try to get past the negotiating? Or maybe a way to force it into a specific format?

Device: https://aliexpress.com/item/1005004018738930.html

Full Log:
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3820,len:0x1718
load:0x403c9700,len:0x4
load:0x403c9704,len:0xc08
load:0x403cc700,len:0x2f08
entry 0x403c990c
I (27) boot: ESP-IDF v5.2-dev-1349-gcf7e743a9b 2nd stage bootloader
I (27) boot: compile time Jul  3 2023 10:59:23
I (28) boot: Multicore bootloader
I (32) boot: chip revision: v0.1
I (35) boot.esp32s3: Boot SPI Speed : 80MHz
I (40) boot.esp32s3: SPI Mode       : DIO
I (45) boot.esp32s3: SPI Flash Size : 2MB
I (50) boot: Enabling RNG early entropy source...
I (55) boot: Partition Table:
I (59) boot: ## Label            Usage          Type ST Offset   Length
I (66) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (73) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (81) boot:  2 factory          factory app      00 00 00010000 00100000
I (88) boot: End of partition table
I (93) esp_image: segment 0: paddr=00010020 vaddr=3c030020 size=0e320h ( 58144) map
I (111) esp_image: segment 1: paddr=0001e348 vaddr=3fc92d00 size=01cd0h (  7376) load
I (113) esp_image: segment 2: paddr=00020020 vaddr=42000020 size=259f4h (154100) map
I (146) esp_image: segment 3: paddr=00045a1c vaddr=3fc949d0 size=00f5ch (  3932) load
I (147) esp_image: segment 4: paddr=00046980 vaddr=40374000 size=0ec6ch ( 60524) load
I (171) boot: Loaded app from partition at offset 0x10000
I (171) boot: Disabling RNG early entropy source...
I (182) cpu_start: Multicore app
I (183) esp_psram: Found 2MB PSRAM device
I (183) esp_psram: Speed: 40MHz
I (184) cpu_start: Pro cpu up.
I (187) cpu_start: Starting app cpu, entry point is 0x403752d8
0x403752d8: call_start_cpu1 at /home/user/esp/esp-idf/components/esp_system/port/cpu_start.c:154

I (0) cpu_start: App cpu up.
I (613) esp_psram: SPI SRAM memory test OK
I (622) cpu_start: Pro cpu start user code
I (622) cpu_start: cpu freq: 160000000 Hz
I (622) cpu_start: Application information:
I (625) cpu_start: Project name:     usb-uvc
I (630) cpu_start: App version:      v5.2-dev-1349-gcf7e743a9b
I (637) cpu_start: Compile time:     Jul  3 2023 10:59:16
I (643) cpu_start: ELF file SHA256:  cec1f690b...
I (648) cpu_start: ESP-IDF:          v5.2-dev-1349-gcf7e743a9b
I (655) cpu_start: Min chip rev:     v0.0
I (659) cpu_start: Max chip rev:     v0.99 
I (664) cpu_start: Chip rev:         v0.1
I (669) heap_init: Initializing. RAM available for dynamic allocation:
[... 144 lines cut ...]
                                2: 60 Hz
      white_balance_temperature 0x0098091a (int)    : min=2800 max=6500 step=1 default=4600 value=4600 flags=inactive
                      sharpness 0x0098091b (int)    : min=0 max=6 step=1 default=5 value=5
         backlight_compensation 0x0098091c (int)    : min=0 max=2 step=1 default=1 value=1

Camera Controls

                  auto_exposure 0x009a0901 (menu)   : min=0 max=3 default=3 value=3 (Aperture Priority Mode)
                                1: Manual Mode
                                3: Aperture Priority Mode
         exposure_time_absolute 0x009a0902 (int)    : min=1 max=5000 step=1 default=625 value=625 flags=inactive
     exposure_dynamic_framerate 0x009a0903 (bool)   : default=0 value=1

```

Linux dmesg:
```
[ 9041.561213] usb 1-4.1.4: new high-speed USB device number 45 using xhci_hcd
[ 9041.709760] usb 1-4.1.4: New USB device found, idVendor=0c45, idProduct=6366, bcdDevice= 0.02
[ 9041.709781] usb 1-4.1.4: New USB device strings: Mfr=2, Product=1, SerialNumber=3
[ 9041.709793] usb 1-4.1.4: Product: Webcam
[ 9041.709800] usb 1-4.1.4: Manufacturer: Sonix Technology Co., Ltd.
[ 9041.709809] usb 1-4.1.4: SerialNumber: Webcam
[ 9041.715401] usb 1-4.1.4: Found UVC 1.00 device Webcam (0c45:6366)
```

Alvin1Zhang, 2023-07-03:
Thanks for reporting.

roma-jam, 2023-07-31:
Hi @berkutta , 

Sorry for such a long response. 
Just to be clear, may I ask which board do you use (if that is a development board)? 

Also, the example of uvc tries to find the interface with 320x240, 1/30 interval, but in your case the camera has only one interface with 320x240, 1/5 interval available.
To negotiate correctly, I would suggest to change FPS define value from 30 to 5 ([main.c, line 27](https://github.com/espressif/esp-idf/blob/master/examples/peripherals/usb/host/uvc/main/main.c#L27))

roma-jam, 2023-07-31:
@berkutta, 

Thanks for providing logs and so much information. 
I will close this issue as it duplicates with another one. Is there is anything else in this case, feel free to reopen. 

Duplicated issue #11790

closed 2023-07-31 by roma-jam