espressif/esp-idf #11792: Ping timeout

baldhead69, 2023-07-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi,

I am trying to ping my AP to confirm that the esp32 lib is working, but a timeout occur.

What may be the problem ?

Code:
```
////////////////////////////////////////////////////////////////////////////////////////////
//

static void test_on_ping_success(esp_ping_handle_t hdl, void *args)
{
    // optionally, get callback arguments
    // const char* str = (const char*) args;
    // printf("%s\r\n", str); // "foo"
    uint8_t ttl;
    uint16_t seqno;
    uint32_t elapsed_time, recv_len;
    ip_addr_t target_addr;
    esp_ping_get_profile(hdl, ESP_PING_PROF_SEQNO, &seqno, sizeof(seqno));
    esp_ping_get_profile(hdl, ESP_PING_PROF_TTL, &ttl, sizeof(ttl));
    esp_ping_get_profile(hdl, ESP_PING_PROF_IPADDR, &target_addr, sizeof(target_addr));
    esp_ping_get_profile(hdl, ESP_PING_PROF_SIZE, &recv_len, sizeof(recv_len));
    esp_ping_get_profile(hdl, ESP_PING_PROF_TIMEGAP, &elapsed_time, sizeof(elapsed_time));
    printf("%lu bytes from %s icmp_seq=%u ttl=%u time=%lu ms\n", recv_len, inet_ntoa(target_addr.u_addr.ip4), seqno, ttl, elapsed_time);
}

////////////////////////////////////////////////////////////////////////////////////////////
//

static void test_on_ping_timeout(esp_ping_handle_t hdl, void *args)
{
    uint16_t seqno;
    ip_addr_t target_addr;
    esp_ping_get_profile(hdl, ESP_PING_PROF_SEQNO, &seqno, sizeof(seqno));
    esp_ping_get_profile(hdl, ESP_PING_PROF_IPADDR, &target_addr, sizeof(target_addr));
    printf("From %s icmp_seq=%d timeout\n", inet_ntoa(target_addr.u_addr.ip4), seqno);
}

////////////////////////////////////////////////////////////////////////////////////////////
//

static void test_on_ping_end(esp_ping_handle_t hdl, void *args)
{
    uint32_t transmitted;
    uint32_t received;
    uint32_t total_time_ms;

    esp_ping_get_profile(hdl, ESP_PING_PROF_REQUEST, &transmitted, sizeof(transmitted));
    esp_ping_get_profile(hdl, ESP_PING_PROF_REPLY, &received, sizeof(received));
    esp_ping_get_profile(hdl, ESP_PING_PROF_DURATION, &total_time_ms, sizeof(total_time_ms));
    printf("%lu packets transmitted, %lu received, time %lums\n", transmitted, received, total_time_ms);
}

////////////////////////////////////////////////////////////////////////////////////////////
//

void initialize_ping()
{
/*    
    // convert URL to IP address
    ip_addr_t target_addr;
    struct addrinfo hint;
    struct addrinfo *res = NULL;
    memset(&hint, 0, sizeof(hint));
    memset(&target_addr, 0, sizeof(target_addr));
    
    int ret = getaddrinfo(MY_PING_URL, NULL, &hint, &res);
    if( ret != 0 )
    {
        printf( "%s: Unknown host [ %s ]\n", MY_PING_TAG, MY_PING_URL );
        printf( "ret = %d\n", ret );
        return;
    }

    struct in_addr addr4 = ((struct sockaddr_in *) (res->ai_addr))->sin_addr;
    inet_addr_to_ip4addr(ip_2_ip4(&target_addr), &addr4);
    freeaddrinfo(res);

    esp_ping_config_t ping_config = ESP_PING_DEFAULT_CONFIG();
    ping_config.target_addr = target_addr;          // target IP address
    ping_config.count = ESP_PING_COUNT_INFINITE;    // ping in infinite mode, esp_ping_stop can stop it
*/

    ip_addr_t target_addr;

    //target_addr.u_addr.ip4.addr = 3372329152;  // u32_t. 192.168.1.201. Tem que escrever 201.1.168.192.    

    int ret = inet_aton( "192.168.1.201", &target_addr.u_addr.ip4 );
    if( ret == 0 )
    {
        printf( "%s: cp could not be converted to addr\n", MY_PING_TAG );        
    }
    printf( "target_addr.u_addr.ip4 = %lu\n", target_addr.u_addr.ip4.addr );

    ret = inet_aton( "192.168.1.1", &target_addr.u_addr.ip4 );
    if( ret == 0 )
    {
        printf( "%s: cp could not be converted to addr\n", MY_PING_TAG );        
    }
    printf( "target_addr.u_addr.ip4 = %lu\n", target_addr.u_addr.ip4.addr );

    esp_ping_config_t ping_config = ESP_PING_DEFAULT_CONFIG();
    ping_config.target_addr = target_addr;  // target IP address
    ping_config.count = ESP_PING_COUNT_INFINITE;    // ping in infinite mode, esp_ping_stop can stop it
    ping_config.timeout_ms = 5000;

    // set callback functions
    esp_ping_callbacks_t cbs;
    cbs.on_ping_success = test_on_ping_success;
    cbs.on_ping_timeout = test_on_ping_timeout;
    cbs.on_ping_end = test_on_ping_end;
    cbs.cb_args = NULL;  // arguments that will feed to all callback functions, can be NULL
    
    esp_ping_handle_t ping;
    esp_ping_new_session(&ping_config, &cbs, &ping);
    esp_ping_start(ping);
}
```

esp-if monitor:
```
From 192.168.1.1 icmp_seq=1 timeout
E (9184) ping_sock: send error=0
I (9283) MY_APP: MALLOC_CAP_INTERNAL = 161860
I (9283) MY_APP: MALLOC_CAP_SPIRAM = 6789840

From 192.168.1.1 icmp_seq=2 timeout
E (14184) ping_sock: send error=0
I (14283) MY_APP: MALLOC_CAP_INTERNAL = 162136
I (14283) MY_APP: MALLOC_CAP_SPIRAM = 6789840

From 192.168.1.1 icmp_seq=3 timeout
E (19184) ping_sock: send error=0
I (19283) MY_APP: MALLOC_CAP_INTERNAL = 162136
I (19283) MY_APP: MALLOC_CAP_SPIRAM = 6789840

From 192.168.1.1 icmp_seq=4 timeout
E (24184) ping_sock: send error=0
I (24283) MY_APP: MALLOC_CAP_INTERNAL = 162136
I (24283) MY_APP: MALLOC_CAP_SPIRAM = 6789840

From 192.168.1.1 icmp_seq=5 timeout
E (29184) ping_sock: send error=0
I (29283) MY_APP: MALLOC_CAP_INTERNAL = 162136
I (29283) MY_APP: MALLOC_CAP_SPIRAM = 6789840

From 192.168.1.1 icmp_seq=6 timeout
E (34184) ping_sock: send error=0
I (34283) MY_APP: MALLOC_CAP_INTERNAL = 162136
I (34283) MY_APP: MALLOC_CAP_SPIRAM = 6789840

From 192.168.1.1 icmp_seq=7 timeout
E (39184) ping_sock: send error=0
I (39283) MY_APP: MALLOC_CAP_INTERNAL = 162136
I (39283) MY_APP: MALLOC_CAP_SPIRAM = 6789840

From 192.168.1.1 icmp_seq=8 timeout
E (44184) ping_sock: send error=0
I (44283) MY_APP: MALLOC_CAP_INTERNAL = 162136
I (44283) MY_APP: MALLOC_CAP_SPIRAM = 6789840

From 192.168.1.1 icmp_seq=9 timeout
E (49184) ping_sock: send error=0
I (49283) MY_APP: MALLOC_CAP_INTERNAL = 162136
I (49283) MY_APP: MALLOC_CAP_SPIRAM = 6789840

From 192.168.1.1 icmp_seq=10 timeout
E (54184) ping_sock: send error=0
I (54283) MY_APP: MALLOC_CAP_INTERNAL = 162136
I (54283) MY_APP: MALLOC_CAP_SPIRAM = 6789840
```

Notes:
esp-idf v5.0.2

Thank's.

nopnop2002, 2023-07-04:
try this.

https://github.com/nopnop2002/esp-idf-ping/tree/master

baldhead69, 2023-07-04:
@nopnop2002 

Your project worked, thank's.

I will take a better look at your code.

My project have a websocket server running on it.

Could it be that the websocket server is causing the error in my project ?

baldhead69, 2023-07-04:
Hi,

I changed this in my code:

* ESP_ERROR_CHECK( esp_wifi_set_ps( WIFI_PS_NONE ) );
* target_addr.type = IPADDR_TYPE_V4;

But i think that the responsible for the non-functioning was only this:  "target_addr.type = IPADDR_TYPE_V4;"

Thank's again @nopnop2002. 

Please espressif, fix the example once and for all.
https://docs.espressif.com/projects/esp-idf/en/v5.0.2/esp32s3/api-reference/protocols/icmp_echo.html?highlight=icmp

david-cermak, 2023-10-11:
@baldhead69 do yo have IPv6 capable local network?

> But i think that the responsible for the non-functioning was only the lack of this: "target_addr.type = IPADDR_TYPE_V4;"

I don't think we should restrict the example to support the IPv4 only networks. In case users don't want to use IPv6 they could simply disable it and the example would work as IPv4 only. Do you see any issue with that?

baldhead69, 2023-10-11:
> @baldhead69 do yo have IPv6 capable local network?
>
>
> I don't think we should restrict the example to support the IPv4 only networks. In case users don't want to use IPv6 they could simply disable it [...]

Hi @david-cermak ,

In principle this problem is solved for me.

Could you help me with this ?

https://www.esp32.com/viewtopic.php?f=13&t=34682&sid=ae1b5c32da704235f4abd52edef48051&start=30

no one from espressif answers anything

david-cermak, 2023-10-31:
> In principle this problem is solved for me.

okay, closing. Also seeing that your linked issue has been addressed, too.

closed 2023-10-31 by david-cermak