espressif/esp-idf #11795: BT SPP Acceptor example crashes on disconnect

aircable, 2023-07-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

release/v5.1 and master

### Operating System used.

Linux

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

External 3.3V

### What is the expected behavior?

not crashing after disconnect

### What is the actual behavior?

I (4603681) BDSPP: ESP_SPP_CLOSE_EVT
BDSPP: ACTIVE
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4001a033  PS      : 0x00060130  A0      : 0x800213fc  A1      : 0x3ffcfcd0  
A2      : 0x00410004  A3      : 0xffff8338  A4      : 0x00410000  A5      : 0x3ffb8360  
A6      : 0x00000064  A7      : 0x00000000  A8      : 0x8014c698  A9      : 0x3ffcfca0  
A10     : 0x4014cdf4  A11     : 0x00000036  A12     : 0x3ffb32dc  A13     : 0x0000002f  
0x4014cdf4: r_hci_tl_send at ??:?

A14     : 0x00000030  A15     : 0x00000001  SAR     : 0x00000019  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00410000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x4001a030:0x3ffcfcd0 0x400213f9:0x3ffcfd00 0x4002f326:0x3ffcfd20 0x40151201:0x3ffcfd50 0x40086740:0x3ffcfd80 0x40019d11:0x3ffcfdb0 0x40055b4d:0x3ffcfdd0 0x4014ac3b:0x3ffcfdf0 0x4014b2ad:0x3ffcfe10
0x40151201: lc_acl_disc_ind_handler_hack at lc_task.c:?

0x40086740: ke_task_schedule at ke_task.c:?

0x4014ac3b: r_rw_schedule at ??:?

0x4014b2ad: btdm_controller_task at ??:?

### Steps to reproduce.

1. Step
Configure the IDF to allow:

- Bluetooth modem sleep enable, internal crystal
- Power mode, 240MUz max, 80MHz min, light sleep enable
- Dynamic CPU  frequency automatic
- Heap tracing comprehensive poisening

3. Step
Run SPP acceptor, make connection, observe the GAP mode change for SNIFF and ACTIVE
5. Step
disconnect and wait about 1 second after SPP_CLOSE_EVENT
and the GAP ACTIVE mode event and see the crash

Seems to me a particular hack (function name) is still in place for a workaround.

### Debug Logs.

```plain
see above
```

### More Information.

Source file showing lc_acl_disc_ind_handler_hack at lc_task.c is not available.
No way for others to find this bug.

Seems like r_hci_tl_send tries to send data while the connection is already closed and crashes.

xiongweichao, 2023-07-04:
Hi @aircable ，

ESP32 does [not support ](https://github.com/espressif/esp-idf/blob/cf7e743a9b2e5fd2520be4ad047c8584188d54da/examples/bluetooth/nimble/power_save/main/Kconfig.projbuild#L37)switching between 240MHz and 80M. You can try to change the minimum CPU frequency to 40M, please let me know the test results.

Thanks

aircable, 2023-07-06:
Exactly the issue.
Thanks. This works:

```
esp_pm_config_t pm_config = {
    .max_freq_mhz = 240, // Wifi min is 80MHz
    // ESP32 does not support switching between 240MHz and 80M. 
    .min_freq_mhz = 40, // minimum is crystal frequency, 40MHz
    .light_sleep_enable = true, // deep sleep only
};
```

aircable, 2023-07-06:
issue resolved

closed 2023-07-06 by aircable