espressif/esp-idf #11796: Is it possible to trigger the I2S start-of-transmit synced with an internal or external signal?

NaivelyWritten, 2023-07-03:
### Is your feature request related to a problem?

Hello,

We have an application where we want to transmit a signal synced with a time base. Currently we are using I2S, but it seems that the I2S transmit function even with the new `preload` function has a non-deterministic delay between the time I enable and the time the actual waveform is transmitted.

### Describe the solution you'd like.

Can the I2S peripheral preload the data from DMA and start transmit immediately?

suda-morris, 2023-07-04:
Interesting! Would you please add more details about your application requirement? Why do you need such high real-time behavior?

Technically speaking, it's doable with the help of the [Event Task Matrix](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c6/api-reference/peripherals/etm.html) module. You can configure the GPIO or GPTimer to generate a signal, and the signal can be used to trigger the I2S to start transmitting. It's all done by the hardware, so it's deterministic.

However, this hardware feature currently only exists on ESP32-C6/H2/P4. And the software support from the I2S side is still in progress. If you can provide an application scenario and expected usage, that could help us to design a better API.

Currently, If you want to see the usage of the ETM, you can refer to [this example](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/timer_group/gptimer_capture_hc_sr04). The general initialization steps of the ETM are mostly similar across different peripherals.

NaivelyWritten, 2023-07-06:
Hi @suda-morris , thank you for the fast reply. Our application requires precise sync (~several nanosecond level) of a trigger signal (a GPTimer alert) and an I2S data stream, which currently in the ESP32 can only be done in software and has an accuracy of ~10us with the (shared) GPIO interrupt and the IDF driver. I will look into the ETM.

Alvin1Zhang, 2025-03-08:
@NaivelyWritten Thanks for reporting, would you please help share if any updates? Thanks.

ProfFan, 2025-03-09:
Hi @Alvin1Zhang thanks for the follow up, I am not actively working on that project anymore (left the company) but I'll see if I can use this functionality in my next project.

closed 2025-04-07 by Alvin1Zhang