espressif/esp-idf #11797: ESP32C6/WiFi 6 ITWT example stops after TWT setup attempt

emc-vishaal, 2023-07-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-C6-DevKitC-1

### Power Supply used.

USB

### What is the expected behavior?

Expect ESP32 to connect to WiFi 6 capable AP and negotiate at TWT setup (though I am not sure what the console output should look like...Readme.MD shows no expected results)

### What is the actual behavior?

Error 0x3018 - TWT Setup timeout

### Steps to reproduce.

1. Set the target using idf.py set-target esp32c6
2. build using idf.py build
3. flash using idf.py flash -p /dev/ttyUSB0 monitor
4. wait for example to run

### Debug Logs.

```plain
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x4086c410,len:0xd48
load:0x4086e610,len:0x2d68
load:0x40875720,len:0x17ec
entry 0x4086c410
[0;32mI (23) boot: ESP-IDF v5.1-dirty 2nd stage bootloader[0m
[0;32mI (24) boot: compile time Jul  4 2023 14:18:55[0m
[0;32mI (24) boot: chip revision: v0.0[0m
[0;32mI (27) boot.esp32c6: SPI Speed      : 40MHz[0m
[0;32mI (32) boot.esp32c6: SPI Mode       : DIO[0m
[0;32mI (36) boot.esp32c6: SPI Flash Size : 2MB[0m
[0;32mI (41) boot: Enabling RNG early entropy source...[0m
[0;32mI (47) boot: Partition Table:[0m
[0;32mI (50) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (57) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (72) boot:  2 factory          factory app      00 00 00010000 00100000[0m
[0;32mI (80) boot: End of partition table[0m
[... 188 lines cut ...]
I (18680) wifi:[ADDBA]TX addba request, tid:0, dialogtoken:1, bufsize:32, A-MSDU:0(not supported), policy:1(IMR), ssn:0(0x0)
I (18690) wifi:[ADDBA]TX addba request, tid:7, dialogtoken:2, bufsize:32, A-MSDU:0(not supported), policy:1(IMR), ssn:0(0x30)
I (18701) wifi:[ADDBA]TX addba request, tid:5, dialogtoken:3, bufsize:32, A-MSDU:0(not supported), policy:1(IMR), ssn:0(0x0)
I (18714) wifi:[ADDBA]RX addba response, status:0, tid:0/tb:1(0xa1), bufsize:32, batimeout:0, txa_wnd:32
I (18721) wifi:[ADDBA]RX addba response, status:0, tid:7/tb:1(0xa1), bufsize:32, batimeout:0, txa_wnd:32
I (18730) wifi:[ADDBA]RX addba response, status:0, tid:5/tb:1(0xa1), bufsize:32, batimeout:0, txa_wnd:32
W (18740) wifi:<ba-add>idx:0, ifx:0, tid:0, TAHI:0x100d4e9, TALO:0xa2e3c9c0, (ssn:0, win:64, cur_ssn:0), CONF:0xc0000005
I (18750) wifi:BcnInt:102400, DTIM:1
I (19756) esp_netif_handlers: sta ip: 192.168.100.228, mask: 255.255.255.0, gw: 192.168.100.1
I (24880) MAIN_DEBUG: 3018
E (24881) itwt_main: <WIFI_EVENT_ITWT_SETUP>twt_id:5, unexpected setup command:0
```

### More Information.

My main goal is to get this example running and integrate TWT/light sleep mode into a project where I want the ESP32 to sleep and wake up every few seconds and use TWT to grab whatever data the access point has waiting for it. 

I have a feeling this might have something to do with my access point or maybe some deeper settings in the ESP32...I'm still quite unfamiliar with this device.
ESP32 is connected via WIFI6 to the access point properly. Using DHCP instead of the static IP in the configuration. Everything else in sdkconfig is left on defaults.

The 0x3018 error is supposedly TWT setup timeout (consistently happens 5 sec after GOT_IP event) - I placed this print statement in the itwt setup handler function after it casts the event_data argument into type (wifi_event_sta_itwt_setup_t *)

I tried to look at the code in got_ip_handler but it seems like none of the source files lead to the source code for esp_wifi_sta_itwt_setup - I can only see the prototype defined in esp_wifi_he.h 

Any help is appreciated, thank you!

xuxiao111, 2023-07-04:
Hi @ emc-vishaal, the log shows that the ESP32C6 does not receive a TWT response from AP. Can you help to check whether the AP supports individual TWT?

emc-vishaal, 2023-07-05:
hi @xuxiao111 
I don't have the exact model number on hand but I have an enterprise grade TP-Link WIFI 6 access point...I'd assume these ones have TWT capability (is there a difference between TWT and individual TWT and are only a select number of router models equipped with the individual version?)

xuxiao111, 2023-07-05:
Hi @emc-vishaal , TWT includes individual TWT and broadcast TWT, and now we only support individual TWT. Also, you can use Wireshark to capture the beacon frame sent by AP and check the HE capabilities to confirm the AP supports individual TWT.
[image]

emc-vishaal, 2023-07-05:
Hi @xuxiao111 
I've installed Wireshark but am not able to see anything similar to your window...did you have to create filters or monitor a specific IP address? I see nothing when I set a capture filter to the IP address of the ESP32

xuxiao111, 2023-07-05:
Hi @emc-vishaal, you can follow this wiki to capture wireless frames. https://wiki.wireshark.org/CaptureSetup/WLAN. Or can you show the detailed info about the TP-Link access point that you used?

emc-vishaal, 2023-07-06:
Hi @xuxiao111 

It appears our AP does support ITWT according to the beacon frame read in wireshark:
[image]

xuxiao111, 2023-07-06:
Hi @emc-vishaal, can you help to capture the itwt setup response frame sent by AP?
[image]

emc-vishaal, 2023-07-06:
Hi @xuxiao111 Looks like neither of these messages exist for me. 
The call after getting an IP to esp_wifi_sta_itwt_setup returns ESP_OK so I'm assuming the ESP32 is sending the TWT Setup message to the AP but I don't see anything with the name "espressif" on the wireshark interface...no entries with the ESP32 in either source or destination.

I wanted to make sure the board is actually connecting to wifi so I ran another example and verified I'm able to send/receive MQTT messages but even though I am able to see MQTT data, I see nothing in wireshark.

xuxiao111, 2023-07-06:
Hi @emc-vishaal, can you find the twt setup response frame from AP? If you can't find it, maybe this is an AP firmware issue, you can update the AP firmware to check whether it's ok in the new firmware version or change to another AP which supports ITWT. (ASUS-AX56U,ASUS-AX86U,ASUS-88U, Linksys-MX4200, Linksysy-MX5300)

emc-vishaal, 2023-07-06:
Hi @xuxiao111 , it appears there is no response frame from the AP. Filtering by frames going to and from the MAC address of the ESP32, the only packets seen are some that are being sent by ESP32 and then at times ACK frames that don't have a source address.
[image]

xuxiao111, 2023-07-06:
[image]
Wireshark version:
[image]

emc-vishaal, 2023-07-07:
Thanks @xuxiao111 
I was on a different version of wireshark, I get something similar to you now. I see a difference in our request packet (from ESP32). What is the exponent you are using in your configuration? I am using exponent = 10 and mantissa=512. Any idea why my wakeup time number is so high? doesn't match up to 512*2^10

Screenshot of ESP32 to AP request frame
[image]

Also have confirmed my AP [TP-LINK OMADA Series EAP610] is updated to latest firmware and according to our previous wireshark screenshots should be able to support ITWT

xuxiao111, 2023-07-07:
Hi @emc-vishaal, the default wakeup time is about 255*256 = 65280 (us), and the wake interval is about 512 * 2^10 = 524288 (us). I think these two times are OK.

emc-vishaal, 2023-07-07:
Hi @xuxiao111 

Right, so the 1784062566400 number next to Target Wake Time: in the frame makes me think something is going wrong with the conversion of the mantissa and exponent from the configuration to whatever is coming out on the frameover wifi...is there anything else you can think of that's wrong with the TWT setup request? I can't access any code past the call to esp_wifi_sta_itwt_setup

xuxiao111, 2023-07-07:
Hi @emc-vishaal, after calling the esp_wifi_sta_itwt_setup, the STA will send an ITWT setup frame, and start a timer to trigger the ITWT setup response frame sent from AP. If the AP can not send the setup response frame during 5s, STA will post an ITWT setup failed event. I don't think anything wrong with the TWT parameters, and if we set the wrong TWT parameters, AP also needs to send us the TWT response frame.

emc-vishaal, 2023-07-07:
Hi @xuxiao111 

Hmm...is there anything to be explored on the issue that I can call the esp_wifi_sta_itwt_setup hundreds of times (I just leave my code running - call it every couple seconds after getting the timeout error) and only see the frame on wireshark from the ESP32 sent out once, maybe twice over a period of >30mins? Or is this also possibly an AP issue? 

When running other examples like MQTT, wireshark never seems to miss a frame that the ESP32 sends.

Is there a list of APs that have been tested specifically with the ESP32C6 in TWT mode? Or at least maybe just the one you're using? Might be a good sanity check for me to buy one though I'd be very surprised if an enterprise level WIFI 6 AP with the latest firmware can't respond to these commands...

xuxiao111, 2023-07-07:
Hi @emc-vishaal
AP list:
1. AUSU: RT-AX56U
2. AUSU: RT-AX86U
3. AUSU: RT-AX88U
4. Linksys: MX5300
5. Linksys: MX4200
We have tested the above APs and can support ITWT. Normally, every time you call this function, there should be an ITWT setup frame sent from ESP32C6.
And also, we have tested twt setup function with RT-AX56U, and it works fine.
[image]
branch: release/v5.1 (71a5663786)

igrr, 2023-07-07:
Another router that is known to work is TP-Link Archer AX73.

emc-vishaal, 2023-07-12:
Thanks @xuxiao111 @igrr 
I've decided to go another direction on this project temporarily and not use the ESP32...inconsistent TWT behavior between APs when setting up TWT turns the whole thing into a a no-go at this point but I'll likely revisit this in the near future. I'll close this thread for now

closed 2023-07-12 by emc-vishaal

VitorBFreitas, 2024-01-11:
Sorry reopen this issue, but i'm with a project that one option is implement ESP32-C6 with TWT. Is there any other router, besides the ones listed above in this issue, known to support iTWT?
@igrr @xuxiao111

VitorBFreitas, 2024-01-25:
@xuxiao111 I got a TP-Link Archer AX73, and itwt example now works. Thank you.

However I figured that any combination of mantissa and exponent, to set iTWT interval, that exceeds 0x7FFFFFFF i got:
```
W (3269) wifi:(itwt)rx response, unexpected setup command:7(reject)
E (3269) itwt: <WIFI_EVENT_ITWT_SETUP>twt_id:0, twt setup request was rejected, setup cmd: 7
```

any value under 0x7FFFFFFF, but that exceeds 0x80000, turns into a combination of a mantissa with exponent 13 that have the same value as set.

if above condition is met, but the value is close to 0x20000000 i got:
```
E (2650) wifi:(itwt)recv invalid iTWT parameters SP: 32768 < WD: 65280
E (2651) itwt: <WIFI_EVENT_ITWT_SETUP>twt_id:0, twt setup failed, status: 258
```

and if value is equal 0x20000000, the router crashes.

xuxiao111, 2024-02-07:
Hi @VitorBFreitas,
I'm sorry for not getting back to you sooner. Can you help check if the fields in the TWT action replied to by the AP are correct by capturing packets?

VitorBFreitas, 2024-02-29:
Hello, @xuxiao111.
Can you provide me some help here?

xuxiao111, 2024-02-29:
Hi @VitorBFreitas,
The beacon frame does not carry TWT setup related parameters, just the flag to tell the station, the AP supports TWT.

xuxiao111, 2024-02-29:
We need to capture the TWT setup response frame sent by AP to check whether the AP response is correct. From the log you print, it seems the AP response has incorrect parameters.

VitorBFreitas, 2024-02-29:
> We need to capture the TWT setup response frame sent by AP to check whether the AP response is correct. From the log you print, it seems the AP response has incorrect parameters.

Can you point me what should I look for in the print that shows that the response has incorrect parameters?

xuxiao111, 2024-02-29:
From this "E (2650) wifi:(itwt)recv invalid iTWT parameters SP: 32768 < WD: 65280".

looks like the TWT cycle time is 32.768ms, but the wakeup time is 65.280ms. It's incorrect.

And for this "any value under 0x7FFFFFFF, but that exceeds 0x80000, turns into a combination of a mantissa with exponent 13 that have the same value as set.". The actual TWT parameters used are obtained from the TWT setup response frame sent from the AP.

VitorBFreitas, 2024-02-29:
> From this "E (2650) wifi:(itwt)recv invalid iTWT parameters SP: 32768 < WD: 65280".
>
> looks like the TWT cycle time is 32.768ms, but the wakeup time is 65.280ms. It's incorrect.
>
> And for this "any [...]

Got it. So should I assume that the TP-Link Archer AX73 is the problem, right?

xuxiao111, 2024-03-01:
yeah, so I want to know the TWT setup parameters sent from AP to double-check.

VitorBFreitas, 2024-03-05:
@xuxiao111 Sounds that you are right. Here is the setup parameters sent to ap and from ap.
[image]

[image]