espressif/esp-idf #11800: Memory leak in lwIP when ending TCP-connections

pggh, 2023-07-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.0

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3

### Power Supply used.

External 3.3V

### What is the expected behavior?

No memory leak.

### What is the actual behavior?

If created netconn-based servers (http and others) and experienced a memory leak for each connection.
Further analysis showed that the memory leaked is memory allocated by the function tcp_alloc (172 bytes) and never freed.
This also happens when using the socket API. I don't know the exact timing and internal states of the connection that must be met to show the leak but in my http scenario it happens for every connection.

Note: I've only tested with IDF 5.0 because an offline installer for 5.1 is not yet available.

This issue might be similar or identical to [#9670](https://github.com/espressif/esp-idf/issues/9670) but shows it in action using a very common use case and has a reproducer.

### Steps to reproduce.

Once the circumstances are known, it pretty easy to reproduce:
1. Use the tcp_server example from the IDF SDK
2. Change `} while (len > 0);` in do_retransmit to `} while (0);`
3. Add the following diagnostic code after `close(sock);`:
```
        close(sock);

        printf("heep free: %d\n", heap_caps_get_free_size(MALLOC_CAP_DEFAULT));
        {
            void* m = malloc(65536);
            if (m == NULL) {
                printf("!!! MALLOC FAILED !!!\n");
            } else {
                free(m);
            }
        }
```
4. use ´curl http://ip_of_esp/´ in a loop (I did it on windows) and after about 1200 requests, the heap memory is exhausted and you get the MALLOC FAILED messages.

The sdkconfig.defaults used to build this reproducer was:
```
CONFIG_BOOTLOADER_CUSTOM_RESERVE_RTC=y
CONFIG_BOOTLOADER_CUSTOM_RESERVE_RTC_SIZE=0x10
CONFIG_ESPTOOLPY_FLASHMODE_QIO=y
CONFIG_ESPTOOLPY_FLASHSIZE_4MB=y
CONFIG_PARTITION_TABLE_CUSTOM=y
CONFIG_ESPTOOLPY_BEFORE_RESET_USB=y
# CONFIG_ETH_USE_SPI_ETHERNET is not set
# CONFIG_ESP_PHY_CALIBRATION_AND_DATA_STORAGE is not set
# CONFIG_ESP_PHY_REDUCE_TX_POWER is not set
CONFIG_ESP_DEFAULT_CPU_FREQ_MHZ_80=y
CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG=y
CONFIG_ESP_BROWNOUT_DET_LVL_SEL_3=y
# CONFIG_ESP32_WIFI_NVS_ENABLED is not set
CONFIG_FREERTOS_UNICORE=y
CONFIG_FREERTOS_TIMER_TASK_PRIORITY=8
CONFIG_FREERTOS_USE_TRACE_FACILITY=y
CONFIG_FREERTOS_USE_STATS_FORMATTING_FUNCTIONS=y
CONFIG_FREERTOS_VTASKLIST_INCLUDE_COREID=y
CONFIG_FREERTOS_PLACE_SNAPSHOT_FUNS_INTO_FLASH=y
# CONFIG_LOG_COLORS is not set
CONFIG_LWIP_SO_RCVBUF=y
CONFIG_LWIP_NETBUF_RECVINFO=y
# CONFIG_LWIP_DHCP_DOES_ARP_CHECK is not set
# CONFIG_LWIP_IPV6 is not set
CONFIG_LWIP_TCP_TMR_INTERVAL=30
CONFIG_LWIP_TCP_SND_BUF_DEFAULT=5760
CONFIG_LWIP_TCP_WND_DEFAULT=5760
# CONFIG_SPI_FLASH_YIELD_DURING_ERASE is not set
```

### More Information.

It's still a mystery to me, what exact sequence of call to the netconn-API is needed to distinguish between a graceful-close (pending data is sent, followed by the FIN-sequence) and an abortif-close (pending data discarded, RST send to peer)
This is especially true when CONFIG_LWIP_SO_LINGER is not enabled. But even with it enabled, I'm not sure setting linger prior to close/delete does what I expect.
It would be very helpful if an expert could clarify this.

david-cermak, 2023-10-31:
Hi @pggh 

Yes, there's some one time allocation for each socket you open. One option is to open and close all system sockets before looking into memory leaks. This is what we do in IDF test code:

https://github.com/espressif/esp-idf/blob/5b39159e668d3af5832cedb500ccc4869bf15492/tools/unit-test-app/components/test_utils/test_utils.c#L32-L41

Plus there're sockets in those wait states (`WAIT_CLOSE`, `TIME_WAIT` and `FIN_WAIT_2`) that should be recycled. Applying lingering option would also help to abruptly close connections.
You can adjust the wait states timeouts with `CONFIG_LWIP_TCP_FIN_WAIT_TIMEOUT`, and maybe reducing maximum segment lifetime in `CONFIG_LWIP_TCP_MSL` would help recycling the closed sockets faster.

david-cermak, 2023-11-28:
Closing as answered. Feel free to reopen.

closed 2023-11-28 by david-cermak