espressif/esp-idf #11802: Question about flash encryption ESP32...

butaikis, 2023-07-04:
Hello. I have a question about flash encryption. At the moment, I upload my encryption key to efuse, enable encryption, encrypt the firmware with it, upload the firmware to flash, everything works.

Tell me, why can't you first upload your bins (already encrypted), and then upload the key and enable encryption? I'm getting an `invalid header` error.

mahavirj, 2023-07-05:
@butaikis 

What steps you followed? Can you please share all logs?

Recently we have added a guide [enable-flash-encryption-externally](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/host-based-security-workflows.html#enable-flash-encryption-externally), that may help you for this case.

butaikis, 2023-07-05:
@mahavirj Hello!
I have a script
1. key encrypted upload efuse
2. enabling flash encryption efuse
3. loading my bins (encrypted)

Is my .bat script

```
:: Efuses set
set path_espefuse=D:\Develop_VS_Code\.platformio\packages\tool-esptoolpy\espefuse.py
set key_enc=D:\Develop_VS_Code\project\_bin\flash_encryption_key.bin
:: Write key encryption
py %path_espefuse% --port COM9 --do-not-confirm burn_key flash_encryption %key_enc%
:: Set security efuses
set efuse_security= FLASH_CRYPT_CNT 1 FLASH_CRYPT_CONFIG 0xf JTAG_DISABLE 1 DISABLE_DL_ENCRYPT 1 DISABLE_DL_DECRYPT 1 DISABLE_DL_CACHE 1
py %path_espefuse% --port COM9 --do-not-confirm --before default_reset burn_efuse %efuse_security% 
:: Set protect write/read mask (read only all efuse)
:: py %path_espefuse% --port COM9 --do-not-confirm write_protect_efuse WR_DIS RD_DIS

::Upload bin
set path_esptool=D:\Develop_VS_Code\.platformio\packages\tool-esptoolpy\esptool.py
set path_input=D:\Develop_VS_Code\project\_bin\production\bin\
set path_fw=D:\Develop_VS_Code\project\_bin\production\fw_1.12\enc_firmware.bin
set path_webui=D:\Develop_VS_Code\project\_bin\production\webui_1.12\storage_webui.bin
set chip=esp32
set baud=921600
set fmode=keep
set ffreq=keep
py %path_esptool% --port COM9 --chip %chip% --baud %baud% write_flash --force 0x1000 %path_input%enc_bootloader.bin 0xD000 %path_input%enc_partitions.bin 0x15000 %path_input%enc_ota_data_initial.bin 0x20000 %path_fw% 0x5FE000 %path_input%storage_user.bin 0x617000 %path_webui%
```

Everything works as it should.

I became interested. Why the firmware does not work if you do this.
1. loading my bins (encrypted)
2. key encrypted upload efuse
2. enabling flash encryption efuse

Example

```
::Upload bin
set path_esptool=D:\Develop_VS_Code\.platformio\packages\tool-esptoolpy\esptool.py
set path_input=D:\Develop_VS_Code\project\_bin\production\bin\
set path_fw=D:\Develop_VS_Code\project\_bin\production\fw_1.12\enc_firmware.bin
set path_webui=D:\Develop_VS_Code\project\_bin\production\webui_1.12\storage_webui.bin
set chip=esp32
set baud=921600
set fmode=keep
set ffreq=keep
py %path_esptool% --port COM9 --chip %chip% --baud %baud% write_flash --force 0x1000 %path_input%enc_bootloader.bin 0xD000 %path_input%enc_partitions.bin 0x15000 %path_input%enc_ota_data_initial.bin 0x20000 %path_fw% 0x5FE000 %path_input%storage_user.bin 0x617000 %path_webui%
:: Efuses set
set path_espefuse=D:\Develop_VS_Code\.platformio\packages\tool-esptoolpy\espefuse.py
set key_enc=D:\Develop_VS_Code\project\_bin\flash_encryption_key.bin
:: Write key encryption
py %path_espefuse% --port COM9 --do-not-confirm burn_key flash_encryption %key_enc%
:: Set security efuses
set efuse_security= FLASH_CRYPT_CNT 1 FLASH_CRYPT_CONFIG 0xf JTAG_DISABLE 1 DISABLE_DL_ENCRYPT 1 DISABLE_DL_DECRYPT 1 DISABLE_DL_CACHE 1
py %path_espefuse% --port COM9 --do-not-confirm --before default_reset burn_efuse %efuse_security% 
:: Set protect write/read mask (read only all efuse)
:: py %path_espefuse% --port COM9 --do-not-confirm write_protect_efuse WR_DIS RD_DIS
```

AdityaHPatwardhan, 2023-07-06:
Hi @butaikis Ideally this should not happen, The contents of esp32 at the end of both scripts are supposed be the same.

Can you please provide following information that would give us more insights.

1) Read the bootloader from the flash of the device used in the 2nd process (where you are getting the invalid header error) and compare it with encrypted binaries available for the bootloader i.e. `enc_bootloader.bin`. They should be an exact match.

2) flash the encrypted binaries again on the device where you are getting an invalid header error and check if it works. IMO it should work.

butaikis, 2023-07-06:
@AdityaHPatwardhan 
Thank you! Indeed, the bootloader is different, apparently in the second case I used ESP32 Flash Download tools and **DoNotChgBin** was not installed there. Because of this, it made changes to my encrypted bootloader. Checked everything, it works!

AdityaHPatwardhan, 2023-07-06:
@butaikis Thanks for confirming

I am not much familiar with the Flash Download tools but it looks like if **DoNotChgBin** is not enabled then bootloader headers are updated and the bootloader binary is changed.
That must have happened in your case.
I am closing the issue since you have confirmed it as resolved.
If there is anything else, please feel free to re-open the issue.

Thanks,
Aditya

closed 2023-07-06 by AdityaHPatwardhan