espressif/esp-idf #11803: ESP_LOGx affects the Priority of the Task

CorsairCat, 2023-07-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Is it default that ESP_LOGx will change the task priority? I'm currently working no a speed sensitive project and it appears that  include the ESP_LOGx will make the task using more cpu time, which is similar to increase the priority? is it what it supposed to be?

The platform I use is ESP32S3

ESP-Marius, 2023-07-05:
Logging will not change the task priority, but it is an expensive operation (both in terms of time and stack) so it will increase the time it takes for the task to perform its actions yes. Which in turns means a task will take more of the total CPU time.

CorsairCat, 2023-07-05:
> Logging will not change the task priority, but it is an expensive operation (both in terms of time and stack) so it will increase the time it takes for the task to perform its actions yes. Which in turns means a task will take more of the total CPU time.

Yes, I mean, I'm using same code, and trying to capture a 1000hz response. however, with the same code, if I use it without ```ESP_LOGW(TAG, "A");``` included, it can only capture 500hz, but it do capture 1000hz if I have that line included. That's why I m confused.

ESP-Marius, 2023-07-05:
Ah, I see.

Hard to give an answer without seeing the full picture, but in short no, logging should not make your thread have higher priority.
It is probably some other side-effect that is causing this. 

Could be that with logging some of your threads will yield more often, causing your capture thread to be less resource starved? Or the extra code causes a difference in flash code layout, causing less cache misses for your critical code path?

CorsairCat, 2023-07-05:
> Ah, I see.
>
> Hard to give an answer without seeing the full picture, but in short no, logging should not make your thread have higher priority. It is probably some other side-effect that is causing [...]

Yep, I supposed its a side effect. It do appears that include a log function call will make this thread take more time. It confused me for a few days, I tried to change their priority first, then analysis the cache stuff. I m  trying to find out why this happened. I just saw some docs said the thread manage system of esp-freertos is different from original freertos, I'm planning to look into that and see why

CorsairCat, 2023-07-06:
I did a deep analysis and figure out where this side effects comes from. I have another task using a queue to interact with this high frequency task. and that LOG command slow down the queue request and therefore reduce the load from the high frequency task.

ESP-Marius, 2023-07-06:
Great, sounds like there is no IDF related bug or anything then.

I'll mark the issue as closed.

closed 2023-07-06 by ESP-Marius