espressif/esp-idf #11804: esp_ble_gap_config_local_privacy doesn't set random mac

jeremyherbert, 2023-07-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1

### Operating System used.

Linux

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-C3-WROOM-02

### Power Supply used.

External 3.3V

### What is the expected behavior?

I run the following code:

```
    static esp_ble_adv_params_t adv_params = {
            .adv_int_min        = 0x20,
            .adv_int_max        = 0x30,
            .adv_type           = ADV_TYPE_IND,
            .own_addr_type      = BLE_ADDR_TYPE_RANDOM,
            .channel_map        = ADV_CHNL_ALL,
            .adv_filter_policy  = ADV_FILTER_ALLOW_SCAN_ANY_CON_ANY,
    };
    esp_ble_gap_config_local_privacy(true);
    ESP_LOGI(TAG, "Advertising started");
    return esp_ble_gap_start_advertising(&adv_params);
```

And I expect advertising to start with a random address.

### What is the actual behavior?

Advertising does not start, and I receive the following error:

```
E (485) BT_BTM: No random address yet, please set random address and try
E (495) BT_APPL: bta_dm_ble_set_adv_params_all(), fail to set ble adv params.
```

### Steps to reproduce.

1. Open `esp_hid_device` example
2. Change `esp_hid_ble_gap_adv_start` to have the code as above
3. Build + flash + run

### More Information.

related: https://github.com/espressif/esp-idf/issues/4574 but this seems to be a regression.

CorsairCat, 2023-07-05:
did you try using this line to create a new Random address?
```
esp_bd_addr_t rand_addr = {0xc3,0x91,0x91,0x26,0xf4,0xc3}; // you can generate this randomly instead of hard-coded 
esp_ble_gap_set_rand_addr(rand_addr);
```
it works on my code.
There is a limitation on the address, both of the highest two bit should be 1, and I'm not sure which side is highest so I set both side to be 0x3c

jeremyherbert, 2023-07-05:
Yes, but I think this is a separate issue - enabling privacy mode used to set a random address automatically, and now it does not. Also, it shouldn't be possible to enable privacy mode with a non-random address.

jeremyherbert, 2023-07-05:
Thanks @zhp0406 but isn't it a requirement for LE privacy to have a random address? so ESP-IDF allows you to enable LE privacy but not with a random address?

CorsairCat, 2023-07-05:
Technically yes? For example now on iOS, it will kept privacy MAC address for same Wi-Fi, so it should allow you to use the address you prefer?

zhp0406, 2023-07-05:
1-The random device address can be one of the following types. The details of these three types of random addresses are introduced in "BLUETOOTH CORE SPECIFICATION Version 5.3 | Vol 6, Part B."

[image]

2-it is not a requirement for LE privacy to have a random address. and ESP-IDF allows enable LE privacy but not with a random address.

3-The API `esp_ble_gap_config_local_privacy(true)` is used to enable the resolution of Resolvable Private Addresses in the Controller. This allows the Controller to utilize the resolving list whenever it receives a local or peer Resolvable Private Address.

4-When the `own_addr_type` is set as BLE_ADDR_TYPE_RPA_PUBLIC, a Resolvable Private Address will be automatically generated. Therefore, if you set `own_addr_type` as BLE_ADDR_TYPE_RPA_PUBLIC, the RPA address will be used directly.

5-static random addresses are generated using the API `esp_ble_gap_set_rand_addr`,and should be generate manually.

jeremyherbert, 2023-07-05:
Thank you for the explanation. I appreciate your help.

closed 2023-07-05 by jeremyherbert