espressif/esp-idf #11806: esp_hid_device bluetooth example doesn't work properly with input passkey

jeremyherbert, 2023-07-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-C3-WROOM-02

### Power Supply used.

External 3.3V

### What is the expected behavior?

After modifying the example to accept a passkey (entered over SPI) I expect the HID device to work immediately after pairing.

### What is the actual behavior?

The HID device does not work, even though pairing was successful. 

### Steps to reproduce.

1. Download, build and flash the attached code to an ESP32-C3, which is a modified version of the sample that does the following:
* sets the IO capability to input only
* enables the passkey to be sent to the device using the spi slave driver. You transmit "p123456" over spi and then it calls `esp_ble_passkey_reply(passkey_auth_target, true, passkey);`

3. Try to connect to the device (I am using mac os 12.4) and you should get a pairing code, send it over SPI
4. Authentication succeeds with `ESP_GAP_BLE_AUTH_CMPL_EVT` being triggered and printed to UART log. Device is listed as connected on the PC, but the event `ESP_HIDD_CONNECT_EVENT` is never sent to the the hidd callback.
5. Power cycle the esp32-c3
6. PC doesn't automatically connect to the device. Double clicking on the device in the bluetooth preferences does connect to the device, and it doesn't ask to pair again. `ESP_HIDD_CONNECT_EVENT` is now triggered and HID messages are finally able to be sent
7. After power cycling again, device connects to the PC automatically and is able to send HID messages.

[esp32-c3-pairing.zip](https://github.com/espressif/esp-idf/files/11955516/esp32-c3-pairing.zip)

### Debug Logs.

```plain
first connection (pairing, but fails to start HID):

ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0x1704
load:0x403cc710,len:0x968
load:0x403ce710,len:0x2f68
entry 0x403cc710
I (30) boot: ESP-IDF v5.1 2nd stage bootloader
I (30) boot: compile time Jul  5 2023 17:52:58
I (30) boot: chip revision: v0.3
I (33) boot.esp32c3: SPI Speed      : 80MHz
[... 229 lines cut ...]
I (384) app_start: Starting scheduler on CPU0
I (389) main_task: Started on CPU0
I (389) main_task: Calling app_main()
I (399) HID_DEV_DEMO: setting hid gap, mode:1
I (399) ESP_HID_GAP: Starting SPI transaction
I (399) BLE_INIT: BT controller compile version [85b425c]
I (409) phy_init: phy_version 970,1856f88,May 10 2023,17:44:12
I (459) BLE_INIT: Bluetooth MAC: 60:55:f9:83:ed:c2
W (479) BT_BTM: BTM_BleWriteAdvData, Partial data write into ADV
I (479) HID_DEV_DEMO: setting ble device
I (489) HID_DEV_BLE: START
I (489) main_task: Returned from app_main()
I (649) ESP_HID_GAP: BLE GAP AUTH SUCCESS
I (649) HID_DEV_BLE: CONNECT
I (649) HID_DEV_DEMO: Send volume key
W (819) BT_L2CAP: le con upd: err_stat=0x1e
I (2749) HID_DEV_DEMO: Send volume key
I (4849) HID_DEV_DEMO: Send volume key
I (6949) HID_DEV_DEMO: Send volume key
I (9049) HID_DEV_DEMO: Send volume key
```

### More Information.

I tried to connect to the device on Ubuntu 22.04 over BLE, but it doesn't even show a pairing code and just works as-is

jeremyherbert, 2023-07-05:
This seems to be something to do with the way the controller is interrogating the GATT server. When I first connect to the device, System Report shows:

```
  ESP BLE HID2:
  Address:	60:55:F9:83:ED:C2
  Vendor ID:	0x16C0
  Product ID:	0x05DF
  Services:	0x400000 < BLE >
```

But after power cycling and manually reconnecting (note that the battery level is now visible):
```
  ESP BLE HID2:
  Address:	60:55:F9:83:ED:C2
  Vendor ID:	0x16C0
  Product ID:	0x05DF
  Battery Level:	100%
  Services:	0x400000 < BLE >
```

So it seems that the GATT interrogation is failing somewhere

zhp0406, 2023-07-05:
Got it, I will check it later.

jeremyherbert, 2023-07-05:
I tried to collect some logs with more debugging enabled (only after the auth is complete):

This is the first connection:

```
I (123317) ESP_HID_GAP: BLE GAP AUTH SUCCESS
D (124337) BT_GATT: gatt_get_ch_state: ch_state=4
D (124337) BT_GATT: gatts_process_read_by_type_req, op_code =8, len = 0
D (124337) BT_GATT: read_attr_value uuid=0x2a00 perm=0x1 sec_flag=0x34 offset=0 read_long=0
D (124347) BT_GAP: gap_ble_s_attr_request_cback : recv type (0x01)
D (124357) BT_GAP: GATT_UUID_GAP_DEVICE_NAME len=0x000c
I (124357) BT_GATT: GATTS_SendRsp: conn_id: 2  trans_id: 11  Status: 0x0000
D (124367) BT_GATT: gatt_get_ch_state: ch_state=4
D (124377) BT_GATT: gatt_sr_process_app_rsp gatt_if=2
D (124377) BT_GATT: gatt_dequeue_sr_cmd
D (124377) BT_GATT: gatt_sr_process_app_rsp ret_code=0
D (124437) BT_GATT: gatt_get_ch_state: ch_state=4
D (124437) BT_GATT: gatts_process_read_by_type_req, op_code =8, len = 0
D (124437) BT_GATT: read_attr_value uuid=0x2803 perm=0x1 sec_flag=0x34 offset=0 read_long=0
D (124447) BT_GATT: gatt_dequeue_sr_cmd
D (124467) BT_GATT: gatt_get_ch_state: ch_state=4
D (124467) BT_GATT: gatts_process_read_by_type_req, op_code =8, len = 0
D (124487) BT_GATT: gatt_get_ch_state: ch_state=4
D (124507) BT_GATT: gatt_get_ch_state: ch_state=4
D (124507) BT_GATT: gatts_write_attr_perm_check op_code=0x12 handle=0x002b offset=0 len=2 sec_flag=0x34 key_size=16
D (124507) BT_GATT: gatts_write_attr_perm_check p_attr->permission =0x0011 min_key_size==0x0000
I (124517) BT_GATT: GATT_GetConnectionInfor conn_id=3
D (124527) BT_GATT: gatt_dequeue_sr_cmd
V (124527) BLE_HIDD: Battery CCC: Notify: On, Indicate: Off
D (124537) BT_GATT: gatt_get_ch_state: ch_state=4
D (124537) BT_GATT: read_attr_value uuid=0x2a19 perm=0x1 sec_flag=0x34 offset=0 read_long=0
I (124547) HID_DEV_DEMO: Send Volume
D (124557) BT_GATT: gatt_get_ch_state: ch_state=4
I (124557) BT_GATT: GATT_GetConnectionInfor conn_id=3
D (124567) BT_GATT: gatt_dequeue_sr_cmd
I (124567) BT_GATT: GATT_GetConnectionInfor conn_id=5
I (124577) BT_GATT: GATTS_HandleValueNotification
D (124577) BT_GATT: gatt_get_ch_state: ch_state=4
D (124587) BLE_HIDD: Battery READ 100
V (124587) BLE_HIDD: HID CONF[0] status: 0x00, len: 0
D (124637) BT_GATT: gatt_get_ch_state: ch_state=4
I (124637) BT_GATT: GATT_GetConnectionInfor conn_id=5
I (124637) BT_GATT: GATTS_HandleValueNotification
D (124637) BT_GATT: gatt_get_ch_state: ch_state=4
V (124647) BLE_HIDD: HID CONF[0] status: 0x00, len: 0
I (126647) HID_DEV_DEMO: Send Volume
D (126647) BT_GATT: gatt_get_ch_state: ch_state=4
I (126647) BT_GATT: GATT_GetConnectionInfor conn_id=5
I (126647) BT_GATT: GATTS_HandleValueNotification
D (126647) BT_GATT: gatt_get_ch_state: ch_state=4
V (126657) BLE_HIDD: HID CONF[0] status: 0x00, len: 0
D (126707) BT_GATT: gatt_get_ch_state: ch_state=4
I (126707) BT_GATT: GATT_GetConnectionInfor conn_id=5
I (126707) BT_GATT: GATTS_HandleValueNotification
D (126707) BT_GATT: gatt_get_ch_state: ch_state=4
V (126717) BLE_HIDD: HID CONF[0] status: 0x00, len: 0
I (128717) HID_DEV_DEMO: Send Volume
D (128717) BT_GATT: gatt_get_ch_state: ch_state=4
I (128717) BT_GATT: GATT_GetConnectionInfor conn_id=5
I (128717) BT_GATT: GATTS_HandleValueNotification
D (128717) BT_GATT: gatt_get_ch_state: ch_state=4
V (128727) BLE_HIDD: HID CONF[0] status: 0x00, len: 0
D (128777) BT_GATT: gatt_get_ch_state: ch_state=4
I (128777) BT_GATT: GATT_GetConnectionInfor conn_id=5
I (128777) BT_GATT: GATTS_HandleValueNotification
D (128777) BT_GATT: gatt_get_ch_state: ch_state=4
V (128787) BLE_HIDD: HID CONF[0] status: 0x00, len: 0
I (130787) HID_DEV_DEMO: Send Volume
D (130787) BT_GATT: gatt_get_ch_state: ch_state=4
I (130787) BT_GATT: GATT_GetConnectionInfor conn_id=5
[... 304 lines cut ...]
D (149477) BT_GATT: gatt_get_ch_state: ch_state=4
V (149487) BLE_HIDD: HID CONF[0] status: 0x00, len: 0
I (151487) HID_DEV_DEMO: Send Volume
D (151487) BT_GATT: gatt_get_ch_state: ch_state=4
I (151487) BT_GATT: GATT_GetConnectionInfor conn_id=5
I (151487) BT_GATT: GATTS_HandleValueNotification
D (151487) BT_GATT: gatt_get_ch_state: ch_state=4
V (151497) BLE_HIDD: HID CONF[0] status: 0x00, len: 0
D (151547) BT_GATT: gatt_get_ch_state: ch_state=4
I (151547) BT_GATT: GATT_GetConnectionInfor conn_id=5
I (151547) BT_GATT: GATTS_HandleValueNotification
D (151547) BT_GATT: gatt_get_ch_state: ch_state=4
V (151557) BLE_HIDD: HID CONF[0] status: 0x00, len: 0
I (153557) HID_DEV_DEMO: Send Volume
D (153557) BT_GATT: gatt_get_ch_state: ch_state=4
I (153557) BT_GATT: GATT_GetConnectionInfor conn_id=5
I (153557) BT_GATT: GATTS_HandleValueNotification
D (153557) BT_GATT: gatt_get_ch_state: ch_state=4
V (153567) BLE_HIDD: HID CONF[0] status: 0x00, len: 0
D (153617) BT_GATT: gatt_get_ch_state: ch_state=4
I (153617) BT_GATT: GATT_GetConnectionInfor conn_id=5
I (153617) BT_GATT: GATTS_HandleValueNotification
D (153617) BT_GATT: gatt_get_ch_state: ch_state=4
V (153627) BLE_HIDD: HID CONF[0] status: 0x00, len: 0
```

jeremyherbert, 2023-07-05:
I think the issue is that the code in `ble_hidd.c` assumes that when the GATTS is connected that GAP authentication is already complete, but when you are using a passkey the GATTS connection (ie `ESP_GATTS_CONNECT_EVT`) happens before GAP authentication is finished.

jeremyherbert, 2023-07-06:
This seems to be to do with encryption not being enabled on the link. I can see the following error in the mac os console:

```
default	11:18:35.096997+1000	BluetoothUIServer	LEHIDConnectionComplete - VID:0x16C0 PID:0x05DF “ESP BLE HID2”: no custom info found
error	11:18:35.207520+1000	BTLEServer	Not creating HID device as the link is not encrypted
```

jeremyherbert, 2023-07-06:
I have found a workaround (or maybe even a fix), if you edit `ble_hidd.c` in the `esp_hid` component and set all of the GATTS permissions on all of the services to `ESP_GATT_PERM_READ_ENCRYPTED` or `ESP_GATT_PERM_READ_ENCRYPTED | ESP_GATT_PERM_WRITE_ENCRYPTED` as appropriate, then the device is able to connect properly the first time, and automatic connection is working correctly through power cycles. 

Since the HID over BLE standard requires encryption anyway, I don't see a problem with doing this.

jeremyherbert, 2023-07-19:
So the real solution is that both iOS and Mac require the report map characteristic and the report reference descriptor to have their permissions set as encrypted, otherwise when the initial connection occurs the device is considered not encrypted and the HID reports are not parsed and enumerated. This isn’t documented anywhere, but it is the way that they behave. The problem only appears when pairing OOB or with passkey input on the device. The battery and device info service do not need to be encrypted. On the second connection where pairing and bonding is already complete, the check is skipped because the connection is encrypted before the GATT operations.

The characteristics probably should be set as encrypted anyway, as this follows the spirit of the BLE HOGP spec requiring all HID data to be transferred over an encrypted connection.

zhp0406, 2023-11-08:
In the latest version（by commit 5727802497b0504167216932af6d08a392845c6e）, we have addressed the issue by setting some characteristic properties to be encrypted. Thank you very much for reporting the issue.

closed 2023-11-09 by Alvin1Zhang

paul356, 2025-11-23:
I find that the issue Jeremy points out still exists. The commit 5727802 is not sufficient. After adding encrypted to the report map and report characteristic I can make an esp32s3 flashed with esp_hid_device work right after bonding.

I use esp-idf 5.4. My changes:

```
--- a/components/esp_hid/src/ble_hidd.c
+++ b/components/esp_hid/src/ble_hidd.c
@@ -386,7 +386,7 @@ static esp_err_t create_hid_db(esp_ble_hidd_dev_t *dev, int device_index)
     add_db_record(_last_db, HIDD_LE_IDX_PROTO_MODE_VAL, (uint8_t *)&s_hid_proto_mode_uuid, ESP_GATT_PERM_READ | ESP_GATT_PERM_WRITE, 1, 1, (uint8_t *)&dev->protocol);
 
     add_db_record(_last_db, HIDD_LE_IDX_REPORT_MAP_CHAR, (uint8_t *)&s_character_declaration_uuid, ESP_GATT_PERM_READ, 1, 1, (uint8_t *)&s_char_prop_read);
-    add_db_record(_last_db, HIDD_LE_IDX_REPORT_MAP_VAL, (uint8_t *)&s_hid_report_map_uuid, ESP_GATT_PERM_READ, HIDD_LE_REPORT_MAP_MAX_LEN, dev->devices[device_index].reports_map.len, (uint8_t *)dev->devices[device_index].reports_map.data);
+    add_db_record(_last_db, HIDD_LE_IDX_REPORT_MAP_VAL, (uint8_t *)&s_hid_report_map_uuid, ESP_GATT_PERM_READ_ENCRYPTED, HIDD_LE_REPORT_MAP_MAX_LEN, dev->devices[device_index].reports_map.len, (uint8_t *)dev->devices[device_index].reports_map.data);
     add_db_record(_last_db, HIDD_LE_IDX_REPORT_MAP_EXT_REP_REF, (uint8_t *)&s_hid_report_map_ext_desc_uuid, ESP_GATT_PERM_READ, 2, 2, (uint8_t *)&s_bat_level_uuid);
 
     size_t index = HIDD_LE_IDX_NB;
@@ -398,18 +398,18 @@ static esp_err_t create_hid_db(esp_ble_hidd_dev_t *dev, int device_index)
                 //Input Report
                 add_db_record(_last_db, index++, (uint8_t *)&s_character_declaration_uuid, ESP_GATT_PERM_READ, 1, 1, (uint8_t *)&s_char_prop_read_notify);
                 report->index = index;
-                add_db_record(_last_db, index++, (uint8_t *)&s_hid_report_uuid, ESP_GATT_PERM_READ, report->value_len, 0, NULL);
-                add_db_record(_last_db, index++, (uint8_t *)&s_character_client_config_uuid, ESP_GATT_PERM_READ | ESP_GATT_PERM_WRITE_ENCRYPTED, 2, 0, NULL);
+                add_db_record(_last_db, index++, (uint8_t *)&s_hid_report_uuid, ESP_GATT_PERM_READ_ENCRYPTED, report->value_len, 0, NULL);
+                add_db_record(_last_db, index++, (uint8_t *)&s_character_client_config_uuid, ESP_GATT_PERM_READ | ESP_GATT_PERM_WRITE, 2, 0, NULL);
             } else if (report->report_type == ESP_HID_REPORT_TYPE_OUTPUT) {
                 //Output Report
                 add_db_record(_last_db, index++, (uint8_t *)&s_character_declaration_uuid, ESP_GATT_PERM_READ, 1, 1, (uint8_t *)&s_char_prop_read_write_write_nr);
                 report->index = index;
-                add_db_record(_last_db, index++, (uint8_t *)&s_hid_report_uuid, ESP_GATT_PERM_READ | ESP_GATT_PERM_WRITE, report->value_len, 0, NULL);
+                add_db_record(_last_db, index++, (uint8_t *)&s_hid_report_uuid, ESP_GATT_PERM_READ_ENCRYPTED | ESP_GATT_PERM_WRITE_ENCRYPTED, report->value_len, 0, NULL);
             } else {
                 //Feature Report
                 add_db_record(_last_db, index++, (uint8_t *)&s_character_declaration_uuid, ESP_GATT_PERM_READ, 1, 1, (uint8_t *)&s_char_prop_read_write);
                 report->index = index;
-                add_db_record(_last_db, index++, (uint8_t *)&s_hid_report_uuid, ESP_GATT_PERM_READ | ESP_GATT_PERM_WRITE, report->value_len, 0, NULL);
+                add_db_record(_last_db, index++, (uint8_t *)&s_hid_report_uuid, ESP_GATT_PERM_READ_ENCRYPTED | ESP_GATT_PERM_WRITE_ENCRYPTED, report->value_len, 0, NULL);
             }
             add_db_record(_last_db, index++, (uint8_t *)&s_hid_report_ref_descr_uuid, ESP_GATT_PERM_READ, 2, 2, (uint8_t *)&report->report_id);
         } else {
```