espressif/esp-idf #11807: Why PPP still new/experimental in menuconfig at IDFv5.0.2?

WangShuoran, 2023-07-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello, In #10249 issue, @david-cermak say:

> Sorry for the late reply. The PPP is officially supported by Espressif, sorry for the confusion. We forgot to remove this note about the experimental status.

But at IDFv4.4.4 and v5.0.2 the menuconfig still have the question:
[image]

Why?

WangShuoran, 2023-11-05:
In IDF v5.1.1 this feature is stable, and SLIP why not stable?

WangShuoran, 2023-11-05:
@methodmissing Can you help me? The SLIP is very old protocol, I want to use this protocol.

david-cermak, 2023-11-14:
@WangShuoran We removed the experimental note from PPPoS option in `v5.1`.

> Why?

This hasn't been tested properly, and when introduced it was marked as experimental.

> The SLIP is very old protocol, I want to use this protocol.

You can check this example: https://github.com/espressif/esp-protocols/tree/master/examples/esp_netif/slip_custom_netif

david-cermak, 2023-11-28:
Hope it's been answered. Please reopen if not.

closed 2023-11-28 by david-cermak