espressif/esp-idf #11809: WPA2 Enterprise using EAP-TLS does not work in ESP32-AT

BPaillard, 2023-07-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

We are working with ESP32 C3-Mini1 and using the ESP-AT firmware.
The firmware has been rebuilt to allow the AT+CJEAP command for Enterprise connection.
At the time, we rebuilt with the latest version of SDK and firmware. Here are the relevant versions:
AT version:2.5.0.0-dev(s-2bb352d - ESP32C3 - Oct 20 2022 08:54:52)
SDK version:v4.3.2-553-g588436db47
compile time(409a018):Oct 20 2022 09:57:31
Bin version:2.5.32768(MINI-1)
(Bin version was marked with "32768" so we can recognize it in operation)
We have verified that the Enterprise connection works well with EAP-PEAP and EAP-TTLS (using username/password)
But we have not been able to make EAP-TLS work (using certificates).
We are using a router with a RADIUS server attached, running Freeradius software.
The proper certificates have been installed in the RADIUS server, as well as the corresponding segments of the "WPA2" partition on the ESP32 module. The RADIUS server has been setup properly to use EAP-TLS.
The issue seems to be a breakdown in the protocol:
- The RADIUS server sends an access challenge for TLS-EAP
- The ESP-AT firmware responds with an NAK, but with a desired auth type of TLS-EAP

This behavior is similar to the behavior desribed in [https://github.com/espressif/esp-idf/issues/5698](url)
The error log from the RADIUS server has the same errors as what we see. In particular: 

(1) eap: WARNING: !!! We requested to use an EAP type as normal.
(1) eap: WARNING: !!! The supplicant rejected that, and requested to use the same EAP type.
(1) eap: WARNING: !!! i.e. the supplicant said 'I don't like X, please use X instead.
(1) eap: WARNING: !!! The supplicant software is broken and does not work properly.
(1) eap: WARNING: !!! Please upgrade it to software that works.

I am attaching print-screens of the intercepted exchange between the RADIUS server and the ESP32 module, as well as the PCAP log.
[image]
[image]
[2023-07-04_23-53.zip](https://github.com/espressif/esp-idf/files/11957200/2023-07-04_23-53.zip)

Luxin812, 2023-07-06:
@BPaillard Maybe you can use IDF's [wifi_enterprise](https://github.com/espressif/esp-idf/tree/release/v4.3/examples/wifi/wpa2_enterprise) demo to test whether it can be successful using EAP-TLS?

kapilkedawat, 2023-07-06:
@BPaillard please share server config, server logs, dut logs(with supplicant logs enabled and esp-idf's log level set to `debug`) and wireless packet capture during the session.

BPaillard, 2023-07-06:
[FreeradiusDebu
[ESP32_Command_Response.txt](https://github.com/espressif/esp-idf/files/11971399/ESP32_Command_Response.txt)
gLog.txt](https://github.com/espressif/esp-idf/files/11971397/FreeradiusDebugLog.txt)
[ESP32_Mini1_DebugLog.txt](https://gith
[freeradius.zip](https://github.com/espressif/esp-idf/files/11971403/freeradius.zip)
ub.com/espressif/esp-idf/files/11971401/ESP32_Mini1_DebugLog.txt)

OK, I rebuilt the ESP-AT firmware with enabling of output logs "debug" level. I also increased the timeouts in our firmware because the rebuilt ESP-AT is much slower to boot. But no other change beyond that.

FreeradiusDebugLog.txt is the debug log from Freeradius
ESP32_Command_Response.txt is a capture of the exchange between our MCU and ESP-AT in the ESP32-Mini1 module
ESP32_Mini1_DebugLog.txt is the debug log from the ESP32 module.
freeradius.zip is a zip containing all the files (including the certs and config files) for freeradius 3.0.

BPaillard, 2023-07-06:
[FreeradiusDebugLog.txt](https://github.com/espressif/esp-idf/files/11971438/FreeradiusDebugLog.txt)

BPaillard, 2023-07-06:
[ESP32_Mini1_DebugLog.txt](https://github.com/espressif/esp-idf/files/11971446/ESP32_Mini1_DebugLog.txt)

BPaillard, 2023-07-06:
Somehow the initial file upload got the files mixed up. So I re-uploaded the ones that are broken... Sorry.

kapilkedawat, 2023-07-07:
Hi @BPaillard,
Can you please try https://github.com/espressif/esp-idf/issues/11809#issuecomment-1622887200? 

Please make sure we do following before testing:

1. Make sure supplicant debugs are enabled https://github.com/espressif/esp-idf/blob/master/components/esp_wifi/Kconfig#L549 
2. Mbedtls logs are enabled https://github.com/espressif/esp-idf/blob/master/components/mbedtls/Kconfig#L139
3. Get wireless sniffer capture b/w the AP and esp32 station
4. You mentioned that with certificates its not working, are you flashing them at correct address? If they are encrypted, are you giving password correctly?

BPaillard, 2023-07-07:
[ESP32_Mini1_DebugLog_w_mbedTLS.txt](https://github.com/espressif/esp-idf/files/11981620/ESP32_Mini1_DebugLog_w_mbedTLS.txt)
[ESP32_Command_Response_2.txt](https://github.com/espressif/esp-idf/files/11981621/ESP32_Command_Response_2.txt)

Hi @kapilkedawat 

I modified the mbedTLS debug and supplicant debug options in menuconfig and rebuilt. I think we have better information! I attached the debug log and Command-Response log.

In response to https://github.com/espressif/esp-idf/issues/11809#issuecomment-1622887200
OK, maybe I should have explained a bit better. The ESP32 module is in our product right now, and fielded to some clients already. I am trying to see if I can enable the "Enterprise" connectivity without modifying the ESP32 firmware. The only way to work with the ESP32 module is to drive it from our MCU.

Right now I do not have the AP instrumentation to capture the wireless stream. But we will be working on that.

Yes, the certificates files are formatted according to the documentation in [ESP32C3-AT][v2.4.0.0]User-Guide.pdf, and flashed in the wpa2_cert, wpa2_key and wpa2_ca partitions. It is possible I guess that the formatting is wrong, but it does follow the documentation. I have also run AtPKI.py on the certificates files and checked that the contents of the 3 partitions is identical to the outputs of AtPKI.py
The 3 certificate files that I use on the client side are the client certificates from the examples/wifi/wifi_enterprise/main folder  and on the radius server I use the server-side certificates from the example. We have also checked the validity of those certificates.

BPaillard, 2023-07-07:
I am not sure if it is relevant, but the ESP-AT firmware that I am using has AMPDU Rx and Tx disabled. This to improve performance when client and server are exchanging small packets at a rapid rate.
To my knowledge this, and enabling AT+CJEAP are the only differences between that firmware and the Espressif factory release.

kapilkedawat, 2023-07-10:
`
D (3888) wpa: TLS: using phase1 config options[0m
[0;31mE (3893) wpa: mbedtls_x509_crt_parse returned -0x2180[0m
[0;31mE (3898) wpa: Failed to set client configurations[0m
[0;31mE (3903) wpa: failed to create ssl handle[0m
[0;32mI (3908) wpa: TLS: Failed to set TLS connection parameters[0m
`
@BPaillard device is not able to parse the certs. Please share the certs and how did you flash them, If possible, please also share the code snippet which you are using.

BPaillard, 2023-07-10:
[OriginalCertFiles.zip](https://github.com/espressif/esp-idf/files/12001469/OriginalCertFiles.zip)
[FlashContents.zip](https://github.com/espressif/esp-idf/files/12001471/FlashContents.zip)

Hi @kapilkedawat 
I attached the original cert files, and the contents of the 3 WPA2 partitions. For these, I have simply created a file, each with the binary contents of the corresponding WPA2 partition (each 8192 bytes in length). Let me know if you would prefer another format. For instance I could create an ASCII file containig hex representations of each byte... Might be more human-readable, but harder to parse.
Unfortunately I cannot share the code used to create the Flash contents. It has been developped in LabVIEW since it needs to be integrated with the rest of our environment.
However, I did use the AtPKI Python script to generate the .bin from the cert files, and did compare to the Flash contents that we generated.

kapilkedawat, 2023-07-10:
@BPaillard Can you flash files directly to the addresses instead of converting them? 
Please print the data before calling esp_wifi_sta_wpa2_ent_set_cert_key() and make sure you are getting the exact same certs that are available in original cert file.

BPaillard, 2023-07-10:
@kapilkedawat 
You mean flash the raw certificale files??? Not the bin files as explained in [ESP32C3-AT][v2.4.0.0]User-Guide.pdf?

We do not call esp_wifi_wpa2_ent_set_cert_key()... We Flash the cert files (and read them back) using the AT+SYSFLASH command (we work with the ESP-AT firmware). This is the same process that we use for the MQTT certs and we checked that the MQTT certs do work on AWS using Amazon-provided certs.

kapilkedawat, 2023-07-10:
Could you please try same setup with [wifi_enterprise](https://github.com/espressif/esp-idf/tree/release/v4.3/examples/wifi/wpa2_enterprise) and see if that works?

BPaillard, 2023-07-10:
Hi @kapilkedawat 

If you mean to build that project and flash it to the module, I already explained that we cannot do that. The module is in our product and only works with the ESP-AT firmware communicating with our MCU. That configuration is already deployed, so my aim at this point is to see if that ESP-AT firmware can connect to Enterprise with EAP-TLS. We already checked that it works with EAP-TTLS and EAP-PEAP.

If you mean, use the same certs from the example on the device and the radius server, that is what we are doing.
The certs that we are using right now are the certs that are found in the examples.wifi/wpa2_enterprise/main folder for that example. Before that, we have tried with certs generated with OpenSSL as explained in the example in reference. That gave the same result.

I am surprised that the ESP-AT firmware cannot parse the certs that are in the WPA2 partitions. Do you see any reason why?

BPaillard, 2023-07-10:
Hi @kapilkedawat 

Re: Your earlier suggestion "Can you flash files directly to the addresses instead of converting them?"

Just to be sure I tried to write the raw contents of the cert files in the Flash partitions without converting them first. I immediately get errors:
0x800b (AT_EAP_GET_CERT_LEN_FAILED) and
0x8003 (AT_EAP_CONN_FAILED) 

I think this is expected. The encoding of the certs as described in [ESP32C3-AT][v2.4.0.0]User-Guide.pdf serves to delimit them and to add information such as the number and size of certs in the partition.

ustccw, 2023-07-12:
@BPaillard Hello, could you please try to replace the following library and test again.

[libespat.zip](https://github.com/espressif/esp-idf/files/12026005/libespat.zip)

From the log:
```
mbedtls_x509_crt_parse returned -0x2180
```
it should be a known bug that the last byte of ca/cert/key should be NULL terminated, when esp-at uses `esp_wifi_sta_wpa2_ent_set_ca_cert()` and `esp_wifi_sta_wpa2_ent_set_cert_key()` APIs.

BPaillard, 2023-07-12:
@ustccw 

YES!!!
After null-terminating the cert, key and CA in the WPA2 certificate partitions, the device does connect using EAP-TLS.

Thank you!

closed 2023-07-12 by BPaillard

ustccw, 2023-07-12:
no worries

Rbv15, 2023-11-23:
Hello. I still have the same issue: ” The supplicant rejected that, and requested to use the same EAP type” radius log message. I tried several ways: Arduino IDE Esp32 Dev (multiple cores and Wifi.begin versions), Expressif IDF 5.0, and now, I will try the Esp32 AT firmware.
Perhaps the trick of null terminating the partitions.
I have other issue.
 Is it possible to set a CA certificate chain to achieve ESP32 WPA2 Enterprise EAP-TLS authentication? I need to connect into a WAN WPA2 Enterprise that requires CA certificate validation, and uses root CA and  intermediate CA certificate. How to put more then one CA certificate? It seems to me that there are separate partitions sets, for different purposes (EAP, HTTPS, MQTT). And the bundle Ca certificates are not involved in EAP. Am I wrong?
Is there any tool to load the contents of a P12 directly to ESP32 partitions for EAP purpose? A p12 with the root CA, intermediate CA, client certificate, and client key, putting them on each related partition? 
Sorry my poor English.

BPaillard, 2023-11-23:
Hi Rbv15

I cannot comment on your second issue. But in my case null-terminating the certificates solved the problem. Note that it is the certificate that needs to be null-terminated, not the partition (add 0x00 at the end of the certificate, not at the end of the partition). It is not the same for ESP32-AT V2.5, where each partition has a fixed number of bytes.
Regarding you second question (again for ESP32-AT firmware V2.5), it is true that each set of certificates has 3 separate storage locations: a) the CA certificate, b) the client certificate, c) the private key.
There are SEPARATE sets of 3 locations for MQTT and for Enterprise. So the CA for WPA2 enterprise does not need to be the same as the CA for MQTT. This is normal, since the CA for WPA2 Enterprise is typically custom, while the CA for MQTT is typically chosen by the broker's organisation (AWS for example).

Rbv15, 2023-11-24:
Hi BPaillard

Thank you, your answer. Yes, the null character to end certificate, not at the limit of memory to hold it.
Make sense what you said about sets of certificates. I thought the same.

Concerning my issues, I appreciate any suggestions or postings of CA chain solutions related to EAP handshake, from anybody.