espressif/esp-idf #11812: multitasking preemptive or cooperative ?

diplfranzhoepfinger, 2023-07-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/freertos_idf.html

Hello, 

after Reading this, 
it is still unclear for me if the Task Scheduler is preemptive or cooperative ? 

Thanks, 
Franz

KaeLL, 2023-07-05:
Priority-based preemptive with the dual-core mods.

ESP-Marius, 2023-07-06:
@KaeLL is correct.

Described in more details [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/freertos_idf.html#smp-scheduler)

diplfranzhoepfinger, 2023-07-06:
TNX

closed 2023-07-06 by diplfranzhoepfinger