espressif/esp-idf #11814: Resolving .local hostname with DNS server running on AP

baldhead69, 2023-07-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi,

I did some tests on my AP( DD-WRT installed ) that is running DHCP server and has dnsmasq enabled.

If i remove " address=/http://myfota.local/192.168.1.201 " from DD-WRT "dns table", the chrome doesn't find "www.myfota.local" and ping from a windows command prompt, ping www.myfota.local doesn't find too, but if i put "address=/http://myfota.local/192.168.1.201 " in DD-WRT "dns table", both chrome and command prompt ping worked.

ESP32-S3 ping doesn't find the host:
```
#define MY_PING_URL "www.myfota.local"

esp_err_t initialize_ping()
{
    // convert URL to IP address
    ip_addr_t target_addr;
    struct addrinfo hint;
    struct addrinfo *res = NULL;
    
    memset(&hint, 0, sizeof(hint));
    memset(&target_addr, 0, sizeof(target_addr));
    
    int ret = getaddrinfo(MY_PING_URL, NULL, &hint, &res);
    if( ret != 0 )
    {
        ESP_LOGI( MY_PING_TAG, LOG_USER1( "Unknown host: [ %s ]" ), MY_PING_URL );
        ESP_LOGI( MY_PING_TAG, LOG_USER1( "ret = %d\n" ), ret );
        return ESP_ERR_NOT_FOUND;
    }

    if( res->ai_family == AF_INET )  // ipv4
    {
        struct in_addr addr4 = ( (struct sockaddr_in *) (res->ai_addr) )->sin_addr;
        inet_addr_to_ip4addr( ip_2_ip4( &target_addr ), &addr4 );

        //target_addr.type = IPADDR_TYPE_V4;

        ESP_LOGI( MY_PING_TAG, LOG_USER1( "IPv4 = %s" ), ip4addr_ntoa( &(target_addr.u_addr.ip4) ) );
    }
    else if( res->ai_family == AF_INET6 )  // ipv6
    {
        struct in6_addr addr6 = ( (struct sockaddr_in6 *) (res->ai_addr) )->sin6_addr;
	inet6_addr_to_ip6addr( ip_2_ip6( &target_addr ), &addr6 );

        //target_addr.type = IPADDR_TYPE_V6;

        ESP_LOGI( MY_PING_TAG, LOG_USER1( "IPv6 = %s" ), ip6addr_ntoa( &(target_addr.u_addr.ip6) ) );
    }
    else  // unknown family
    {
        ESP_LOGI( MY_PING_TAG, LOG_USER1( "Unknown family = %d" ), res->ai_family );        
    }

    freeaddrinfo(res);

    esp_ping_config_t ping_config = ESP_PING_DEFAULT_CONFIG();
    ping_config.target_addr = target_addr;  // target IP address
    ping_config.count = ESP_PING_COUNT_INFINITE;    // ping in infinite mode, esp_ping_stop can stop it
    ping_config.timeout_ms = 5000;

    // set callback functions
    esp_ping_callbacks_t cbs;
    cbs.on_ping_success = test_on_ping_success;
    cbs.on_ping_timeout = test_on_ping_timeout;
    cbs.on_ping_end = test_on_ping_end;
    cbs.cb_args = NULL;  // arguments that will feed to all callback functions, can be NULL
    
    esp_ping_handle_t ping;
    
    esp_err_t e = esp_ping_new_session(&ping_config, &cbs, &ping);
    if( e != ESP_OK )
    {
        ESP_LOGI( MY_PING_TAG, LOG_USER1( "Nao foi possivel criar a sessao ping !!!\nError = %s\n" ), esp_err_to_name( e ) );
        return e;
    }
        
    e = esp_ping_start(ping);
    if( e != ESP_OK )
    {
        ESP_LOGI( MY_PING_TAG, LOG_USER1( "Nao foi possivel iniciar o ping !!!\nError = %s\n" ), esp_err_to_name( e ) );
        return e;
    }

    ESP_LOGI( MY_PING_TAG, LOG_USER1( "My Ping Start" ) );
    return ESP_OK;
}
```

Any suggestions on how to configure lwip "dns client" to work this way ?

Notes:
* esp-idf v5.0.2
* ESP32-S3
* Windows 11
* Image in annex: Dnsmasq_Infrastructure

baldhead69, 2023-07-06:
I changed  "address=/http://myfota.local/192.168.1.201" to "address=/http://myfota.com/192.168.1.201" from DD-WRT "dns table" and worked.

It seems that the lwip lib automatically changes the protocol for resolving the name, based on the TLD (top level domain).

TLD = .local use protocol Mdns
TLD = .com use protocol Dns

As my AP( with DD WRT ) and my apache server were not running/sending the mdns packet to the network, the name was not resolved by lwip in esp32-s3.
When i put .com in the TLD the lwip lib used the dns protocol and my AP( with DD WRT ) responded with the dns packet.

Is this normal functioning of lwip's dns/mdns ?
Is there any rfc that says to use Mdns for .local and DNS for internet address ?

igrr, 2023-07-06:
In ESP-IDF this behavior is controlled by [CONFIG_LWIP_DNS_SUPPORT_MDNS_QUERIES](https://docs.espressif.com/projects/esp-idf/en/release-v4.4/esp32/api-reference/kconfig.html#config-lwip-dns-support-mdns-queries) option and can be disabled. For background, please see https://github.com/espressif/esp-idf/issues/5521.

I don't think there is a simple way to disable it in Arduino, though, other than using Arduino as an IDF component or recompiling Arduino libraries using the lib-builder.

baldhead69, 2023-07-06:
@igrr 

I have a websocket secure server ( wss ) running on esp32-s3 too that uses Mdns "server" to broadcast(multicast) its name/ip for a app in my android phone find it.

If i disable "CONFIG_LWIP_DNS_SUPPORT_MDNS_QUERIES", Mdns "server" will broadcast(multicast) wss server name/ip ?

david-cermak, 2023-07-28:
> If i disable "CONFIG_LWIP_DNS_SUPPORT_MDNS_QUERIES", Mdns "server" will broadcast(multicast) wss server name/ip ?

Yes, the mdns service would still work. `CONFIG_LWIP_DNS_SUPPORT_MDNS_QUERIES` only controls how lwip resolves local names with `getaddrinfo()`. It is unrelated to `mdns` component behavior.

closed 2023-10-31 by david-cermak