espressif/esp-idf #11815: RTC timer preservation after reset

diplfranzhoepfinger, 2023-07-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

https://github.com/diplfranzhoepfinger/RTC-timer-preservation-after-reset

diplfranzhoepfinger, 2023-07-06:
and it works like proposed: 

see: [https://gist.github.com/diplfranzhoepfinger/937d60cc8bb981f24b59edc52885f094](https://gist.github.com/diplfranzhoepfinger/937d60cc8bb981f24b59edc52885f094)

it preserves Time during the Reset.

diplfranzhoepfinger, 2023-07-06:
there are 2 ways where the RTC timer is not preserved: 

Power on

Reset by Pressing Reset on the Devboard.

diplfranzhoepfinger, 2023-07-06:
[https://github.com/diplfranzhoepfinger/RTC-timer-preservation-after-reset/tree/feature/reset_with_assert](https://github.com/diplfranzhoepfinger/RTC-timer-preservation-after-reset/tree/feature/reset_with_assert)

simulates a assert catched: 

also time is preserved.

KaeLL, 2023-07-06:
What's your question?

franz-hoepfinger-at-hm-edu, 2023-07-06:
Currently no Question. 
My Customer will test this code on his side. If it works perfectly we will close the Ticket. If it doesn't work we will add the Question here.

ESP-Marius, 2023-07-12:
If there are no issues i'll just close this one then. Feel free to re-open if you do encounter anything.

closed 2023-07-12 by ESP-Marius