espressif/esp-idf #11816: idf.py build -B option doesn't work

MartyMcFlyInTheSky, 2023-07-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

Using `idf.py build -B build01` should prompt idf.py to create build tree inside the build01 directory according to the docs:

> Create the build directory if needed. The sub-directory build is used to hold build output, although this can be changed with the -B option.

(the same is also printed by idf.py build --help)

### What is the actual behavior?

I get this output:

```
:\dev\sams_playground>idf.py build -B build01
C:\dev\esp\esp-idf\tools\check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
Usage: idf.py build [OPTIONS]
Try 'idf.py build --help' for help.

Error: No such option: -B
```

### Steps to reproduce.

execute `idf.py build -B build01`

igrr, 2023-07-06:
The build directory option is a global option, it applies to most idf.py commands. You need to specify it before the command, i.e. `idf.py -B my_dir build` or `idf.py -B my_dir flash monitor `

closed 2023-07-06 by MartyMcFlyInTheSky

dobairoland, 2023-07-10:
@MartyMcFlyInTheSky Could give us an idea where that should be documented? Where were you looking to solve this?