espressif/esp-idf #11817: Hart 0 意外重启

woshiwstt50123, 2023-07-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

我在“运行 OpenOCD[](https://docs.espressif.com/projects/esp-idf/zh_CN/release-v5.1/esp32c6/api-guides/jtag-debugging/index.html#jtag-debugging-run-openocd)”这一步中，在命令行输入“openocd -f board/esp32c6-builtin.cfg”后，得到以下错误：
PS C:\Program Files\PowerShell\7> openocd -f board/esp32c6-builtin.cfg
Open On-Chip Debugger v0.12.0-esp32-20230419 (2023-04-18-22:02)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselecting 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Warn : Transport "jtag" was already selected
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : esp_usb_jtag: serial (60:55:F9:F7:18:38)
Info : esp_usb_jtag: Device found. Base speed 24000KHz, div range 1 to 255
Info : clock speed 24000 kHz
Info : JTAG tap: esp32c6.cpu tap/device found: 0x0000dc25 (mfg: 0x612 (Espressif Systems), part: 0x000d, ver: 0x0)
Info : datacount=2 progbufsize=16
Info : Examined RISC-V core; found 2 harts
Info :  hart 0: XLEN=32, misa=0x40903105
Info : starting gdb server for esp32c6 on 3333
Info : Listening on port 3333 for gdb connections
Info : Hart 0 unexpectedly reset!
Info : Reset cause (12) - (Software CPU reset)
Info : Hart 0 unexpectedly reset!
Info : Reset cause (12) - (Software CPU reset)
Info : Hart 0 unexpectedly reset!
Info : Reset cause (12) - (Software CPU reset)
Info : Hart 0 unexpectedly reset!
Info : Reset cause (12) - (Software CPU reset)
Info : Hart 0 unexpectedly reset!
Info : Reset cause (12) - (Software CPU reset)
Info : Hart 0 unexpectedly reset!
Info : Reset cause (12) - (Software CPU reset)
Info : Hart 0 unexpectedly reset!
Info : Reset cause (12) - (Software CPU reset)
Info : Hart 0 unexpectedly reset!
Info : Reset cause (12) - (Software CPU reset)
Info : Hart 0 unexpectedly reset!
Info : Reset cause (12) - (Software CPU reset)
Info : Hart 0 unexpectedly reset!
Info : Reset cause (12) - (Software CPU reset)
Info : Hart 0 unexpectedly reset!
Info : Reset cause (12) - (Software CPU reset)
Info : Hart 0 unexpectedly reset!
Info : Reset cause (12) - (Software CPU reset)
shutdown command invoked
Hart0意外重启好几次。
我不知道发生了什么，也没有在issue列表中搜索到类似的问题。这个对我造成了极大困扰，希望能尽快找到问题所在。

igrr, 2023-07-06:
"Hart 0 unexpectedly reset" message by itself doesn't indicate any issue. Most often it is the result of the application restarting the chip using esp_restart() API.
Which application have you flashed to the board?

woshiwstt50123, 2023-07-06:
example文件夹里的hello_world程序

closed 2023-07-06 by woshiwstt50123

igrr, 2023-07-06:
Makes sense then, hello_world example does call esp_restart.