espressif/esp-idf #11818: OPENOCD server 无法正常打开

woshiwstt50123, 2023-07-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

烧录hello_world程序时，在flash device阶段遇到问题。
端口设置确认没有问题，目标芯片选择的是esp32c6的“builtin usb jtag”，编译完成没有报错。烧录时提示打开openocd。打开openocd时出现以下错误：
c:\Users\TangTao\esp\tools\openocd-esp32\v0.12.0-esp32-20230419\openocd-esp32\bin\openocd.exe

Open On-Chip Debugger v0.12.0-esp32-20230419 (2023-04-18-22:02)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html

c:\Users\TangTao\esp\tools\openocd-esp32\v0.12.0-esp32-20230419\openocd-esp32\bin\openocd.exe

[OpenOCD]
Open On-Chip Debugger v0.12.0-esp32-20230419 (2023-04-18-22:02)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html

[OpenOCD]
debug_level: 2

[OpenOCD]
Info : only one transport option; autoselecting 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and P
[OpenOCD]
ID to 0x1002
Info : esp_usb_jtag: capabilities descriptor set to 0x30a
Info : esp_usb_jtag: target chip id set t
[OpenOCD]
o 1

[OpenOCD]
Warn : Transport "jtag" was already selected

[OpenOCD]
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections

[OpenOCD]
❌ Error: esp_usb_jtag: could not find or open
[OpenOCD]
❌  device!
c:\Users\TangTao\esp\tools\openocd-esp32\v0.12.0-esp32-20230419/openocd-esp32/share/openocd/scripts/target/esp_common.cfg:4: Error: 
at file "c:\Users\TangTao\esp\tools\openocd-esp32\v0.12.0-esp32-20230419/openocd-esp32/share/openocd/scripts/target/esp_common.cfg", line 4

[OpenOCD]
[Stopped] : OpenOCD Server
[Flash]
Can't perform JTag flash, because OpenOCD server is not running!

igrr, 2023-07-06:
Please check the device manager to see if the USB Serial/JTAG device is detected by the system.

woshiwstt50123, 2023-07-06:
确认被操作系统识别。操作系统里显示这个端口的名字是“USB串行设备（COM7）”

igrr, 2023-07-06:
That looks like a USB-Serial bridge (CP2102). The name of the built-in USB of C6 is "USB Serial/JTAG Debug Unit". Please check the VID and PID of the device to confirm.

woshiwstt50123, 2023-07-07:
不是。“Silicon Labs CP210x USB to UART Bridge”是另外一个USB插槽插入后的名字。它和“USB串行设备（COM7）”不是同一个接口。

woshiwstt50123, 2023-07-07:
它的VID和PID有两行：
USB\VID_303A&PID_1001&REV_0102&MI_00
USB\VID_303A&PID_1001&MI_00

igrr, 2023-07-07:
When installing IDF, have you selected the option to install the USB driver (https://dl.espressif.com/dl/idf-driver/idf-driver-esp32-usb-jtag-2021-07-15.zip)? Please check once.

woshiwstt50123, 2023-07-07:
有，在“.espressif\tools\idf-driver\idf-driver-esp32-usb-jtag-2021-07-15”下面。

woshiwstt50123, 2023-07-07:
奇怪，我没有在vscode的插件里看到安装这个驱动的选项。
[image]

igrr, 2023-07-07:
> ID to 0x1002

> VID_303A&PID_1001

Looks like VS Code is configured to use a different JTAG adapter than the one you are using. You probably need to select "ESP32-C6 chip (via builtin USB-JTAG)" instead of "ESP32-C6 chip (via ESP USB Bridge)" in VS Code.

woshiwstt50123, 2023-07-11:
我把原来的文件全部删掉，然后重新复制一份，这样才能成功下载进去。是不是在使用“ESP32-C6 chip (via ESP USB Bridge)”或其他类似选项时会引入新的文件让“ESP32-C6 chip (via builtin USB-JTAG)”的选项失效？

woshiwstt50123, 2023-07-26:
Error: esp_usb_jtag: could not find or open device!这个问题现在又出现了。我已经确认选择的是“ESP32-C6 chip (via builtin USB-JTAG)”，并且USB串行设备（COM7）也已经识别到

jeongyeham, 2023-09-05:
您好，请检查您的JTAG驱动程序是否正确！
[image]