espressif/esp-idf #11820: Some command will fail in empty git repo repository

Simakeng, 2023-07-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esp-idf-v5.1-rc2

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

running command in empty git repository should compiles like normal

### What is the actual behavior?

yeild an error.

### Steps to reproduce.

1. create an empty folder
2. create `main/main.c`, `main/CMakeList.txt`, './CMakeList.txt'
3. run `git init`
4. run `idf.py set-target esp32c3`

Problem recreate

[image]

[image]

[image]

### Build or installation Logs.

```plain
C:\Users\simak\Desktop\Atom>idf.py set-target esp32c3
C:\Espressif\frameworks\esp-idf-v5.1-rc2-2\tools\check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Build directory 'C:\Users\simak\Desktop\Atom\build' not found. Nothing to clean.
Executing action: set-target
Set Target to: esp32c3, new sdkconfig will be created.
Running cmake in directory C:\Users\simak\Desktop\Atom\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=C:\Espressif\python_env\idf5.1_py3.8_env\Scripts\python.exe -DESP_PLATFORM=1 -DIDF_TARGET=esp32c3 -DCCACHE_ENABLE=1 C:\Users\simak\Desktop\Atom"...
-- Existing sdkconfig 'C:/Users/simak/Desktop/Atom/sdkconfig' renamed to 'C:/Users/simak/Desktop/Atom/sdkconfig.old'.
-- Found Git: C:/Espressif/tools/idf-git/2.39.2/cmd/git.exe (found version "2.39.2.windows.1")
fatal: detected dubious ownership in repository at 'C:/Espressif/frameworks/esp-idf-v5.1-rc2-2'
'C:/Espressif/frameworks/esp-idf-v5.1-rc2-2' is owned by:
        'S-1-5-32-544'
but the current user is:
        'S-1-5-21-4017592808-955350438-2807299510-1001'
To add an exception for this directory, call:

        git config --global --add safe.directory C:/Espressif/frameworks/esp-idf-v5.1-rc2-2
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 12.2.0
[... 52 lines cut ...]
        'S-1-5-21-4017592808-955350438-2807299510-1001'
To add an exception for this directory, call:

        git config --global --add safe.directory C:/Espressif/frameworks/esp-idf-v5.1-rc2-2
fatal: detected dubious ownership in repository at 'C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/openthread/openthread'
'C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/openthread/openthread/.git' is owned by:
        'S-1-5-32-544'
but the current user is:
        'S-1-5-21-4017592808-955350438-2807299510-1001'
To add an exception for this directory, call:

        git config --global --add safe.directory C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/openthread/openthread
-- Components: app_trace app_update bootloader bootloader_support bt cmock console cxx driver efuse esp-tls esp_adc esp_app_format esp_coex esp_common esp_eth esp_event esp_gdbstub esp_hid 
esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_mm esp_netif esp_netif_stack esp_partition esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_system esp_timer esp_wifi espcoredump esptool_py fatfs freertos hal heap http_parser idf_test ieee802154 json log lwip main mbedtls mqtt newlib nvs_flash openthread partition_table 
protobuf-c protocomm pthread riscv sdmmc soc spi_flash spiffs tcp_transport ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant
-- Component paths: C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/app_trace C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/app_update C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/bootloader C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/bootloader_support C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/bt C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/cmock C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/console C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/cxx C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/driver C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/efuse C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/esp-tls C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/esp_adc C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/esp_app_format C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/esp_coex C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/esp_common C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/esp_eth C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/esp_event C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/esp_gdbstub C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/esp_hid C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/esp_http_client C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/esp_http_server C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/esp_https_ota C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/esp_https_server C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/esp_hw_support C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/esp_lcd C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/esp_local_ctrl C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/esp_mm C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/esp_netif C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/esp_netif_stack C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/esp_partition C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/esp_phy C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/esp_pm C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/esp_psram C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/esp_ringbuf C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/esp_rom C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/esp_system C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/esp_timer C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/esp_wifi C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/espcoredump C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/esptool_py C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/fatfs C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/freertos C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/hal C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/heap C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/http_parser C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/idf_test C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/ieee802154 C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/json C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/log C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/lwip C:/Users/simak/Desktop/Atom/main C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/mbedtls C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/mqtt C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/newlib C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/nvs_flash C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/openthread C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/partition_table C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/protobuf-c C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/protocomm C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/pthread C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/riscv C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/sdmmc C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/soc C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/spi_flash C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/spiffs C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/tcp_transport C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/ulp C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/unity C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/usb C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/vfs C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/wear_levelling C:/Espressif/frameworks/esp- [... cut ...] visioning C:/Espressif/frameworks/esp-idf-v5.1-rc2-2/components/wpa_supplicant
-- Configuring incomplete, errors occurred!
See also "C:/Users/simak/Desktop/Atom/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1, output of the command is in the C:\Users\simak\Desktop\Atom\build\log\idf_py_stderr_output_31108 and C:\Users\simak\Desktop\Atom\build\log\idf_py_stdout_output_31108
```

### More Information.

Files:
[Atom.zip](https://github.com/espressif/esp-idf/files/11966963/Atom.zip)

Simakeng, 2023-07-06:
CMakeOutput.log:

[CMakeOutput.log](https://github.com/espressif/esp-idf/files/11966976/CMakeOutput.log)

Dazza0, 2024-01-12:
@Simakeng It seems to Windows specific issue regarding file/folder permissions.

Your project is placed under your own user directory tree (i.e., `C:/Users/simak/...`) while your ESP-IDF repository is outside of your user directory tree (i.e., `C:/Espressif/frameworks/esp-idf-v5.1-rc2-2`). Git will check that if the current user is the owner of a git repository before attempting to use it. Since your ESP-IDF folder is placed outside your user directory tree, you are now the owner of that repository.

You can workaround this by adding your esp-idf directory to the list of safe directories via git configuration. Something like...

```
git config --global --add safe.directory C:/Espressif/frameworks/esp-idf-v5.1-rc2-2
```

This [article](https://medium.com/@thecodinganalyst/git-detect-dubious-ownership-in-repository-e7f33037a8f) provides more details about this issue.

closed 2024-01-12 by Dazza0