espressif/esp-idf #11822: Windows Command Prompt Enviroment Variables

anlpereira, 2023-07-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Somehow I lost my Desktop Shortcut that opens the Command Prompt(ESP-IDF 5.0 CMD) that is created in the installation of the ESP_IDF framework. So I can't open my enviroment anymore.  I was looking for on the web how to create a CMD Shortcut and add the command to open the framework but it doesn't work.
Inside the CMD Properties I have this.
Destiny:  C:\WINDOWS\system32\cmd.exe /k ""C:\Espressif\idf_cmd_init.bat" esp-idf-149e3c9332fe2f82c8aa07db831daed9"

Begin: C:\Espressif\frameworks\esp-idf-v5.0.1\

What is this: esp-idf-149e3c9332fe2f82c8aa07db831daed9

I really apreciate any help. Thank you very much.

dobairoland, 2023-07-10:
Please re-run the installer again and it will create the shortcuts. The download will be shorter - you can point it to the already downloaded copy of ESP-IDF.

closed 2023-07-10 by dobairoland

anlpereira, 2023-07-10:
I could run it just entering this command inside the prompt command: idf_cmd_init.bat esp-idf-149e3c9332fe2f82c8aa07db831daed9
I just called this batch script with this esp-idf.... parameter.