espressif/esp-idf #11824: when I stop nimble ,system assert by accident

luacerbb, 2023-07-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

idf v4.4.4

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-DevKitC-V4

### Power Supply used.

USB

### What is the expected behavior?

stop ble task success

### What is the actual behavior?

sometimes assert when stop ble task.

### Steps to reproduce.

stop ble function:
static void _bd_ble_stop(void)
{
	log_debug(TAG,"ble stop");
	int32_t  ret = ble_gap_adv_stop();
	if (ret) 
	{
		log_debug(TAG, "Error in stopping advertisement with err code = %d",ret);
	}
	
//	if((ret = ble_gap_terminate(g_conn_handle, BLE_ERR_REM_USER_CONN_TERM)) != 0)
//	{
//		log_err(TAG,"ble terminate fail:%d",ret);
//	}

	//关闭nimble任务运行
	ret = nimble_port_stop();
	if (ret == 0) 
	{
		HAL_SleepMs(50);
		nimble_port_deinit();
		ret = esp_nimble_hci_and_controller_deinit();
		if (ret != ESP_OK) 
		{
			log_err(TAG, "esp_nimble_hci_and_controller_deinit() failed with error: %d", ret);
		}
		//释放RAM空间，减少很多	
		esp_bt_mem_release(ESP_BT_MODE_BTDM);
	}
	else
	{
		log_err(TAG,"stop ble err: %d",ret);
	}
}

### Debug Logs.

```plain
[dbg] 07:05:55:50 _bd_ble_stop(1331): ble stop
[dbg] 07:05:55:50 _bd_ble_stop(1335): Error in stopping advertisement with err code = 2
E (43929) NimBLE: ble_hs_stop_terminate_timeout_cb,1 connection(s) still up 

[inf] 07:05:55:50 bleprph_host_task(1011): BLE Host Task Stoped
ram_lld_pdu_tx_flush HCI packet count mismatch (1, 2)

abort() was called at PC 0x40083123 on core 0
0x40083123: lock_acquire_generic at /mnt/hgfs/SUB_PROJECT/esp-idf-v4.4.4/components/newlib/locks.c:139

Backtrace: 0x40081da2:0x3ffdf690 0x40092051:0x3ffdf6b0 0x4009869e:0x3ffdf6d0 0x40083123:0x3ffdf740 0x4008325d:0x3ffdf770 0x400832d6:0x3ffdf790 0x401a8786:0x3ffdf7c0 0x401ab7b9:0x3ffdfad0 0x401b6df1:0x3ffdfb00 0x40097169:0x3ffdfb30 0x4010af3b:0x3ffdfb80 0x4011fa1a:0x3ffdfbb0 0x4010ee9a:0x3ffdfbd0 0x4001791d:0x3ffdfbf0 0x4010f725:0x3ffdfc10 0x4010f802:0x3ffdfc50 0x400188f5:0x3ffdfc80 0x4010f0e3:0x3ffdfcc0 0x40116395:0x3ffdfce0 0x4008854d:0x3ffdfd00 0x40046c1f:0x3ffdfd20 0x401175f9:0x3ffdfd50 0x40118a27:0x3ffdfd70 0x40054d4b:0x3ffdfdc0 0x4010da45:0x3ffdfde0
0x40081da2: panic_abort at /mnt/hgfs/SUB_PROJECT/esp-idf-v4.4.4/components/esp_system/panic.c:408

0x40092051: esp_system_abort at /mnt/hgfs/SUB_PROJECT/esp-idf-v4.4.4/components/esp_system/esp_system.c:137

0x4009869e: abort at /mnt/hgfs/SUB_PROJECT/esp-idf-v4.4.4/components/newlib/abort.c:46

0x40083123: lock_acquire_generic at /mnt/hgfs/SUB_PROJECT/esp-idf-v4.4.4/components/newlib/locks.c:139

0x4008325d: _lock_acquire_recursive at /mnt/hgfs/SUB_PROJECT/esp-idf-v4.4.4/components/newlib/locks.c:167

0x400832d6: __retarget_lock_acquire_recursive at /mnt/hgfs/SUB_PROJECT/esp-idf-v4.4.4/components/newlib/locks.c:323

0x401a8786: _vfprintf_r at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:853 (discriminator 2)

0x401ab7b9: vprintf at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vprintf.c:34 (discriminator 5)

0x401b6df1: esp_log_writev at /mnt/hgfs/SUB_PROJECT/esp-idf-v4.4.4/components/log/log.c:200

0x40097169: esp_log_write at /mnt/hgfs/SUB_PROJECT/esp-idf-v4.4.4/components/log/log.c:210

0x4010af3b: host_rcv_pkt at /mnt/hgfs/SUB_PROJECT/esp-idf-v4.4.4/components/bt/host/nimble/esp-hci/src/esp_nimble_hci.c:346 (discriminator 3)

0x4011fa1a: vhci_send at ??:?

0x4010ee9a: r_eif_send at ??:?

0x4010f725: hci_tx_start at hci_tl.c:?

0x4010f802: r_hci_tl_send at ??:?

0x4010f0e3: r_hci_send_2_host_hack at ??:?

0x40116395: r_llc_common_nb_of_pkt_comp_evt_send at ??:?

0x4008854d: ram_lld_pdu_tx_flush at ??:?

0x401175f9: r_lld_evt_init_wrapper at ??:?

0x40118a27: r_lld_init at ??:?

0x4010da45: btdm_controller_task at ??:?
```

rahult-github, 2023-07-12:
Hi @luacerbb  , 

 I tried the stack deinit / init loop code in the blecent example in order to reproduce the issue, but haven't seen this issue so far. 
 
 Since backtrace is provided, the line number in esp_nimble_hci.c  in the official release doesn't match with the one mentioned in the backtrace. Can you please share the file from your setup placed at :  /mnt/hgfs/SUB_PROJECT/esp-idf-v4.4.4/components/bt/host/nimble/esp-hci/src/esp_nimble_hci.c  .

luacerbb, 2023-07-13:
@rahult-github Thanks for the feedback
[esp_nimble_hci.zip](https://github.com/espressif/esp-idf/files/12035792/esp_nimble_hci.zip)

The file don't matched with the official release version  to solve another bug. https://github.com/espressif/esp-idf/issues/11771

rahult-github, 2023-07-13:
Hi @luacerbb  

Thank you. May i request  you to please use attached esp_nimble_hci_fix.c file ( replace your original file with this new file ) and check once ? 

[esp_nimble_hci_fix.zip](https://github.com/espressif/esp-idf/files/12036436/esp_nimble_hci_fix.zip)

luacerbb, 2023-07-13:
ok,I will try, but you  know,this is occur by accident,I am not sure how long it will take for the problem to recur

rahult-github, 2023-07-13:
Yes, understood. This seems to be a corner case and not always issue.

rahult-github, 2023-08-07:
The above change has been [merged](https://github.com/espressif/esp-idf/commit/4258c409b0bc9e534fd6259bbd3ec4cb1d2e23c1) in github. Please feel free to reopen if issue still observed.

closed 2023-08-07 by rahult-github