espressif/esp-idf #11825: invalid conversion from 'int' to 'ip_event_t' when compiling esp_sntp-client with g++

MartyMcFlyInTheSky, 2023-07-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

Building a esp-idf project with g++ compiler works.

### What is the actual behavior?

Using espressifs new esp_sntp.h wrapper around lwip sntp the app doesn't build using the current g++ fork (I believe 12.1) with all warnings as errors (-Wall). This is the output at the respective place:

```
C:/dev/esp/esp-idf/components/esp_netif/include/esp_netif_sntp.h:53:34: warning: invalid conversion from 'int' to 'ip_event_t' [-fpermissive]
   53 |             .ip_event_to_renew = 0,                 \
      |                                  ^
      |                                  |
      |                                  int
C:/dev/esp/esp-idf/components/esp_netif/include/esp_netif_sntp.h:63:13: note: in expansion of macro 'ESP_NETIF_SNTP_DEFAULT_CONFIG_MULTIPLE'
   63 |             ESP_NETIF_SNTP_DEFAULT_CONFIG_MULTIPLE(1, {server})
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:/dev/aurora-light-controller/components/aurora_sntp/aurora_sntp.cpp:23:40: note: in expansion of macro 'ESP_NETIF_SNTP_DEFAULT_CONFIG'
   23 |             esp_sntp_config_t config = ESP_NETIF_SNTP_DEFAULT_CONFIG("pool.ntp.org");
      |                                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~ 
```

C++ is a bit more strict in treating enum types, so you have to give it an enum for it to compile. Default value 0 corresponds to ip_event_t::IP_EVENT_STA_GOT_IP. Would be an easy fix, but I have to use `-fpermissive` for now which is very annoying. I believe this fix has been applied already to the master branch, would be nice if you could retrofit it to release/v5.1 as we're currently checked out on that!

### Steps to reproduce.

Compile project with esp sntp client wrapper and initialize the esp_sntp client using the `ESP_NETIF_SNTP_DEFAULT_CONFIG_MULTIPLE `macro.

david-cermak, 2023-07-28:
Thanks for reporting this problem and sorry for the inconvenience.
We forgot to backport 2ff12460dd59be26ea485daf3ace5ecac4bb14a0 to `v5.1`. Will fix.

xmow49, 2023-09-08:
The problem is still present, any news ?
Thanks

AxelLin, 2023-10-03:
v5.1 fix: ef451cab0fec03b24d71552054c8489c2bd6217b

david-cermak, 2023-10-26:
Fixed in https://github.com/espressif/esp-idf/commit/597810bf96f90d93a98cc8641f90e5bd09a75ff0

closed 2023-10-26 by david-cermak