espressif/esp-idf #11829: SD SPI external CS

Syrinx98, 2023-07-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi, i briefly describe my situation: 
I have the SPI bus connected with a w5500 and a SD Card, the particularity is that i have an external CS for both devices and i control them with an I/O expander, i managed to make it work with w5500, simply setting the CS to zero when i have to do the transaction.
The problem is that i don't understand how to do it with sdcard, i took the initialization code and try write "-1" to the Chip Select configuration, like thi:

`void init_spi()
{
	esp_err_t ret;
	ESP_LOGI(SPI_INIT_TAG, "Using SPI peripheral");

	// di default la frequenza SDMMC_FREQ_DEFAULT è inizializzata a (20MHz)
	host.max_freq_khz = 5000;

	const spi_bus_config_t bus_config = 
	{
		.mosi_io_num = CONFIG_SPI_MOSI_GPIO,
		.miso_io_num = CONFIG_SPI_MISO_GPIO,
		.sclk_io_num = CONFIG_SPI_CLK_GPIO,
		.quadwp_io_num = -1,
		.quadhd_io_num = -1,
		.max_transfer_sz = 0,
		.flags = 0,
	};
	ret = spi_bus_initialize(host.slot, &bus_config, SDSPI_DEFAULT_DMA);
	
	if (ret != ESP_OK) 
	{
		ESP_LOGE(SPI_INIT_TAG, "Failed to initialize bus");
		return;
	}
	
	
	// mount sd
	esp_vfs_fat_sdmmc_mount_config_t mount_config = 
	{
		.format_if_mount_failed = false,
		.max_files = 5,
		.allocation_unit_size = 16 * 1024
	};
	
	sdmmc_card_t *card = NULL;
	const char mount_point[] = MOUNT_POINT;
	
	ESP_LOGI(SPI_INIT_TAG, "Initializing SD card");
	
	sdspi_device_config_t slot_config = SDSPI_DEVICE_CONFIG_DEFAULT();
	slot_config.gpio_cs = -1;
	slot_config.host_id = SPI2_HOST;

	ESP_LOGI(SPI_INIT_TAG, "Mounting filesystem");
	ret = esp_vfs_fat_sdspi_mount(mount_point, &host, &slot_config, &mount_config, &card);

	if (ret != ESP_OK) 
	{
		if (ret == ESP_ERR_TIMEOUT || (ret == ESP_ERR_NOT_SUPPORTED || ret == ESP_ERR_INVALID_ARG)) 
		{ 
			ESP_LOGE(SPI_INIT_TAG, "SD CARD NON INSERITA, NON SUPPORTATA O NON CONFIGURATA CORRETTAMENTE");
		}
		else if (ret == ESP_ERR_NOT_FOUND) 
		{
			ESP_LOGE(SPI_INIT_TAG, "ESP_ERR_NOT_FOUND");
		}
		return;
	}
	ESP_LOGI(SPI_INIT_TAG, "Filesystem mounted");
}`

It gives me "ESP_ERR_INVALID_ARG", i cannot set the CS pin for the SD Card to -1, how can i do it?

igrr, 2023-07-07:
This is probably a duplicate of https://github.com/espressif/esp-idf/issues/9219.

nopnop2002, 2023-07-11:
@Syrinx98 
When using SDSPI (SD card over SPI), SDSPI uses the SPI2_HOST BUS.
SPI3_HOST must be used when using other spi devices at the same time as SDSPI.

igrr, 2023-07-11:
@nopnop2002 Sharing a single SPI bus for an SD card an another device is supported since v4.3. I think this issue is slightly different, it's about providing a way for the SDSPI driver (and possibly SPI Ethernet driver) to control the CS line via a GPIO expander.

nopnop2002, 2023-07-11:
Thank you for your feedback.

I remember being told here a while ago how to add another SPI device to the same SPI bus as SDSPI.

https://github.com/espressif/esp-idf/issues/6510