espressif/esp-idf #11831: ESP-IDF Toolchain Breaks Encapsulation on Linux During Install

marandumich, 2023-07-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.5-104-g8b94183c9c, v4.4.1, v4.4.1-405-g6c5fb29c2c

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

When switching between ESP-IDF versions, I expected activating the relevant toolchain version to be as simple as `. ./esp-idf-versionX/export.sh`.

I am working with several projects which use different ESP-IDF versions. This is required, as I am working in a multidisciplinary research/education environment, for internal deployability and consistency.

### What is the actual behavior?

ESP-IDF breaks project-specific incapsulation during its install process by storing config files in `~/.espressif`. 

When versionY is installed, running `. ./esp-idf-versionX/export.sh` can cause crashes. In my case, I get the resulting message:
```
Adding ESP-IDF tools to PATH...
Traceback (most recent call last):
  File "/home/marand/tmp/esp/esp2/tools/idf_tools.py", line 1858, in <module>
    main(sys.argv[1:])
  File "/home/marand/tmp/esp/esp2/tools/idf_tools.py", line 1854, in main
    action_func(args)
  File "/home/marand/tmp/esp/esp2/tools/idf_tools.py", line 1176, in action_export
    tools_info = filter_tools_info(tools_info)
  File "/home/marand/tmp/esp/esp2/tools/idf_tools.py", line 1118, in filter_tools_info
    targets = get_user_defined_targets()
  File "/home/marand/tmp/esp/esp2/tools/idf_tools.py", line 1098, in get_user_defined_targets
    if env == idf_env_json['idfSelectedId']:
KeyError: 'idfSelectedId'
```

### Steps to reproduce.

```
# Clone relevant esp-idf versions
mkdir -p esp1 && git clone --recursive https://github.com/espressif/esp-idf.git -b release/v4.4 esp1
mkdir -p esp2 && git clone --recursive https://github.com/espressif/esp-idf.git -b v4.4.1 esp2
mkdir -p esp3 && git clone --recursive https://github.com/espressif/esp-idf.git esp3
pushd esp3
    git checkout v4.4.1-405-g6c5fb29c2c  # this is not a formal tag
    git submodule sync
    gt submodule update --recursive
popd

# Setup release/v4.4
pushd esp1
    ./install.sh
popd

# Compare with v4.4.1
pushd esp2
    . ./export.sh
popd

# Compare with old 4.4.1
pushd esp3
    . ./export.sh
popd
```

### More Information.

This crash is part of instantiating the toolchain, and does not involve compilation of code

igrr, 2023-07-07:
Sorry for the trouble. This issue was fixed in v4.4.4 (https://github.com/espressif/esp-idf/commit/640f1c844b5cd935e809b0195df06f1639eb9979, https://github.com/espressif/esp-idf/issues/9837).
There is a workaround for older v4.4.x releases, mentioned in the linked issue.

marandumich, 2023-07-07:
Thank you for your rapid reply. To clarify, are you suggesting that I upgrade all existing projects to use `v4.4.4` or later? I have attempted your suggestions and run into new errors. Broadly, it seems that

## All Versions of Interest Are Incompatible
I have adjusted my scripts demonstrating this to mitigate environment variable contamination.  This loops over 4 ESP-IDF versions, permutes the `install/export` order to see which are compatible with each other. It seems that none are. Output is attached.
```
#!/usr/bin/env bash

# Clone relevant esp-idf versions
mkdir -p esp1 && git clone --recursive https://github.com/espressif/esp-idf.git -b release/v4.4 esp1
mkdir -p esp2 && git clone --recursive https://github.com/espressif/esp-idf.git -b v4.4.1 esp2
mkdir -p esp3 && git clone --recursive https://github.com/espressif/esp-idf.git esp3
mkdir -p esp4 && git clone --recursive https://github.com/espressif/esp-idf.git -b v4.4.4 esp4
pushd esp3 &> /dev/null
    git checkout v4.4.1-405-g6c5fb29c2c  &> /dev/null  # this is not a formal tag
    git submodule sync &> /dev/null  # simplify stdout
    gt submodule update --recursive &> /dev/null  # simplify stdout
popd &> /dev/null

reset() {
    unset $(env | awk -F= '{print $1}' | grep -E ^IDF)  # IDF env vars
    rm -rf ~/.espressif  # install directory
}

# Install and run compatibility check
arr=( esp1 esp2 esp3 esp4 )
for v1 in ${arr[@]}; do
    # wipe env vars
    reset
    
    # install
    tag1="$(cd "$v1"; git describe --tag)"
    echo "installing $tag1"
    pushd $v1 &>/dev/null
        ./install.sh &> /dev/null
    popd&>/dev/null
    echo "done!"
    
    # loop over to check install
    for v2 in ${arr[@]}; do
        if [ "$v1" = "$v2" ]; then continue; fi  # skip self-check
        (cd $v2; . ./export.sh)  # subshell to avoid env vars
        tag2="$(cd "$v2"; git describe --tag)"
        echo "^result for $tag2 using $tag1 install"; echo
    done
    echo;echo
done
```

### Output
```
fatal: destination path 'esp1' already exists and is not an empty directory.
fatal: destination path 'esp2' already exists and is not an empty directory.
fatal: destination path 'esp3' already exists and is not an empty directory.
fatal: destination path 'esp4' already exists and is not an empty directory.
installing v4.4.5-104-g8b94183c9c
done!
Setting IDF_PATH to '/home/marand/tmp/esp2'
Detecting the Python interpreter
Checking "python" ...
Checking "python3" ...
Python 3.8.10
"python3" has been detected
Adding ESP-IDF tools to PATH...
Traceback (most recent call last):
  File "/home/marand/tmp/esp2/tools/idf_tools.py", line 1858, in <module>
    main(sys.argv[1:])
  File "/home/marand/tmp/esp2/tools/idf_tools.py", line 1854, in main
    action_func(args)
  File "/home/marand/tmp/esp2/tools/idf_tools.py", line 1176, in action_export
    tools_info = filter_tools_info(tools_info)
[... 159 lines cut ...]
    if env == idf_env_json['idfSelectedId']:
KeyError: 'idfSelectedId'
^result for v4.4.1 using v4.4.4 install

Setting IDF_PATH to '/home/marand/tmp/esp3'
Detecting the Python interpreter
Checking "python" ...
Checking "python3" ...
Python 3.8.10
"python3" has been detected
Adding ESP-IDF tools to PATH...
Traceback (most recent call last):
  File "/home/marand/tmp/esp3/tools/idf_tools.py", line 1860, in <module>
    main(sys.argv[1:])
  File "/home/marand/tmp/esp3/tools/idf_tools.py", line 1856, in main
    action_func(args)
  File "/home/marand/tmp/esp3/tools/idf_tools.py", line 1177, in action_export
    tools_info = filter_tools_info(tools_info)
  File "/home/marand/tmp/esp3/tools/idf_tools.py", line 1119, in filter_tools_info
    targets = get_user_defined_targets()
  File "/home/marand/tmp/esp3/tools/idf_tools.py", line 1099, in get_user_defined_targets
    if env == idf_env_json['idfSelectedId']:
KeyError: 'idfSelectedId'
^result for v4.4.1-405-g6c5fb29c2c using v4.4.4 install
```

igrr, 2023-07-07:
Yes, I would recommend upgrading from v4.4.x (x<=4) to the latest v4.4.x bugfix release (v4.4.5 at the time of writing). If you can't upgrade IDF to that release, the workarounds are:
* Delete ~/.espressif/idf-env.json every time before running the install/export scripts.
* Set different IDF_TOOLS_PATH environment variables before working with each of the versions. This will ensure that the problematic idf-env.json file would not be shared by different versions.

Again, if you have the option, I'd recommend updating to the latest bugfix release.

marandumich, 2023-07-07:
Duly noted. Thank you for the pointers.

closed 2023-07-07 by marandumich