espressif/esp-idf #11832: BLE - Nimble. Save bonding information in NVS

guirespi, 2023-07-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.2

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32 WROVER KIT

### Power Supply used.

USB

### What is the expected behavior?

When enabling BLE Nimble to save Bonding information in NVS, after the ESP32 reboots, the pairing request should not appear.

### What is the actual behavior?

When enabling BLE Nimble to save Bonding information in NVS, after the ESP32 reboots, I have to accept the pairing again event if the pairing was accepted before the reboot. If the ESP32 does not reboot, the pairing request is not longer required.

### Steps to reproduce.

Use whatever BLE peripheral example and enable the option in SDK Config.

### More Information.

I search through the IDF folder and I found that in some examples this steps to initialize BLE peripheral:

```
void app_main(void){
    int rc;

    /* Initialize NVS — it is used to store PHY calibration data */
    esp_err_t ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
        ESP_ERROR_CHECK(nvs_flash_erase());
        ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK(ret);

    ESP_LOGI(TAG, "ESP_WIFI_MODE_STA");
    wifi_init_sta();
    do_ping_cmd();
    ESP_ERROR_CHECK(esp_nimble_hci_and_controller_init());

    nimble_port_init();
    /* Initialize the NimBLE host configuration. */
    ble_hs_cfg.reset_cb = bleprph_on_reset;
    ble_hs_cfg.sync_cb = bleprph_on_sync;
    ble_hs_cfg.gatts_register_cb = gatt_svr_register_cb;
    ble_hs_cfg.store_status_cb = ble_store_util_status_rr;

    rc = gatt_svr_init();
    assert(rc == 0);

    /* Set the default device name. */
    rc = ble_svc_gap_device_name_set("nimble-bleprph");
    assert(rc == 0);

    /* XXX Need to have template for store */
    ble_store_config_init();

    nimble_port_freertos_init(bleprph_host_task);
}

```

Even when I try this type of init in my project, the behaviour is even more erratic. Sometimes the pairing is not needed after a reboot. Even before the reboot, with this code, the pairing request is needed again.

ESPAbhinav, 2023-07-13:
Hi  @guirespi ,
I request you to please sync to the latest release v4.4 codebase and try it once.

guirespi, 2023-07-13:
Hi @ESPAbhinav, I updated my IDF to v4.4.5 and at the first try the pairing request after reboot is not longer needed. I havea to cellphones, one with Android 12 and another one with Android 13. Everything was alright using the code I mention at my first comment, but adding the flags BLE_SM_PAIR_KEY_DIST_ENC|BLE_SM_PAIR_KEY_DIST_ID to the sm_our_key_dist and sm_their_key_dist members. The bonding saved in the NVS works great until I flash other binary to the esp32 without this change, then the Android 12 phone doesnt finisht to connect the esp32 and always disconnect and even doesnt send the pairint request anymore. The Android 13 phone starts sending pairing request each time it connects to the ESP32, event without reboot.

After seeing this behaviour, I flash the version with the changes i mentionedabove and the behaviour is still the same, the Android 12 phone nevers connect and the Android 13 phone always need pairing request. So, is there a problem with the template used to save the bonding information? Knowing that at first instance the code works, I will try it in the IDF v4.4.2 later.

ESPAbhinav, 2023-07-21:
Hi, @guirespi
We tried to reproduce your issue with Android 11 and 13  and have the following result.

1. nvs works fine on Android 11.
2. I have obtained the result on two Android-13 phones out of which one failed to bond across the reboots.
3. When we tried to debug it is observed that nvs store is storing all the information of previously established bonds correctly. It seems like a problem is arising when the device is sending new peer_addr across the reboot and hence we are getting idx value -1 while searching required value_sec. 

So to confirm things can you please share the debug enable logs on Android 13 and Android 12 for one run in the following manner?
First-time pair and connect the nrf with the epsressif device, then reboot the espressif device and then attempt the connection. These logs will be helpful to point out the reason.

guirespi, 2023-07-21:
Hi, @ESPAbhinav 
I'm a bit confused, do you want the the esp32's BLE debug logs or the Android 12 and 13 logs on our mobile app development enviroment?

ESPAbhinav, 2023-07-24:
Hi, @guirespi,
So we want the esp32's debug enable logs when you are getting issues with nvs on Android 13 and Android 12.
You can follow the below-mentioned process for taking logs.

1. To enable the debug logs 

                            Component Config  --> Log Output --> Default Log Verbosity --> Debug

                            Component Config --> Bluetooth --> NimBLE Options --> NimBLE Host log verbosity --> Debug Logs 

2. To perform the operations 

-     First-time pair and connect the nRF-app from Android 13  with the epsressif device, then reboot the espressif device and then attempt the connection.
-     Repeat the same process for Android 12.

ESPAbhinav, 2023-08-22:
Hi @guirespi ,
Any Update on this ?

ESPAbhinav, 2023-09-20:
Closing this issue . Please feel free to reopen the issue in case of any updates.

closed 2023-11-20 by tomassebestik