espressif/esp-idf #11833: ESP-TLS: "requesting more data than fits"

bmkuter, 2023-07-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am working on a device with 2 TLS connections, and the second connection is repeatedly returning `requesting more data than fits`. My config file should limit this to a 16k buffer via

```
#
# mbedTLS
#
CONFIG_MBEDTLS_INTERNAL_MEM_ALLOC=y
# CONFIG_MBEDTLS_DEFAULT_MEM_ALLOC is not set
# CONFIG_MBEDTLS_CUSTOM_MEM_ALLOC is not set
CONFIG_MBEDTLS_ASYMMETRIC_CONTENT_LEN=y
CONFIG_MBEDTLS_SSL_IN_CONTENT_LEN=16384
CONFIG_MBEDTLS_SSL_OUT_CONTENT_LEN=4096
CONFIG_MBEDTLS_DEBUG=y
# CONFIG_MBEDTLS_DEBUG_LEVEL_WARN is not set
# CONFIG_MBEDTLS_DEBUG_LEVEL_INFO is not set
CONFIG_MBEDTLS_DEBUG_LEVEL_DEBUG=y
# CONFIG_MBEDTLS_DEBUG_LEVEL_VERBOSE is not set
CONFIG_MBEDTLS_DEBUG_LEVEL=3
```

My code so far tries to read 3000 bytes per call to `esp_tls_con_read()`, which works as expected in my other connection. The main difference between the two connections is one only ever returns under the buffer size, while the other is streaming a lot of data (800 KB+). 

Here is the relevant code snippet so far:

```
void tls_rx_task(void *parameters)
{
    int index = *((int *) parameters);
    tls_message_t message = {0};
    ESP_LOGW("", "Hello World from tls_task[%d]!", index);

    if(tls_connection_array[index].tls == 0)
    {
        ESP_LOGW("",  "tls_task[%d]: TLS context not initialized. Closing task...", index);
        vTaskDelete( NULL );
    }

    while (1)
    {
        // If the mutex can be taken...
        if(xSemaphoreTake(cloud_mutex, (TickType_t) 500/portTICK_PERIOD_MS) == pdTRUE) // Try to take the mutex within 10 ticks
        {
            if(tls_connection_array[index].rx_bytes_available == 0 && tls_connection_array[index].tls != NULL)
            {
                ESP_LOGE(TAG, "Free memory before TLS read: %d bytes", esp_get_free_heap_size());
                tls_connection_array[index].rx_bytes_available = esp_tls_conn_read(tls_connection_array[index].tls, message.buffer, MESSAGE_SIZE);
                ESP_LOGE(TAG, "Free memory after  TLS read: %d bytes", esp_get_free_heap_size());

                if (tls_connection_array[index].rx_bytes_available > 0)
                {
                    ESP_LOGW("", "tls_task[%d]: Read %d bytes", index, tls_connection_array[index].rx_bytes_available);
                    message.size = tls_connection_array[index].rx_bytes_available;
                    xQueueSend(tls_connection_array[index].message_queue, &message, portMAX_DELAY);
                }
                else if (tls_connection_array[index].rx_bytes_available == -0x6900)
                {
                    tls_connection_array[index].rx_bytes_available = 0;
                }
                else if (tls_connection_array[index].rx_bytes_available < 0 )
                {
                    if (tls_connection_array[index].tls) esp_tls_conn_destroy(tls_connection_array[index].tls);
                    ESP_LOGW("", "tls_task[%d]: MBEDTLS_ERR_NET_RECV_FAILED ", index);
                    ESP_LOGE(TAG, "Free memory after  MBEDTLS_ERR_NET_RECV_FAILED: %d bytes", esp_get_free_heap_size());
                    memset(&tls_connection_array[index], 0x00, sizeof(tls_manager_struct_t));
                    xSemaphoreGive(cloud_mutex);
                    vTaskDelete( NULL );
                }
            }
            // Give the mutex back before you potentially delay the task.
            if(tls_connection_array[index].rx_bytes_available != 0)
            {
                ESP_LOGW("", "tls_task[%d]: Ignoring read, message already in queue of %d bytes", index, tls_connection_array[index].rx_bytes_available);
            }
            xSemaphoreGive(cloud_mutex);
        }
        else 
        {
            ESP_LOGW("", "tls_task[%d]: Mutex not available for reading, trying later...", index);
        }
        vTaskDelay(1000/portTICK_PERIOD_MS);
    }

    ESP_LOGW("","tls_task[%d]: ERROR. OUT OF MAIN LOOP", index);

    vTaskSuspend(NULL);
}
```

```
E (78427) example: Free memory after  TLS read: 38968 bytes
E (78428) example: Free memory before TLS read: 38968 bytes
I (78434) mbedtls: ssl_msg.c:5441 => read

I (78439) mbedtls: ssl_msg.c:3887 => read record

I (78445) mbedtls: ssl_msg.c:1926 => fetch input

I (78450) mbedtls: ssl_msg.c:2066 in_left: 0, nb_want: 5

I (78456) mbedtls: ssl_msg.c:2086 in_left: 0, nb_want: 5

I (78463) mbedtls: ssl_msg.c:2089 ssl->f_recv(_timeout)() returned 5 (-0xfffffffb)

I (78471) mbedtls: ssl_msg.c:2111 <= fetch input

I (78476) mbedtls: ssl_msg.c:1926 => fetch input

W (78482) mbedtls: ssl_msg.c:1934 requesting more data than fits

W (78489) mbedtls: ssl_msg.c:4639 mbedtls_ssl_fetch_input() returned -28928 (-0x7100)

W (78497) mbedtls: ssl_msg.c:3918 ssl_get_next_record() returned -28928 (-0x7100)

W (78506) mbedtls: ssl_msg.c:5501 mbedtls_ssl_read_record() returned -28928 (-0x7100)

E (78514) esp-tls-mbedtls: read error :-0x7100:
E (78520) example: Free memory after  TLS read: 38968 bytes
W (78527) : tls_task[1]: MBEDTLS_ERR_NET_RECV_FAILED
E (78532) example: Free memory after  TLS read: 73416 bytes
E (79428) example: Free memory before TLS read: 81088 bytes
```

Harshal5, 2023-08-07:
Hello @bmkuter,

Sorry for the delayed response. Thanks for reporting the issue.
Could you please share a minimal issue-reproducible code for the same? It would really help into debugging this issue.

Thanks!

Harshal5, 2023-09-11:
@bmkuter

Closing this issue as of now due to lack of feedback. Please feel free to re-open.

closed 2023-09-11 by Harshal5

CRodriguezAirz, 2024-08-20:
Hello @bmkuter , same issue here. Were you able to resolve it?

bmkuter, 2024-08-20:
Hello! Its been a while, so let me dig up the code. Last I checked I got it working somehow, but I remember it was strange.