espressif/esp-idf #11837: Checkout V5.1 command

ginodecock, 2023-07-09:
Hi,

Please change your documentation to correct the checkout command to:
git clone -b v5.1 --recursive https://github.com/espressif/esp-idf.git esp-idf-v5.1

==> extra space between V5.1 and --recursive

igrr, 2023-07-09:
Hi @ginodecock, I've checked the [getting started guide](https://docs.espressif.com/projects/esp-idf/en/v5.1/esp32/get-started/linux-macos-setup.html#step-2-get-esp-idf) and the command there looks correct. Could you please give a link to the page where you have noticed the misspelled command?

ginodecock, 2023-07-09:
Hi,

You can find it here: https://github.com/espressif/esp-idf/releases/tag/v5.1 at the using git section.

AdityaHPatwardhan, 2023-07-10:
Hi @ginodecock Thanks for pointing out the issue.
I have fixed the typo in the release notes https://github.com/espressif/esp-idf/releases/tag/v5.1
Please take a look once. 
I am closing the issue since it seems to be resolved, feel free to re-open if you think otherwise.

closed 2023-07-10 by AdityaHPatwardhan