espressif/esp-idf #11838: how can I encode 100 symbols completely in RMT?

lazypikapig, 2023-07-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

    I want to use the RMT module to send an infrared sequence with 100 symbols, but it seems to hit RMT_ENCODING_MEM_FULL. So how could I send the whole 100 or more symbols?
    I used global variable to record the results(encoded symbols) of each step of encoding. Here is my code of encoding, based on esp-idf example:

    `static size_t rmt_encode_ir_nec(rmt_encoder_t *encoder, rmt_channel_handle_t channel, const void *primary_data, size_t data_size, rmt_encode_state_t *ret_state)
{
    rmt_ir_nec_encoder_t *nec_encoder = __containerof(encoder, rmt_ir_nec_encoder_t, base);
    rmt_encode_state_t session_state = 0;
    rmt_encode_state_t state = 0;
    size_t encoded_symbols = 0;
    ir_nec_transmit_command_code_t *ir_nec_transmit_command_code = (ir_nec_transmit_command_code_t *)primary_data;
    rmt_encoder_handle_t copy_encoder = nec_encoder->copy_encoder;
    rmt_encoder_handle_t bytes_encoder = nec_encoder->bytes_encoder;
    switch (nec_encoder->state) {
    case 0: // send leading code
        encoded_symbols += copy_encoder->encode(copy_encoder, channel, &nec_encoder->nec_leading_symbol,
                                                sizeof(rmt_symbol_word_t), &session_state);
        g_encoded_symbols_after_leading_code = encoded_symbols;
        if (session_state & RMT_ENCODING_COMPLETE) {
            nec_encoder->state = 1; // we can only switch to next state when current encoder finished
        }
        if (session_state & RMT_ENCODING_MEM_FULL) {
            state |= RMT_ENCODING_MEM_FULL;
            goto out; // yield if there's no free space to put other encoding artifacts
        }
    // fall-through
    case 1: // send command 1
        encoded_symbols += bytes_encoder->encode(bytes_encoder, channel, &ir_nec_transmit_command_code->command1,
                                                 sizeof(uint16_t), &session_state);
 
        g_encoded_symbols_after_command1 = encoded_symbols;
        if (session_state & RMT_ENCODING_COMPLETE) {
            nec_encoder->state = 2; // we can only switch to next state when current encoder finished
        }
        if (session_state & RMT_ENCODING_MEM_FULL) {
            state |= RMT_ENCODING_MEM_FULL;
            goto out; // yield if there's no free space to put other encoding artifacts
        }
    // fall-through
    case 2: // send command 2
        encoded_symbols += bytes_encoder->encode(bytes_encoder, channel, &ir_nec_transmit_command_code->command2,
                                                 sizeof(uint16_t), &session_state);
        g_encoded_symbols_after_command2 = encoded_symbols;                                          
        if (session_state & RMT_ENCODING_COMPLETE) {
            nec_encoder->state = 3; // we can only switch to next state when current encoder finished
        }
        if (session_state & RMT_ENCODING_MEM_FULL) {
            state |= RMT_ENCODING_MEM_FULL;
            goto out; // yield if there's no free space to put other encoding artifacts
        }
    // fall-through
    case 3: // send command 3
        encoded_symbols += bytes_encoder->encode(bytes_encoder, channel, &ir_nec_transmit_command_code->command3,
                                                 sizeof(uint16_t), &session_state);
 
        g_encoded_symbols_after_command3 = encoded_symbols;                                        
        if (session_state & RMT_ENCODING_COMPLETE) {
            nec_encoder->state = 4; // we can only switch to next state when current encoder finished
        }
        if (session_state & RMT_ENCODING_MEM_FULL) {
            state |= RMT_ENCODING_MEM_FULL;
            goto out; // yield if there's no free space to put other encoding artifacts
        }
    // fall-through
    case 4: // send middle code 1
        encoded_symbols += copy_encoder->encode(copy_encoder, channel, &nec_encoder->nec_middle_symbol[0],
                                                sizeof(rmt_symbol_word_t), &session_state);
        g_encoded_symbols_after_middle_code1 = encoded_symbols;
        if (session_state & RMT_ENCODING_COMPLETE) {
            nec_encoder->state = 5; // we can only switch to next state when current encoder finished
        }
        if (session_state & RMT_ENCODING_MEM_FULL) {
            state |= RMT_ENCODING_MEM_FULL;
            goto out; // yield if there's no free space to put other encoding artifacts
        }
    // fall-through
    case 5: // send middle code 2
        encoded_symbols += copy_encoder->encode(copy_encoder, channel, &nec_encoder->nec_middle_symbol[1],
                                                sizeof(rmt_symbol_word_t), &session_state);
        g_encoded_symbols_after_middle_code2 = encoded_symbols;
        if (session_state & RMT_ENCODING_COMPLETE) {
            nec_encoder->state = 6; // we can only switch to next state when current encoder finished
        }
        if (session_state & RMT_ENCODING_MEM_FULL) {
            state |= RMT_ENCODING_MEM_FULL;
            goto out; // yield if there's no free space to put other encoding artifacts
        }
    // fall-through
    case 6: // send command 4
        encoded_symbols += bytes_encoder->encode(bytes_encoder, channel, &ir_nec_transmit_command_code->command4,
                                                 sizeof(uint16_t), &session_state);
        g_encoded_symbols_after_command4 = encoded_symbols;  
        if (session_state & RMT_ENCODING_COMPLETE) {
            nec_encoder->state = 7; // we can only switch to next state when current encoder finished
        }
        if (session_state & RMT_ENCODING_MEM_FULL) {
            state |= RMT_ENCODING_MEM_FULL;
            goto out; // yield if there's no free space to put other encoding artifacts
        }
    // fall-through
    case 7: // send command 5
        encoded_symbols += bytes_encoder->encode(bytes_encoder, channel, &ir_nec_transmit_command_code->command5,
                                                 sizeof(uint16_t), &session_state);
        g_encoded_symbols_after_command5 = encoded_symbols;  
        if (session_state & RMT_ENCODING_COMPLETE) {
            nec_encoder->state = 8; // we can only switch to next state when current encoder finished
        }
        if (session_state & RMT_ENCODING_MEM_FULL) {
            state |= RMT_ENCODING_MEM_FULL;
            goto out; // yield if there's no free space to put other encoding artifacts
        }
    // fall-through
    case 8: // send command 6
        encoded_symbols += bytes_encoder->encode(bytes_encoder, channel, &ir_nec_transmit_command_code->command6,
                                                 sizeof(uint16_t), &session_state);
        g_encoded_symbols_after_command6 = encoded_symbols;  
        if (session_state & RMT_ENCODING_COMPLETE) {
            nec_encoder->state = 9; // we can only switch to next state when current encoder finished
        }
        if (session_state & RMT_ENCODING_MEM_FULL) {
            state |= RMT_ENCODING_MEM_FULL;
            g_mem_full_flag = 1;
            goto out; // yield if there's no free space to put other encoding artifacts
        }
    // fall-through
    case 9: // send ending code
        encoded_symbols += copy_encoder->encode(copy_encoder, channel, &nec_encoder->nec_ending_symbol,
                                                sizeof(rmt_symbol_word_t), &session_state);
        g_encoded_symbols_after_end_code = encoded_symbols;
        if (session_state & RMT_ENCODING_COMPLETE) {
            nec_encoder->state = 0; // back to the initial encoding session
            state |= RMT_ENCODING_COMPLETE;
        }
        if (session_state & RMT_ENCODING_MEM_FULL) {
            state |= RMT_ENCODING_MEM_FULL;
            goto out; // yield if there's no free space to put other encoding artifacts
        }
    }
out:
    *ret_state = state;
    g_encoded_symbols = encoded_symbols;
    return encoded_symbols;
}`

And the result output:
g_encoded_symbols_after_leading_code = 1
g_encoded_symbols_after_command1 = 17
g_encoded_symbols_after_command2 = 33
g_encoded_symbols_after_command3 = 49
g_encoded_symbols_after_middle_code1 = 50
g_encoded_symbols_after_middle_code2 = 51
g_encoded_symbols_after_command4 = 67
g_encoded_symbols_after_command5 = 83
g_encoded_symbols_after_command6 = 96
g_encoded_symbols_after_end_code = 0
g_mem_full_flag = 1

It seems that memory was full in command6 encoding.
how can I encode 100 symbols completely?

suda-morris, 2023-07-10:
@lazypikapig  when there's no memory for saving the encoding artifacts, the scheme will **yield** in the middle (see the `goto` statements in your encoding function). The encoding scheme can resume when half of the encoded RMT symbols are transmitted.

lazypikapig, 2023-07-10:
@suda-morris 
thanks for your reply. 
the function 'rmt_isr_handle_tx_threshold' is saying 'continue pingpong transmission' indeed, but I just find out that it just only call my encoder again, without something like 'rmt_ll_tx_start' or any other function used to transmit again.
could you remind me some hints?
thanks again!

suda-morris, 2023-07-11:
@lazypikapig No, we don't need to call a "start" function because when we are doing the ping-pong operation, the transmission is still undergoing

lazypikapig, 2023-07-11:
@suda-morris

closed 2023-07-12 by lazypikapig