espressif/esp-idf #11841: tcp link often occur disconnect

luacerbb, 2023-07-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

idf 4.4.4

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

cutom board

### Power Supply used.

External 5V

### What is the expected behavior?

socket work well

### What is the actual behavior?

tcp often  disconnect and need a long time can restore, but  when tcp disconnect ,udp work well. if  dev power on again, tcp can  work well right away.

### Steps to reproduce.

when dev work several hours,tcp will disconnect,i will try recreat tcp link,but it wil disconnted for a long time, if I restart system ,it will create tcp link right away.

david-cermak, 2023-07-28:
> tcp often disconnect and need a long time can restore, but when tcp disconnect ,udp work well. if dev power on again, tcp can work well right away.

How many sockets are you using? How do you reconnect? Could you please share some portions of your code, especially the way how you handle disconnections and restoring the connection?

david-cermak, 2023-10-12:
@luacerbb Feel free to reopen with more details.

closed 2023-10-12 by david-cermak